amzn/amzn-drivers #193: get feature comment discrepancy

rzezeski, 2021-11-13:
In the common code definitions for the get feature command is the following comment:

```
struct ena_admin_get_set_feature_common_desc {
	/* 1:0 : select - 0x1 - current value; 0x3 - default
	 *    value
	 * 7:3 : reserved3
	 */
	u8 flags;
```

The comment indicates that bits 0-1 should hold a value of `0x1` to get the current value and `0x3` to get the default value. However, the `ena_com_get_feature_ex()` function always sets it to zero by virtue of a `memset()` with no follow up modification to `feat_common.flags`.

In my illumos port I follow the comments and set these bits to `0x1`, and it seems to work fine. But I though it would be good to double check the intended behavior here and see if anyone has insight into this.

davidarinzon, 2021-11-15:
Hi @rzezeski,
Thank you for pointing this out, the intention is as per the documentation in the comment, but this functionality is not yet supported, therefore, setting it to 0x0 will be sufficient. You may consider these fields as reserved for now, we will update this documentation accordingly in the future.
Please let me know if you have additional questions.

rzezeski, 2021-11-17:
Okay, that's good to know. Thank you.

closed 2021-11-19 by davidarinzon