amzn/amzn-drivers #196: ENA: Differences to Linux in-tree ena driver

Vincinator, 2021-12-10:
Hey Hey!

Hope you can help me understand the differences to the in-tree ena driver variant. 

What are the differences between the ena linux in-tree driver `drivers/net/ethernet/amazon/ena`  and the ena kernel module code provided in this repository? 
How can I determine the ena version (of this repo) used for `drivers/net/ethernet/amazon/ena` of Linux? 

Thanks!
Vincent

ShayAgros, 2021-12-12:
Hi,
the two code bases don't contain exactly the same code (for example, GIthub contains Local Page Caches and module parameters while the upstream code doesn't). Work is being done to make them as identical as possible over time.

Unfortunately we can no longer update the driver version number in the upstream code, so the only way to find if some code in Github exists upstream is to look for a cover letter that specifies it (something like [this link](https://patchwork.ozlabs.org/project/netdev/list/?series=&submitter=79035&state=*&q=&archive=both&delegate=)) or to actually install this driver and check for your feature.

If you want to ask about a specific upstream version of the driver and its support for a feature please let me know in this ticket

Vincinator, 2021-12-14:
Thanks @ShayAgros!

The monitoring of [network performance](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/monitoring-network-performance-ena.html) is a feature some of our users are looking for. Since this feature requires ENA driver version above 2.2.10, the questions was raised to upgrade to a newer driver version.

Based on the above, I will go forward with the Github code.

Thanks a lot for your help

davidarinzon, 2021-12-19:
Hi @Vincinator 
Do you require any further assistance?

Vincinator, 2021-12-21:
@davidarinzon No - I will close this issue. 
Thank you!

closed 2021-12-21 by Vincinator