amzn/amzn-drivers #198: get-vfio-with-wc.sh fails on Ubuntu Server 20.04 LTS (64-bit x86): failed due to unavailability of vmlinux

rodgarrison, 2021-12-17:
This is used to work on Ubuntu Server 20.04. I've tested the most current x86-64-bit Ubuntu Server 20.04 LTS (ami-0b49a4a6e8e22fa16) on both a t2.mico instance and c5n.metal instance:

Steps to reproduce:

* git clone https://github.com/amzn/amzn-drivers
* cd amzn-drivers/userspace/dpdk/enav2-vfio-patch
* sudo apt-get update
* edit `/etc/apt/sources.list` and uncomment `deb-src http://us-east-1.ec2.archive.ubuntu.com/ubuntu/ focal main restricted` and `deb-src http://us-east-1.ec2.archive.ubuntu.com/ubuntu/ focal-updates main restricted` sources
* sudo ./get-vfio-with-wc.sh

```
.
 .
  .
Get:1 http://us-east-1.ec2.archive.ubuntu.com/ubuntu focal-updates/main linux-aws-5.11 5.11.0-1022.23~20.04.1 (dsc) [4646 B]
Get:2 http://us-east-1.ec2.archive.ubuntu.com/ubuntu focal-updates/main linux-aws-5.11 5.11.0-1022.23~20.04.1 (tar) [186 MB]
Get:3 http://us-east-1.ec2.archive.ubuntu.com/ubuntu focal-updates/main linux-aws-5.11 5.11.0-1022.23~20.04.1 (diff) [10.7 MB]
.
 .
  .
make -C /lib/modules/5.11.0-1022-aws/build M=/home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio modules
make[1]: Entering directory '/usr/src/linux-headers-5.11.0-1022-aws'
  CC [M]  /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/mdev/mdev_core.o
  CC [M]  /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/mdev/mdev_sysfs.o
  CC [M]  /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/mdev/mdev_driver.o
  LD [M]  /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/mdev/mdev.o
  CC [M]  /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/mdev/vfio_mdev.o
  MODPOST /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/Module.symvers
  CC [M]  /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/mdev/mdev.mod.o
  LD [M]  /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/mdev/mdev.ko
  BTF [M] /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/mdev/mdev.ko
Skipping BTF generation for /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/mdev/mdev.ko due to unavailability of vmlinux
  CC [M]  /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/mdev/vfio_mdev.mod.o
  LD [M]  /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/mdev/vfio_mdev.ko
  BTF [M] /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/mdev/vfio_mdev.ko
Skipping BTF generation for /home/ubuntu/amzn-drivers/userspace/dpdk/enav2-vfio-patch/tmp/linux-aws-5.11-5.11.0/drivers/vfio/mdev/vfio_mdev.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/linux-headers-5.11.0-1022-aws'
Done

[3] Install module
mv: cannot stat '/usr/lib/modules/5.11.0-1022-aws/kernel/drivers/vfio/pci/vfio-pci.ko': No such file or directory
```

shaibran, 2021-12-21:
@rodgarrison  Thanks for the input, we'll review this

shaibran, 2021-12-23:
@rodgarrison  Hi, the problem you see is with the latest Ubuntu AMI, which apparently built the vfio and vfio-pci as part of the kernel, not as kernel modules. We will try get reasoning from AMI team.

Meanwhile, there are two main options:

1. if you prefer to use the latest version that comes with the AMI, unzip [buildscript.zip](https://github.com/amzn/amzn-drivers/files/7769634/buildscript.zip)
and execute the script as root in order to rebuild the kernel with VFIO drivers as modules.

2. alternatively,  you can downgrade to older kernel:

- revert the kernel to 5.4.0:
```
sudo apt install -y linux-image-5.4.0-1060-aws linux-headers-5.4.0-1060-aws linux-tools-5.4.0-1060-aws
```

- b. update grub (for the fresh instance the GRUB_DEFAULT value must be changed in /etc/default/grub to `"1>2"):
```
$ grep GRUB_DEFAULT /etc/default/grub
GRUB_DEFAULT="1>2"
```
c. update the grub and reboot:
```
sudo update-grub
sudo reboot
```

rodgarrison, 2022-02-01:
Apologies for late, late reply. I abandoned Ubuntu for Amazon's Linux:
```
$ cat /proc/version 
Linux version 5.10.75-79.358.amzn2.x86_64 (mockbuild@ip-10-0-40-76) (gcc10-gcc (GCC) 10.3.1 20210422 (Red Hat 10.3.1-1), GNU ld version 2.35-21.amzn2.0.1) #1 SMP Thu Nov 4 21:08:30 UTC 2021
```

So I did return to this issue.

techthumb, 2022-03-25:
@shaibran Did you get an update from the AMI team?

shaibran, 2022-05-30:
issue is since vfio and vfio-pci are now being built as part of the kernel, not as kernel modules. instruction to workaround it are already updated in the [readme](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#622-vfio-pci).
we have a task to push these changes to the kernel though.

closed 2022-05-30 by amitbern-aws