amzn/amzn-drivers #199: DPDK 19.05 patches folder contains an duplicate/redundant patch

talawahtech, 2021-12-22:
The folder of patches for DPDK 19.05 seems to contain a duplicate/redundant patch: `0004-net-ena-fix-L4-checksum-Tx-offload.patch`. This prevents `git am ../amzn-drivers/userspace/dpdk/19.05/*.patch` from applying the patches cleanly.

shaibran, 2021-12-23:
@talawahtech  Thank you for notifying us, let us review this

shaibran, 2021-12-29:
@talawahtech thank you. The file 0004-net-ena-fix-L4-checksum-Tx-offload.patch is indeed redundant and should be removed, for the moment, removing it should resolve the problem.
We noticed same issue also in:
userspace/dpdk/18.08/0019-net-ena-fix-L4-checksum-Tx-offload.patch
userspace/dpdk/18.11/0016-net-ena-fix-L4-checksum-Tx-offload.patch
userspace/dpdk/19.02/0006-net-ena-fix-L4-checksum-Tx-offload.patch
userspace/dpdk/19.05/0004-net-ena-fix-L4-checksum-Tx-offload.patch
we'll address all the above.

closed 2022-01-16 by akiyano