amzn/amzn-drivers #200: ENA VFIO patch fails with Linux kernel 5.15

talawahtech, 2021-12-30:
As of kernel 5.15 the VFIO-PCI source code has been restructured, and most of the code that was previously in `drivers/vfio/pci/vfio_pci.c` has been moved to `drivers/vfio/pci/vfio_pci_core.c` as such the patch used by get-vfio-with-wc.sh will fail.

The section of code to be patched hasn't changed, so creating a new 5.15 specific patch should be straightforward.

shaibran, 2022-01-02:
@talawahtech thanks, we have a fix ready and will publish it after regression tests

grahamc, 2022-01-04:
Hi @shaibran I don't suppose you have an update on the regression testing? We're tracking this issue for rolling out 5.15 to NixOS users.

shaibran, 2022-01-13:
@grahamc Hey, tests were completed; please see  https://github.com/amzn/amzn-drivers/pull/204

closed 2022-01-16 by akiyano