amzn/amzn-drivers #202: How do I verify that the ENA VFIO patch is working

talawahtech, 2021-12-30:
I am testing DPDK on AWS using the Seastar framework with the built-in HTTPD application. I have gotten it running and I applied the ENA specific DPDK patches as well as the ENA VFIO patch (which required [some modification](https://github.com/amzn/amzn-drivers/issues/200)), however I am not seeing any difference in performance or behavior with/without the ENA VFIO patch.

I confirmed that the modified version of the VFIO module is in fact being loaded based on the dmesg entries noting that loading an out-of-tree module taints the kernel, but I am wondering if there is something else I need to do to see the benefits of this patch or at least verify that it is working as intended.

Instance: c5n.xlarge
OS: Fedora 34
Kernel: 5.15.11-100.fc34
DPDK Version: 19.05
Seastar Version: GitHub master

shaibran, 2022-01-02:
@talawahtech Hey, the easiest way is to check localize address of the prefetchable BAR of the used device and then check if it has WC attribute.

1. get the BAR memory; Below you can see that Memory at fe900000 is prefetchable
```
# lspci -v -s 00:06.0
00:06.0 Ethernet controller: Amazon.com, Inc. Elastic Network Adapter (ENA)
	Subsystem: Amazon.com, Inc. Elastic Network Adapter (ENA)
	Physical Slot: 6
	Flags: fast devsel, IOMMU group 0
	Memory at febf8000 (32-bit, non-prefetchable) [size=16K]
	Memory at fe900000 (32-bit, prefetchable) [size=1M]
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [b0] MSI-X: Enable- Count=9 Masked-
	Kernel driver in use: vfio-pci
	Kernel modules: ena
```
2. start DPDK application which uses the 00:06.0 device. It should allocate and map needed resources. 
3. After the application has been started, check in another terminal what is the attribute of this memory (should be write-combining):
```
# cat /sys/kernel/debug/x86/pat_memtype_list | grep fe900000
PAT: [mem 0x00000000fe800000-0x00000000fe900000] write-combining
PAT: [mem 0x00000000fe900000-0x00000000fea00000] write-combining
```

talawahtech, 2022-01-03:
Thanks @shaibran, here is the output that I get when Seastar HTTPD is running:
```
$ sudo lspci -v -s 00:06.0
00:06.0 Ethernet controller: Amazon.com, Inc. Elastic Network Adapter (ENA)
        Subsystem: Amazon.com, Inc. Elastic Network Adapter (ENA)
        Physical Slot: 6
        Flags: bus master, fast devsel, latency 0, IOMMU group 0
        Memory at febf8000 (32-bit, non-prefetchable) [size=16K]
        Memory at fe900000 (32-bit, prefetchable) [size=1M]
        Capabilities: [70] Express Endpoint, MSI 00
        Capabilities: [b0] MSI-X: Enable+ Count=9 Masked-
        Kernel driver in use: vfio-pci
        Kernel modules: ena
```

```
$ sudo cat /sys/kernel/debug/x86/pat_memtype_list | grep fe900000
PAT: [mem 0x00000000fe800000-0x00000000fe900000] write-combining
PAT: [mem 0x00000000fe900000-0x00000000fea00000] uncached-minus
PAT: [mem 0x00000000fe900000-0x00000000fea00000] uncached-minus
```
The output is the same whether I use the modified kernel module or the original one, so I guess it is safe to say it isn't working?

Here are the commands I used  to load the kernel module, bind the device, and run Seastar HTTPD.

```
sudo modprobe vfio enable_unsafe_noiommu_mode=1 
sudo modprobe vfio-pci

sudo ifconfig eth1 down
sudo seastar/dpdk/usertools/dpdk-devbind.py -b vfio-pci 00:06.0

sudo seastar/build/release/apps/httpd/httpd --port 8080 --network-stack native --dpdk-pmd
```

talawahtech, 2022-01-06:
@shaibran @semihalf-krawczyk-michal I confirm that the change in PR https://github.com/amzn/amzn-drivers/pull/204 to also install vfio-pci-core.ko resolves this issue for me:

```
sudo cat /sys/kernel/debug/x86/pat_memtype_list | grep fe900000
PAT: [mem 0x00000000fe800000-0x00000000fe900000] write-combining
PAT: [mem 0x00000000fe900000-0x00000000fea00000] write-combining
PAT: [mem 0x00000000fe900000-0x00000000fea00000] write-combining
```
I also saw a 40% performance increase for my Seastar HTTPD workload! I will update the issue subject to reflect the actual cause.

Are there any plans to get this change into the upstream kernel?

shaibran, 2022-01-09:
@talawahtech no plan to up stream

closed 2022-01-09 by akiyano

talawahtech, 2022-01-10:
> @talawahtech no plan to up stream

@shaibran any specific reason why not? Does this patch have a negative performance impact on non-DPDK workloads? Any chance it will be included in [Amazon Linux 2022 ](https://github.com/amazonlinux/amazon-linux-2022) by default?