amzn/amzn-drivers #206: Feature request: add support for n-tuple filtering

aneagoe, 2022-01-20:
Hello,

We're looking for an easier way to use kernel bypass together with the ENA driver. For that, openonload would be a perfect fit. However, at the moment, it doesn't work due to ENA's lack of support for n-tuple filtering.
How feasible would it be to add such a feature?
For more background, please see https://github.com/Xilinx-CNS/onload/issues/28.

akiyano, 2022-01-20:
Hi @aneagoe,
Thanks for your request.
n-tuple filtering is in our long term plans, but there isn't a concrete timeline yet.

nirvana-msu, 2023-01-19:
There are other use-cases for n-tuple filtering support beside *AF_XDP*/kernel bypass with *OpenOnload*.

[Accelerated RFS](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/9/html/monitoring_and_managing_system_status_and_performance/configuring-an-operating-system-to-optimize-access-to-network-resources_monitoring-and-managing-system-status-and-performance#accelerated-rfs_configuring-an-operating-system-to-optimize-access-to-network-resources) also requires it. This is what can let you achieve the best latency while still using kernel for handling TCP stack. It does it by routing packet directly to the correct NIC queue  / correct CPU your application needs this packet on, rather than assigning queue randomly based on hash (which normally would mean an extra IPI call, even if you use regular RFS).

Btw, another requirement for *aRFS* is that NIC must support `ndo_rx_flow_steer()` `net_device` function. Is it documented anywhere for AWS EC2 ENA (or is there any way to check if from the running instance)? Specifically interested in this for `c6g` instances.

@davidarinzon / @akiyano I'm wondering if n-tuple filtering support could be prioritized for ENA driver?

Thanks.

davidarinzon, 2023-01-22:
Hi @nirvana-msu 
Thank you for sharing this information and the additional use-cases.
You're welcome to reach me out at darinzon@amazon.com for further discussion.

Regarding `ndo_rx_flow_steer()`, this option is not implemented by the driver at this point.

ShakesB33r, 2023-04-04:
+1 to being able to benefit from both aRFS as well as openonload, would be nice to see the enhancements mentioned in this thread implemented.

FrothyB, 2023-11-03:
Are there any plans to implement these features or should we assume that they will not be landing anytime soon?

davidarinzon, 2023-11-14:
Hi @FrothyB 

We are looking into the option of adding n-tuple support, but I can't provide a timeframe at this point.
Stay tuned for ENA Linux driver releases and release notes.

davidarinzon, 2024-09-12:
Hi,

We've added AF_XDP support as well as flow steering support in the latest release (2.13.0g), you're welcome to check them and see if it unblocks our efforts.

davidarinzon, 2024-09-20:
Hi @aneagoe 

It is mentioned that the feature is supported from 7th generation instance types in the documentation (https://github.com/amzn/amzn-drivers/tree/master/kernel/linux/ena#flow-steering-ntuple). This translates to Nitro v5 (https://docs.aws.amazon.com/ec2/latest/instancetypes/ec2-nitro-instances.html).

Can you please try it on c7gn or r8g and see if it works for you?

We will improve the documentation to better correlate the instance type and nitro families.

aneagoe, 2024-09-23:
@davidarinzon thanks, now I understand. I will check on an hpc7g instance in that case.
A side question, is there a possibility that c7a or c7g instances will be migrated to Nitro v5 or that this feature would trickle down to Nitro v4? Otherwise, it feels a bit restrictive atm...

davidarinzon, 2024-09-26:
Hi @aneagoe 

The feature is available from Nitro v5 and on.

You may also use the recently made available c8g and m8g (https://aws.amazon.com/blogs/aws/run-your-compute-intensive-and-general-purpose-workloads-sustainably-with-the-new-amazon-ec2-c8g-m8g-instances/).

davidarinzon, 2024-10-15:
Resolving this ticket, please feel free to re-open it in case there are questions or issues.

closed 2024-10-15 by davidarinzon

daniel-trnk, 2024-11-16:
My apologies if this is based on a misunderstanding... I am trying to use accelerated RFS on a c7gn.metal instance, but I don't seem to be able to set up ntuple-filters on multiple different interfaces (eg. eth11 and eth12) at the same time?

```
[root@prod]# ethtool -K eth11 ntuple on
[root@prod]# ethtool -K eth12 ntuple on
[root@prod]# ethtool -U eth11 flow-type tcp4 dst-ip 10.10.10.87 dst-port 11111 action 2
Added rule with ID 63
[root@prod]# ethtool -U eth12 flow-type tcp4 dst-ip 10.132.102.137 dst-port 11112 action 3
rmgr: Cannot insert RX class rule: Invalid argument
Cannot insert classification rule
```

I am able to add multiple filters on the same interface.

Is this expected?

davidarinzon, 2024-11-17:
Hi @daniel-trnk 

Thanks for reaching out.
While we're looking into this, can you please share your instance ID and a timeline?

David

daniel-trnk, 2024-11-17:
@davidarinzon the instance ID is `i-04998996b6e7e937f` (`ap-northeast-1a`) although we do only spin it up when we are working on the code / testing.
Re the timeline: I changed the instance type to `c7gn` (from `c7g`) on Nov 16 after reading through this issue and then tried to set it ups as described above in different combinations.

davidarinzon, 2024-11-17:
Hi @daniel-trnk 

Thank you for sharing the information.
Quoting from the documentation about the feature at https://github.com/amzn/amzn-drivers/tree/master/kernel/linux/ena#flow-steering-ntuple, "The table is shared across all interfaces on the host, meaning that an entry used by one interface cannot be used by another interface until the rule is removed."
As the table is shared, two rules can't have the same location/ID.
In case such is not specified using `loc <number>`, a default value is added, in this case `63`. Your first command succeeds, and you can see the `63` as a returned ID. The second command fails, and it fails as the same location/ID (`63`) is sent.
We recommend using the `loc` option.

daniel-trnk, 2024-11-17:
@davidarinzon thank you very much for the clarification. I indeed did not understand the significance of those sentences. If I explicitly set `loc XX`, it works. What added to my confusion is probably that when you add multiple rules on the same interface, it automatically sets a new `loc value`. My apologies for not realizing that.

This brings me to another question I should have realized from the beginning: this implies, that the max number of ntuple rules is the number of cores? I would have expected it to be whatever `/proc/sys/net/core/rps_sock_flow_entries` is set to?

Again, I apologize if I am misunderstanding how this works or if the feature even does what I think it does: My use case is that we have processes pinned to specific Cores which are subscribing to a potentially large number of websockets or interacting with Rest endpoints across multiple NICs, sharing NICs at times. I can distinguish them by client IP / Port and was hoping to lower latency and ensure better utilization of ENA NICs and their  limits by using accelerated RFS - but that would require more than one rule per Core...

ShayAgros, 2024-11-18:
> Is the limit to the number of cores a hardware or core design issue?

This is not a property that can be changed for the current nitro generation, but *might* change in future generations. It depends very much on understanding the use-cases of our customers.

Regarding the use-case you described. For the current diagram
```
                                                        
 ┌────────────────────────────────────────────────────┐ 
 │  instance                                          │ 
 │                                                    │ 
 │                                                    │ 
 │                                                    │ 
 │    ┌─────────────┐         ┌─────────────┐         │ 
 │    │eth11        │         │eth12        │         │ 
 │    │             │         │             │         │ 
 │    │q0  q1  q2   │         │q0  q1  q2   │         │ 
 │    │┌─┐ ┌─┐ ┌─┐  │         │┌─┐ ┌─┐ ┌─┐  │         │ 
 │    ││ │ │ │ │ │  │         ││ │ │ │ │ │  │         │ 
 │    │└─┘ └─┘ └─┘  │         │└─┘ └─┘ └─┘  │         │ 
 │    └─────────────┘         └─────────────┘         │ 
 │                                                    │ 
 └────────────────────────────────────────────────────┘ 
```

And an application / websocket bound to CPU0.

Are you trying to bring all traffic from a specific address to (for example) queue 0 on `eth11` and `eth12` so that you can map the queues themselves to the same core and have a better cache locality for the same traffic ?

daniel-trnk, 2024-11-18:
Understood. Thanks for the explanation.

Yes. However, because there may be multiple processes on different cores connecting to the same remote address / port, I have to focus on the local client address and port. So In your example, I would like to bring all traffic coming in on eth11 to a specific local IP and port (what my websocket client is binding to) a specific queue - say q0 - that is mapped to the core - e.g. 0 - said client is running on.  I may have a similar setup for q0 to core 0 for an address/port combination on eth12 as the same client will try to connect to a websocket from that address. 
A different client running on core 3 would act in the same way with different local ports but the same local client addresses and I would try to use q1 for that.

Again, this idea is based on a very superficial understanding of how these queues work - and I apologize if I misunderstand the underlying concepts.

More abstract: Our use case is probably unusual - or at least outside of what most reference architectures focus on - in the sense that we are client-centric:
* We have a pool of source IPs distributed across all 15 NICs of the instance.
* We have a number of different processes with multiple threads pinned to specific Cores.
* Each process has a I/O-Thread on a specific core.
* These I/O-Threads each subscribe to a number of different Websocket streams and interact with a number of Rest endpoints.
* Each of these server endpoints also has a pool of IPs and we do multiple connections from each I/O thread to each Endpoint using different Source IP:Source Port -> Target IPs combinations from the pools of Source / Target IPs respectively and then de-dupe the data (this is done for reliability and latency).

Our goal is to get the data to the I/O core with the lowest possible latency and to use caches and any limits as efficiently as possible.

ShayAgros, 2024-11-20:
Well `io_uring` is a new(ish) mechanism for asynchronous operations. In the setup you described it would probably make managing multiple IO operations faster. It doesn't include/require any special driver support, but rather provides the same functionality you're using now with a lot less user-kernel space context switches (which should improve CPU usage and latency).

When it comes to latency, you probably won't beat the performance of DPDK (or Linux AF XDP), however using these tools comes with a significant effort:
1. DPDK takes the specific network device (e.g. eth11) out of the kernel's control. This means that you can no longer rely on the kernel for load balancing, ARP or TCP stack. A DPDK application would need to poll almost directly the driver's IO queues and manage the distribution of traffic between the different threads.
2. AF XDP would require you to part from the network stack goodies (e.g. no automatic load balancing by the kernel) but retain the userspace infrastructure (e.g. you can still poll a unix socket). AF XDP is the native Linux solution to what DPDK tries to achieve and also requires you to almost directly manage the driver's queues.

Both solutions would require you to implement the network stack, and would *probably* provide better performance but that's mostly because you end up creating a tailored experience with a significant effort.
Also the DPDK driver currently **doesn't support ntuple filtering** (might have the support added to it in the future).

Since the device currently doesn't support more steering rules, the next thing to try would probably be to steer the traffic in the network stack itself. You could reuse the same source port for all traffic destined to a specific thread (e.g. using SO_REUSEPORT and fine-grain it using [BPF-Powered SO_REUSEPORT](https://medium.com/all-things-ebpf/ebpf-powered-load-balancing-for-so-reuseport-30acb395e1d6)).

daniel-trnk, 2024-11-20:
Thank you for the advice. So in summary: for either case (EPoll w packet steering / DPDK / io_uring / AF XDP) we can't rely on any HW support for steering packets to the right queue, correct?

oliora, 2025-07-31:
Are there any plans to add this feature for Nitro v4 or introduce Nitro v5 instances with Intel CPU?

akiyano, 2025-09-18:
@oliora Nitro V5 X86 instances are now available. For example:
https://aws.amazon.com/ec2/instance-types/m8i/
https://aws.amazon.com/ec2/instance-types/r8i/