amzn/amzn-drivers #207: How to prepare a packet for transmission with DPDK

rodgarrison, 2022-02-01:
DPDK can and cannot be faulted for its totally inadequate documentation of how to prepare a DPDK mbuf to TX over a DPDK TXQ. If one does not know already, don't go looking to DPDK to tell you. There are too many NICs with too much variation to step into that game. On the other hand the documentation also fails to point programmers to a direction to resolve this question even though this has got to be FAQ#2 after building.

Now in the case of AWS ENA NICs one might expect the DPDK doc to point to an Amazon PDF/HTTP URI which would answer this question. Is there such an animal?

I'll place this into context better:

* Programmers use this repo to among other things program DPDK AWS ENA NICs running over VFIO. We need good drivers.
* Programmers will create DPDK memory pool from which they get mbufs to hold packet data
* The mbuf has memory section --- a blob of memory of finite size --- into which the programmer needs to put any ethernet, IPv4 headers, and the payload. 
* Knowing what to put, in what, order must be supplied by vendors since its NIC H/W specific or least to a good approximation
* Here my vendor is AWS for its ENA NICs; I happen be working on `c5n` bare metal instances for what's worth

As best I can tell --- but I haven't quite gotten to running this code yet --- the memory layout is:

```
+-------------------------------------+ start of data room in mbuf 
|+-----------------------------------+| cache aligned address start of ethernet headers (packed memory)
|| uint8_t dst_mac_[6];              ||
|| uint8_t src_mac_[6];              ||
|| uint16_t eth_type_;               ||
|+-----------------------------------+| cache aligned address start of ipv4 headers (packed memory)
|| uint8_t ihl_ : 4;                 ||
|| uint8_t version_ : 4;             ||
|| uint8_t ecn_ : 2;                 ||
|| uint8_t dscp_ : 6;                ||
|| uint16_t tot_len_;                ||
|| uint16_t id_;                     ||
|| uint16_t frag_off_;               ||
|| uint8_t ttl_;                     ||
|| uint8_t protocol_;                ||
|| uint16_t check_;                  ||
|| uint32_t src_ip_;                 ||
|| uint32_t dst_ip_;                 ||
|+-----------------------------------+| cached aligned address of payload
|| data payload                      ||
|+-----------------------------------+|
+-------------------------------------+
```
Is there a smart, better, official way to confirm this hopefully with field descriptions?

rodgarrison, 2022-02-03:
To help the community and toward improving DPDK-ENA integration I will provide a GIT url which among other things,

* Gives very clear build instructions of this, DPDK
* Gives very clear one time set up instructions of TLB 
* Provides a minimal-complete UDP client/server app on AWS-ENA NIC through DPDK
* Provides clear build instructions for former
* Provides the config for prev bullet clearly (hint 12-factor app ENV key/value pairs)
* Provides extended write up of the theory of mbuf packet preparation

The last point would be stronger if it was rejoined by an AWS publication/URI to confirm

zorikm, 2022-02-03:
Rod, thank you so much for these insightful suggestions.
Please let us discuss and get back to you with our comments and conclusions.

rodgarrison, 2022-02-08:
@zorikm No problem ... the least I can do is cary some water here. 

I want to mention in passing what perhaps is a the logical end of DPDK user space work ... something that's well beyond my expertise but may right up Amazon's engineering know-how to shoot-down OR contemplate elsewhere for the future.

If we're gonna go all user space NIC work ... how about we parlay this into any of the reduced-attack-service-uni-kernels and show up with something like:

https://news.ycombinator.com/item?id=30247022

plus DPDK user space apps ... it's got a as much kernel NIC stuff removed as makes sense. 

The DPDK library I'm working on is but 1/3rd of a larger distributed app. That app happens to align well with AWS graviton instances. A larger number of cores on cheaper HW makes it easier to have scalable user space NIC on a core that's colocated (as possible) with another core thread to which the DPDK lcore delegates. 

Packing the whole shebang into a smaller kernel ... maybe is a silly ask ... maybe makes sense eventually ... not sure.

rodgarrison, 2022-03-07:
Apologies for long delay. Promised link:

https://github.com/rodgarrison/reinvent

shaibran, 2023-08-10:
ENA currently implements only the generic DPDK PMD API, so any application which uses the API is not device specific but common to all devices.
Application writing is out of ENA team scope and knowledge. However, we do intend to enhance our documentation with detailed build instruction for all LTS and improve DPDK-ENA integration and ease of use so we will consider referencing your github from our best practices section or from a new section in the readme where we can reference application examples.
Thank you!

closed 2023-08-10 by shaibran