amzn/amzn-drivers #210: Compile error with latest UEK kernel 4.1.12 - OEL6U10

TactfulElf, 2022-02-19:
Compile Error with latest UEK kernel 4.1.12 on OEL6010

make -C /lib/modules/4.1.12-124.48.6.el6uek.x86_64/build M=/usr/src/amzn-drivers-2.6.1/kernel/linux/ena modules
make[1]: Entering directory `/usr/src/kernels/4.1.12-124.48.6.el6uek.x86_64'
  CC [M]  /usr/src/amzn-drivers-2.6.1/kernel/linux/ena/ena_netdev.o
In file included from /usr/src/amzn-drivers-2.6.1/kernel/linux/ena/ena_netdev.h:9,
                 from /usr/src/amzn-drivers-2.6.1/kernel/linux/ena/ena_netdev.c:25:
/usr/src/amzn-drivers-2.6.1/kernel/linux/ena/kcompat.h:751: error: redefinition of 'page_ref_count'
compilation terminated due to -Wfatal-errors.
make[2]: *** [/usr/src/amzn-drivers-2.6.1/kernel/linux/ena/ena_netdev.o] Error 1
make[1]: *** [_module_/usr/src/amzn-drivers-2.6.1/kernel/linux/ena] Error 2
make[1]: Leaving directory `/usr/src/kernels/4.1.12-124.48.6.el6uek.x86_64'
make: *** [all] Error 2

The initial UEK kernel that comes with OEL6U10 has package kernel-uek-4.1.12-124.16.4.el6uek.x86_64 which doesn't have this issue.

The latest UEK kernel package kernel-uek-4.1.12-124.48.6.el6uek.x86_64 in the repo does have this issue

It appears Oracle backported page_ref_count from kernel-uek-4.1.12-124.43.1el6uek.x86_64 inclusive.

The patch below should fix >=  kernel-uek-4.1.12-124.43.1.el6uek.x86_64 packages.  It will however not compile on the initial UEK kernel package supplied with OEL6U10 (kernel-uek-4.1.12-124.16.4.el6uek.x86_64), there doesn't appear to be a way to determine with more granularity the patches applied to UEK 4.1.12 

[ena261-oel6u10_page_ref_count-backpropagation.txt](https://github.com/amzn/amzn-drivers/files/8102311/ena261-oel6u10_page_ref_count-backpropagation.txt)

davidarinzon, 2022-02-20:
Thank you @TactfulElf for raising this issue, we will look into this

akiyano, 2023-05-18:
@TactfulElf, 
This issue was resolved in the latest release of the driver - https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.8.6

closed 2023-05-18 by akiyano