amzn/amzn-drivers #211: Parallel computation  problems with EFA

chenshixinnb, 2022-02-24:
#!/bin/bash
module add GROMACS/2021-foss-2020b
module rm OpenMPI/4.0.5-GCC-10.2.0
module rm UCX/1.9.0-GCCcore-10.2.0
module rm libfabric/1.11.0-GCCcore-10.2.0
export PATH=/opt/amazon/openmpi/bin:$PATH
export LD_LIBRARY_PATH=/opt/amazon/openmpi/lib64:$LD_LIBRARY_PATH
export LIBRARY_PATH=/opt/amazon/openmpi/lib64:$LIBRARY_PATH
export CPATH=/opt/amazon/openmpi/include:$CPATH

export PATH=/opt/amazon/efa/bin:$PATH
export LD_LIBRARY_PATH=/opt/amazon/efa/lib64:$LD_LIBRARY_PATH
export LIBRARY_PATH=/opt/amazon/efa/lib64:$LIBRARY_PATH
export CPATH=/opt/amazon/efa/include:$CPATH

mpirun -v gmx_mpi mdrun -v -cpi inputfile -deffnm inputfile --mca pml cm --mca mtl ofi --mca pml_base_verbose 10 --mca mtl_base_verbose 100 -x FI_LOG_LEVEL=warn

chenshixinnb, 2022-02-24:
slurm.out：
[c-96-4-worker0001:31403] mca: base: components_open: component cm open function successful
[c-96-4-worker0001:31544] mca: base: components_register: registering framework mtl components
[c-96-4-worker0001:31544] mca: base: components_register: found loaded component ofi
[c-96-4-worker0001:31544] mca: base: components_register: component ofi register function successful
[c-96-4-worker0001:31544] mca: base: components_open: opening mtl components
[c-96-4-worker0001:31544] mca: base: components_open: found loaded component ofi
[c-96-4-worker0001:31544] mca: base: components_open: component ofi open function successful
[c-96-4-worker0001:31544] mca: base: components_open: component cm open function successful
[c-96-4-worker0001:31414] mca: base: components_register: registering framework mtl components
[c-96-4-worker0001:31414] mca: base: components_register: found loaded component ofi
[c-96-4-worker0001:31414] mca: base: components_register: component ofi register function successful

[c-96-4-worker0001:31472] mca:base:select: Auto-selecting mtl components
[c-96-4-worker0001:31472] mca:base:select:(  mtl) Querying component [ofi]
[c-96-4-worker0001:31472] mca:base:select:(  mtl) Query of component [ofi] set priority to 25
[c-96-4-worker0001:31472] mca:base:select:(  mtl) Selected component [ofi]
[c-96-4-worker0001:31472] select: initializing mtl component ofi
[c-96-4-worker0001:31388] select: initializing pml component cm
[c-96-4-worker0001:31388] mca:base:select: Auto-selecting mtl components
[c-96-4-worker0001:31388] mca:base:select:(  mtl) Querying component [ofi]
[c-96-4-worker0001:31388] mca:base:select:(  mtl) Query of component [ofi] set priority to 25
[c-96-4-worker0001:31388] mca:base:select:(  mtl) Selected component [ofi]

[c-96-4-worker0001:31430] mtl_ofi_component.c:366: mtl:ofi:provider: rdmap0s6-rdm
[c-96-4-worker0001:31432] mtl_ofi_component.c:366: mtl:ofi:provider: rdmap0s6-rdm
[c-96-4-worker0001:31433] mtl_ofi_component.c:366: mtl:ofi:provider: rdmap0s6-rdm
[c-96-4-worker0001:31424] mtl_ofi_component.c:366: mtl:ofi:provider: rdmap0s6-rdm
[c-96-4-worker0001:31414] mtl_ofi_component.c:366: mtl:ofi:provider: rdmap0s6-rdm

[c-96-4-worker0002:29112] check:select: checking my pml cm against process [[47917,1],0] pml cm
[c-96-4-worker0002:29117] check:select: checking my pml cm against process [[47917,1],0] pml cm
[c-96-4-worker0002:29085] check:select: checking my pml cm against process [[47917,1],0] pml cm
[c-96-4-worker0002:29052] check:select: checking my pml cm against process [[47917,1],0] pml cm
[c-96-4-worker0002:29032] check:select: checking my pml cm against process [[47917,1],0] pml cm

[c-96-4-worker0001:31492] *** An error occurred in MPI_Bcast
[c-96-4-worker0001:31492] *** reported by process [3140288513,140724603453511]
[c-96-4-worker0001:31492] *** on communicator MPI_COMM_WORLD
[c-96-4-worker0001:31492] *** MPI_ERR_OTHER: known error not in list
[c-96-4-worker0001:31492] *** MPI_ERRORS_ARE_FATAL (processes in this communicator will now abort,
[c-96-4-worker0001:31492] ***    and potentially your MPI job)
[c-96-4-worker0001:31516] *** An error occurred in MPI_Bcast

[c-96-4-worker0001:03697] check:select: checking my pml cm against process [[52300,1],0] pml cm
[c-96-4-worker0001:03728] check:select: checking my pml cm against process [[52300,1],0] pml cm
[c-96-4-worker0001:03701] check:select: checking my pml cm against process [[52300,1],0] pml cm
[c-96-4-worker0001:03729] check:select: checking my pml cm against process [[52300,1],0] pml cm
[c-96-4-worker0001:03725] check:select: checking my pml cm against process [[52300,1],0] pml cm
[c-96-4-worker0001:03672] check:select: checking my pml cm against process [[52300,1],0] pml cm
[c-96-4-worker0001:03625] *** An error occurred in MPI_Bcast
[c-96-4-worker0001:03625] *** reported by process [3427532801,4498047342758330405]
[c-96-4-worker0001:03625] *** on communicator MPI_COMM_WORLD
[c-96-4-worker0001:03625] *** MPI_ERR_OTHER: known error not in list
[c-96-4-worker0001:03625] *** MPI_ERRORS_ARE_FATAL (processes in this communicator will now abort,
[c-96-4-worker0001:03625] ***    and potentially your MPI job)
[c-96-4-worker0001:03562] 79 more processes have sent help message help-mpi-errors.txt / mpi_errors_are_fatal
[c-96-4-worker0001:03562] Set MCA parameter "orte_base_help_aggregate" to 0 to see all help / error messages

chenshixinnb, 2022-02-24:
[cloudam@c-96-4-worker0001 ~]$ fi_info -p efa -t FI_EP_RDM
provider: efa
    fabric: EFA-fe80::4fc:a6ff:fe55:bb98
    domain: rdmap0s6-rdm
    version: 111.20
    type: FI_EP_RDM
    protocol: FI_PROTO_EFA

chenshixinnb, 2022-02-24:
> slurm.out： [c-96-4-worker0001:31403] mca: base: components_open: component cm open function successful [c-96-4-worker0001:31544] mca: base: components_register: registering framework mtl components [...]

This is the error log

wzamazon, 2022-02-24:
Hi, we will continue debugging through the open mpi issue https://github.com/open-mpi/ompi/issues/9858

closed 2022-02-27 by firasj