amzn/amzn-drivers #213: Request for documentation / information

rodgarrison, 2022-03-19:
[As per](https://github.com/rodgarrison/reinvent#benchmarks) my first-take sense of AWS ENA NICs under DPDK is terrible. It underperforms `ncat` by an order of 10 almost two in other cases. Now, as I write there towards the bottom, this is probably a reflection of bad application code likely resulting from using default threshold values or too-small ring sizes. And I will have to revisit setting the link speed.

But in an effort to avoid re-tilling ground can you provide guidance on any of the following:

* [A better description of the threshold fields here](https://doc.dpdk.org/api/structrte__eth__thresh.html). The DPDK one liner in the code doesn't help me much.
* [Can you recommend a setting for tx_rs_thresh here](https://doc.dpdk.org/api/structrte__eth__txconf.html#ab7d7dbc33adbb1740718c9a841555dce) 

Elsewhere, for example during the EC2 launch screen for `c5n.metal` instances, the web pages says `c5n.metal` instances are capable of 100Gbp with EFA enabled: 

```
This instance type supports the Elastic Fabric Adapter (EFA). EFAs support OS-bypass functionality, which enables High
Performance Computing (HPC) and Machine Learning (ML) applications to communicate directly with the network interface
hardware. EFAs provide lower latency and higher throughput than traditional TCP channels.
```

Was that performance achieved on this driver? Put another way, is there any hope to get to 100Gbs using DPDK? I'm guessing 100Gbp is the rounded value that aggregates:

* all 32 TXQs and all RXQs are in use
* and, as a result, the TXQ (or RXQ) performance for 1 queue will max out at a correspondingly smaller value i.e. 1/32nd or 1/64th of the 100Gbp aggregate value.

rodgarrison, 2022-03-19:
Comparison of DPDK to ncat which uses 8192 bytes per packet and therefore sends fewer packets and therefore less work is a bad comparison. I am closing this request. I will reopen it with the original questions either with better stats or omit the stats as out of scope.

shaibran, 2022-03-21:
Let us review your request and we'll get back to you shortly

shaibran, 2022-03-22:
**Regarding documentation:**

- ENA uses only the [standard DPDK features](https://doc.dpdk.org/guides/nics/overview.html). See DPDK features [documentation ](https://doc.dpdk.org/guides/nics/features.html). If the API documentation is missing, it's very unlikely that ENA will make use of it.  This is DPDK API and its documentation is out of ENA scope. We suggest to request a better documentation of this API directly from the DPDK community.
- We do document all non-standard behaviors for the ENA driver and we're constantly working on the [ENA DPDK README](https://github.com/amzn/amzn-drivers/blob/master/userspace/dpdk/README.md) to provide best practices, troubleshooting and FAQs. 
- We'll review the datasheet reference you provided and see what equivalent ENA data we can expose in order to improve app developers user experience.

**Regarding threshold recommendation:**
We're not using any of the threshold registers in our PMD. 

**Regarding line speed:**
ENA is a virtual device, so there is no link. Thus the default link speed for the ENA is RTE_ETH_SPEED_NUM_NONE.
The maximum performance is limited by the VF's capabilities and the instance type. As the driver is not aware on which instance type it is running, using appropriate limiters for the instance should be performed on the application side, as it can get access to more variety of tools to do that (e.g. it can read the instance meta data or pass the maximum BW as an argument).

**Regarding performance results:**
You can find the[ sample results](https://github.com/amzn/amzn-ec2-ena-utilities/blob/main/ena-dts/RESULTS.md) for the c5n.18xlarge. It was achieved by using the benchmarking tool based on the pktgen and the DTS.

rodgarrison, 2022-03-28:
Reopening this item with one further request for clarification. As per 

https://www.amazon.com/Data-Plane-Development-Kit-DPDK/dp/0367373955/ref=sr_1_1?keywords=data+plane+development+kit&qid=1648425589&s=books&sprefix=data+plane+devel%2Cstripbooks%2C85&sr=1-1

virtio devices (see chapter 12 pg 236) are partially or wholly interrupt driven (INTx/MSI-X).

[And as per](https://github.com/rodgarrison/reinvent#benchmarks) the number one performance constraint on the TX side is whether or not the TXQ's ring is full. As you can see code spends a lot of CPU time busy-spinning on rte_eth_tx_burst trying to flush packets out once the TXQ's ring is full.

which leads to this question:

- is the user mode virtio driver provided in this repo interrupt driven? If so how is the ENA driver using it poll driven which is what one would naively expect per DPDK doc? Not sure of details but an interrupt-rate-bouned-pathway surely has to be slower than one which is 100% poll driven

shaibran, 2022-03-29:
By default, the ENA PMD works fully in the in the polling mode. The cleanup is performed right after transmit (since v22.03 it's right before the transmit).
Regarding ```As you can see code spends a lot of CPU time busy-spinning on rte_eth_tx_burst trying to flush packets out once the TXQ's ring is full``` --> That's because application is pushing the packets faster than the hardware can handle it

closed 2022-04-02 by rodgarrison