amzn/amzn-drivers #214: Interrupt coalesce CQE mode support

talawahtech, 2022-04-06:
I noticed a [recent commit ](https://github.com/amzn/amzn-drivers/commit/7c6b128179c3e37189933b064ac4f1b64781fde2) that mentions supporting the latest ethtool API for interrupt coalescing params. Specifically to accommodate the the addition of CQE mode.

My attempt to use the feature on Fedora 34 (kernel 5.15) with ENA 2.7.0 (installed via DKMS) results in the follwing error:
```
$ sudo ethtool -C eth0 cqe-mode-rx on
netlink error: cannot modify an unsupported parameter (offset 36)
netlink error: Invalid argument
```

Taking a closer look at the code, it seems that the function api is supported, but there is no actual code change to implement support for the feature. Is this feature on the roadmap?

davidarinzon, 2022-04-06:
Hi @talawahtech,

Thank you for reaching out.
The change itself was added in order to maintain compilation compatibility with the Linux Kernel.
We'll get back to you on whether this feature is going to be supported in the future.

davidarinzon, 2022-05-22:
Hi @talawahtech 

Do you have any further inquiries regarding this topic?

talawahtech, 2022-05-23:
Hi David,

No, I don't.

closed 2022-09-13 by davidarinzon