amzn/amzn-drivers #216: Unable to activate write_combining for igb_uio Ubuntu 20.04.4 LTS 5.13.0-1021-aws

techthumb, 2022-04-12:
```

$ uname -a
Linux ip-10-21-120-92 5.13.0-1021-aws #23~20.04.2-Ubuntu SMP Thu Mar 31 11:36:15 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

$ cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.4 LTS"

$ /root/dpdk-stable/usertools/dpdk-hugepages.py --show
Node Pages Size Total
0    4096  2Mb    8Gb

Hugepages mounted on /dev/hugepages

$ cat /sys/kernel/debug/x86/pat_memtype_list
PAT memtype list:
PAT: [mem 0x00000000bffec000-0x00000000bffee000] write-back
PAT: [mem 0x00000000bffed000-0x00000000bfff0000] write-back
PAT: [mem 0x00000000bffef000-0x00000000bfff0000] write-back
PAT: [mem 0x00000000bfff0000-0x00000000bfff1000] write-back
PAT: [mem 0x00000000fe800000-0x00000000fe900000] write-combining
PAT: [mem 0x00000000febf0000-0x00000000febf2000] uncached-minus
PAT: [mem 0x00000000febf2000-0x00000000febf3000] uncached-minus
PAT: [mem 0x00000000febf4000-0x00000000febf8000] uncached-minus
PAT: [mem 0x00000000febf6000-0x00000000febf7000] uncached-minus
PAT: [mem 0x00000000fed00000-0x00000000fed01000] uncached-minus
PAT: [mem 0x00000000fed00000-0x00000000fed01000] uncached-minus

$ cat /sys/kernel/debug/x86/pat_memtype_list | grep fe900000
PAT: [mem 0x00000000fe800000-0x00000000fe900000] write-combining

$ curl -H "X-aws-ec2-metadata-token: $TOKEN" http://169.254.169.254/latest/meta-data/ami-id
ami-04505e74c0741db8d
```

```playbook.yaml
---
- name: Set Up Machine
  hosts: all
  tasks:
    - name: Update apt repo and cache
      apt: update_cache=yes force_apt_get=yes cache_valid_time=3600
    - name: Upgrade all packages on servers
      apt: upgrade=dist force_apt_get=yes
    - name: Install build-essential
      apt: pkg=build-essential state=present update_cache=yes
    - name: Check if a reboot is needed on all servers
      register: reboot_required_file
      stat: path=/var/run/reboot-required get_md5=no
    - name: Reboot the box if kernel updated
      reboot:
        msg: "Reboot initiated by Ansible for kernel updates"
        connect_timeout: 5
        reboot_timeout: 300
        pre_reboot_delay: 0
        post_reboot_delay: 30
        test_command: uptime
      when: reboot_required_file.stat.exists
    - name: DPDK Stable
      git:
        repo: git://dpdk.org/dpdk-stable
        dest: /root/dpdk-stable
        version: v21.11
    - name: DPDK Kernel Mods
      git:
        repo: git://dpdk.org/dpdk-kmods
        dest: /root/dpdk-kmods
    - name: Exclude ens6 from being managed by OS
      command: ip link set dev ens6 down
      ignore_errors: True
    - name: Set up HugePages
      command: /root/dpdk-stable/usertools/dpdk-hugepages.py --setup 8G
    - name: Add the uio module
      community.general.modprobe:
        name: uio
        state: present
    - name: make and install igb_uio.ko
      shell: |
        cd /root/dpdk-kmods/linux/igb_uio
        make
        insmod /root/dpdk-kmods/linux/igb_uio/igb_uio.ko wc_activate=1
        /root/dpdk-stable/usertools/dpdk-devbind.py --bind=igb_uio 00:06.0

```

techthumb, 2022-04-13:
Seeing this issue with Amazon Linux 2 as well.

```
$ uname -a
Linux ip-10-21-120-191.ec2.internal 5.10.106-102.504.amzn2.x86_64 #1 SMP Tue Mar 29 23:15:13 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

$ cat /etc/system-release
Amazon Linux release 2 (Karoo)

$ /root/dpdk-stable/usertools/dpdk-devbind.py --status

Network devices using kernel driver
===================================
0000:00:05.0 'Elastic Network Adapter (ENA) ec20' if=eth0 drv=ena unused= *Active*
0000:00:06.0 'Elastic Network Adapter (ENA) ec20' if=eth1 drv=ena unused= *Active*

$ /root/dpdk-stable/usertools/dpdk-hugepages.py --setup 8G

$ /root/dpdk-stable/usertools/dpdk-hugepages.py --show
Node Pages Size Total
0    4096  2Mb    8Gb

Hugepages mounted on /dev/hugepages

$ cat /sys/kernel/debug/x86/pat_memtype_list | grep fe900000
PAT: [mem 0x00000000fe800000-0x00000000fe900000] write-combining
PAT: [mem 0x00000000fe900000-0x00000000fea00000] write-combining

$ modprobe uio

$ insmod /root/dpdk-kmods/linux/igb_uio/igb_uio.ko wc_activate=1

$ cat /sys/kernel/debug/x86/pat_memtype_list | grep fe900000
PAT: [mem 0x00000000fe800000-0x00000000fe900000] write-combining
PAT: [mem 0x00000000fe900000-0x00000000fea00000] write-combining

$ /root/dpdk-stable/usertools/dpdk-devbind.py --status

Network devices using kernel driver
===================================
0000:00:05.0 'Elastic Network Adapter (ENA) ec20' if=eth0 drv=ena unused=igb_uio *Active*
0000:00:06.0 'Elastic Network Adapter (ENA) ec20' if=eth1 drv=ena unused=igb_uio *Active*

$ ip link set dev eth1 down

$ /root/dpdk-stable/usertools/dpdk-devbind.py --status

Network devices using kernel driver
===================================
0000:00:05.0 'Elastic Network Adapter (ENA) ec20' if=eth0 drv=ena unused=igb_uio *Active*
0000:00:06.0 'Elastic Network Adapter (ENA) ec20' if=eth1 drv=ena unused=igb_uio

$ cat /sys/kernel/debug/x86/pat_memtype_list | grep fe900000
PAT: [mem 0x00000000fe800000-0x00000000fe900000] write-combining
PAT: [mem 0x00000000fe900000-0x00000000fea00000] write-combining

$ /root/dpdk-stable/usertools/dpdk-devbind.py --bind igb_uio 00:06.0

$ /root/dpdk-stable/usertools/dpdk-devbind.py --status

Network devices using DPDK-compatible driver
============================================
0000:00:06.0 'Elastic Network Adapter (ENA) ec20' drv=igb_uio unused=ena

Network devices using kernel driver
===================================
0000:00:05.0 'Elastic Network Adapter (ENA) ec20' if=eth0 drv=ena unused=igb_uio
*Active*

$ cat /sys/kernel/debug/x86/pat_memtype_list | grep fe900000
PAT: [mem 0x00000000fe800000-0x00000000fe900000] write-combining
```

shaibran, 2022-04-14:
Hi @[techthumb](https://github.com/techthumb) this is s a known issue for DPDK v21.11 - there was a regression in the DPDK bus code which prevents the driver's requests to map the memory as WC when using igb_uio.ko.
Quick workaround is to cherry-pick this commit from DPDK v22.03: 6b92f184680283b4b3da4ebc722b5feb7ef280fb

we'll add this to our  README, thank you!

techthumb, 2022-04-14:
Same problem with v22.03 on Ubuntu

```
$ git status
HEAD detached at v22.03
nothing to commit, working tree clean
$ cat /sys/kernel/debug/x86/pat_memtype_list | grep fe900000
PAT: [mem 0x00000000fe800000-0x00000000fe900000] write-combining
$ cat /sys/kernel/debug/x86/pat_memtype_list
PAT memtype list:
PAT: [mem 0x00000000bffec000-0x00000000bffee000] write-back
PAT: [mem 0x00000000bffed000-0x00000000bfff0000] write-back
PAT: [mem 0x00000000bffef000-0x00000000bfff0000] write-back
PAT: [mem 0x00000000bfff0000-0x00000000bfff1000] write-back
PAT: [mem 0x00000000fe800000-0x00000000fe900000] write-combining
PAT: [mem 0x00000000febf0000-0x00000000febf2000] uncached-minus
PAT: [mem 0x00000000febf2000-0x00000000febf3000] uncached-minus
PAT: [mem 0x00000000febf4000-0x00000000febf8000] uncached-minus
PAT: [mem 0x00000000febf6000-0x00000000febf7000] uncached-minus
PAT: [mem 0x00000000fed00000-0x00000000fed01000] uncached-minus
PAT: [mem 0x00000000fed00000-0x00000000fed01000] uncached-minus
```

shaibran, 2022-04-27:
Let us investigate this, will keep you updated

shaibran, 2022-05-01:
we suspect you might be missing step 3 in this [guide](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#622-vfio-pci), need to start DPDK application in the background, which will be using the ENA HW - otherwise the memory will not be mapped.
We will improve our documentation to so WC verification section is clearer.

techthumb, 2022-05-01:
Many thanks for looking into this. 

I confirm that I am not starting a DPDK application prior to binding the NIC. 

Note: I am using igb_uio and not vfio-pci. 

Starting a DPDK application without initializing the NIC feels counter intuitive.

I'll follow these steps and report back tomorrow.

closed 2022-05-30 by amitbern-aws