amzn/amzn-drivers #218: Issue Xmitting Packets on Multi Process DPDK

naveen1anand, 2022-04-25:
I am using the DPDK Primary - Secondary process on arm64 - AWS instance with the ENA driver. Primary process sets Port config / mbuf pools, and secondary process sends out packet. I see Packets transmitted over the ENA driver, with Opackets / Obytes increasing. But don't see packets received on other aws machine.

If I transmit same packets over a single process,  it's received on other AWS machine. It fails, only If I run along with another Primary process.  
Anything I need to take care of here. Mainly, on ENA with igb_uio/vfio bind NIC card.

shaibran, 2022-05-01:
multi process support is still experimental but should basically function.
can you please share more details? 
DPDK version; do you use igb_iou or vfio; traffic characteristics the instance ID and region (which of the instances is tx/rx); etc.
I will try to see if I can detect anything on the framework logs; it will help if you could specify the time when issue occurred.

naveen1anand, 2022-05-02:
DPDK versions both 19.11 and 21.08 have tried -- I see issue on both versions. I use igb_uio.
From driver i see tx-poll/obytes and all tx related xstats incrementing on Primary process, but don't packets on pdump.

###### NIC extended statistics 
rx_good_packets: 4
tx_good_packets: 2449
rx_good_bytes: 168
tx_good_bytes: 161634
rx_missed_errors: 0
rx_errors: 0
tx_errors: 0
rx_mbuf_allocation_errors: 0
rx_q0_packets: 0
rx_q0_bytes: 0
rx_q0_errors: 0
tx_q0_packets: 2449
tx_q0_bytes: 161634
wd_expired: 0
dev_start: 1
dev_stop: 0
tx_drops: 0
bw_in_allowance_exceeded: 0
bw_out_allowance_exceeded: 0
pps_allowance_exceeded: 0
conntrack_allowance_exceeded: 0
linklocal_allowance_exceeded: 0
rx_q0_cnt: 0
rx_q0_bytes: 0
rx_q0_refill_partial: 0
rx_q0_bad_csum: 0
rx_q0_mbuf_alloc_fail: 0
rx_q0_bad_desc_num: 0
rx_q0_bad_req_id: 0
tx_q0_cnt: 2449
tx_q0_bytes: 161634
tx_q0_prepare_ctx_err: 0
tx_q0_linearize: 0
tx_q0_linearize_failed: 0
tx_q0_tx_poll: 2449
tx_q0_doorbells: 2449
tx_q0_bad_req_id: 0
tx_q0_available_desc: 1022

Currently, am trying only Tx traffic from secondary, but need Rx also.
My instance ID details:

Instance ID : i-01fa59c56a9e07818 (l2-arm-instance-1)
Region: us-east-1
Avilability Zone: us-east-1b
Mon May 02 2022 14:54:26 GMT+0530 (India Standard Time)

shaibran, 2022-06-06:
thanks, we'll review the info

shaibran, 2022-07-03:
Hi @naveen1anand we attempted to reproduce this in-house, however, the issue did be reproduced on the latest DPDK or the one stated above.
It’s most likely the an application issue.

closed 2022-07-11 by amitbern-aws