amzn/amzn-drivers #220: Instance becomes irresponsible after ena device reset (needing reboot).

iuridiniz, 2022-05-09:
Hello everyone,

We are getting a issue in our instances running in AWS with ena freezing up them.

---

Over the past four weeks, we got two freezing in two different instances, both instances have a similar workload. 

In the first event, one instance freezes up, requiring a reboot through AWS web console, few hours later, the other instance was irresponsible too requiring the same process. 

In that event, our team was suspecting some AWS infrastructure problem due timing and the similar symptoms.

But, in the last weekend, the same event occurred, one instance freeze up, and the other instance also freeze up within almost 24 hours interval. These time we done more research and found this messages in kern.log:

```
[1371895.517318] ena 0000:00:05.0 ens5: Found a Tx that wasn't completed on time, qid 2, index 0. 5756000 usecs have passed since last napi execution. Missing Tx timeout value 5000 msecs
... [several repeat of above message with same qid, but different index'es and increased number of usecs ... until device reset]
[1371911.325136] ena 0000:00:05.0 ens5: Found a Tx that wasn't completed on time, qid 2, index 1023. 21560000 usecs have passed since last napi execution. Missing Tx timeout value 5000 msecs
[1371911.339996] ena 0000:00:05.0 ens5: The number of lost tx completions is above the threshold (1019 > 128). Reset the device
```

After that device reset, some daemons complain about no TX buffer available (Fair enough because TX queue is full):

```
lldpd[819]: unable to send packet on real device for ens5: No buffer space available
```

We also got some complains about kernel hung tasks:

```
[1372035.208263] INFO: task kworker/0:1:2983 blocked for more than 120 seconds.
[1372035.214174]       Tainted: G           OE     4.19.0-18-amd64 #1 Debian 4.19.208-1
[1372035.222276] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[1372035.230478] kworker/0:1     D    0  2983      2 0x80000000
[1372035.235600] Workqueue: events linkwatch_event
[1372035.240115] Call Trace:
[1372035.243628]  __schedule+0x29f/0x840
[1372035.247747]  ? __switch_to_asm+0x35/0x70
[1372035.252041]  ? __switch_to_asm+0x41/0x70
[1372035.256346]  schedule+0x28/0x80
[1372035.260226]  schedule_preempt_disabled+0xa/0x10
[1372035.264817]  __mutex_lock.isra.8+0x286/0x470
[1372035.269293]  ? __switch_to_asm+0x35/0x70
[1372035.273581]  linkwatch_event+0xa/0x30
[1372035.277730]  process_one_work+0x1a7/0x3a0
[1372035.282052]  worker_thread+0x30/0x390
[1372035.286178]  ? create_worker+0x1a0/0x1a0
[1372035.290471]  kthread+0x112/0x130
[1372035.294390]  ? kthread_bind+0x30/0x30
[1372035.298537]  ret_from_fork+0x22/0x40

[1372035.302647] INFO: task kworker/u8:2:9108 blocked for more than 120 seconds.
[1372035.308514]       Tainted: G           OE     4.19.0-18-amd64 #1 Debian 4.19.208-1
[1372035.316551] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[1372035.324753] kworker/u8:2    D    0  9108      2 0x80000000
[1372035.329862] Workqueue: writeback wb_workfn (flush-259:0)
[1372035.334896] Call Trace:
[1372035.338400]  __schedule+0x29f/0x840
[1372035.342452]  schedule+0x28/0x80
[1372035.346343]  io_schedule+0x12/0x40
[1372035.350356]  wbt_wait+0x19b/0x300
[1372035.354350]  ? trace_event_raw_event_wbt_step+0x120/0x120
[1372035.359444]  rq_qos_throttle+0x31/0x40
[1372035.363676]  blk_mq_make_request+0x111/0x530
[1372035.368177]  generic_make_request+0x1a4/0x400
[1372035.372713]  submit_bio+0x45/0x130
[1372035.376812]  ? xfs_setfilesize_trans_alloc.isra.14+0x3d/0x90 [xfs]
[1372035.382376]  xfs_submit_ioend+0x98/0x1d0 [xfs]
[1372035.387018]  xfs_do_writepage+0x1dc/0x480 [xfs]
[1372035.391661]  write_cache_pages+0x1aa/0x440
[1372035.396111]  ? xfs_vm_releasepage+0x80/0x80 [xfs]
[1372035.400885]  xfs_vm_writepages+0x64/0xa0 [xfs]
[1372035.405459]  do_writepages+0x41/0xd0
[1372035.409612]  ? __blk_mq_try_issue_directly+0x176/0x1c0
[1372035.414562]  __writeback_single_inode+0x3d/0x300
[1372035.419237]  writeback_sb_inodes+0x1e7/0x440
[1372035.423750]  __writeback_inodes_wb+0x5f/0xc0
[1372035.428244]  wb_writeback+0x25b/0x2f0
[1372035.432434]  wb_workfn+0x337/0x400
[1372035.436496]  process_one_work+0x1a7/0x3a0
[1372035.440850]  worker_thread+0x30/0x390
[1372035.445072]  ? create_worker+0x1a0/0x1a0
[1372035.449390]  kthread+0x112/0x130
[1372035.453364]  ? kthread_bind+0x30/0x30
[1372035.512571]  ret_from_fork+0x22/0x40
```

We did ping the instance, but we got no response and we had to ask to reboot via AWS web console. But according to `ENA Linux Driver Best Practices and Performance Optimization Guide`, this is not needed:

> Q: What is ENA device reset?
> A: ENA device reset is a self healing mechanism that is triggered when the driver detects unexpected device behavior. Example of such behavior could be an unresponsive device, missing keep-alive events from the device, Tx completions timeouts, netdev > timeout etc. The device reset is a rare event, lasts less than a millisecond and might incur loss of traffic during this time, which is expected to be recovered by the transport protocol in the instance kernel.

_The device reset [...] lasts less than a millisecond and might incur loss of traffic during this time, which is expected to be recovered by the transport protocol in the instance kernel._

We think there is a bug on device reset, since our instances becomes irresponsible after device reset, requiring a reboot to come back online

One interesting thing:
Immediately after asking to reboot, the instance become responsive again (it responds to ping messages, so it was a network freeze and it is not freeze anymore after some thing occurred, but what?), it does shutdown going irresponsible (ping stop responding due instance shutting down) and it goes responsive again (when instance is up). 

Due this, we think that there's something preventing the device reset to work properly, but we don't known what is. We will appreciate any kind of helping to find what is causing these problems:
- the TX timeouts triggering the reset; and (We think related to our workload)
- the network freeze after device reset.

In the both events the CPU usage was low (less than 25%) and there was enough free memory.

-------

Extra info:

**kernel version:** stock 4.19.0-18-amd64 
**ena version:**  2.5.0g
**instance type:** m5a.2xlarge
**distro version:** Debian GNU/Linux 10 (buster)

We are not able right now to upgrade ena to version 2.7.1, but this upgrade is planned by our team.

Previously, The instances were using ena version 2.1.0 (stock) and we upgrade them to version 2.5.0g 
I think that we've upgraded them in 2020 due an ENA issue (due a device reset too, but with no ping responses at any time) that we thinking that was fixed by version due no more events like that, if you guys think that old issue was important, we can provide more details about that.

-------

Workload details:

We are running a process with `SCHED_FIFO` that spawns several threads (sometimes with up to 400 threads) with the same priority (`SCHED_FIFO`), we think this could contribute to TX timeouts.

But we have graphs about  number of threads and (during the events) the number of threads (around 250 in the last freeze) were under the max threads of that day (364 threads max in the last freeze, with no issues in that time)

We are planning to fix that issue too (by firing fewer threads with this priority or by not using `SCHED_FIFO` priority).  We also are studding make some PERF's in production to check the current workload.

-------

Anyway, any help will be appreciated.

zorikm, 2022-05-09:
Hi Iuri. 
From the first brief look, it doesn't look like the ENA driver is the one to be blamed here.  
Looks like instance vCPUs are locked up (wb_workfn task and linkwatch_event task). 
With vCPUs not being available for more than 120 sec, ENA wouldn't be able to function properly, and ENA reset wouldn't help as well. 
I don't know what type of workload you run (400 threads on 4 vCPUs seems to be fairly intense, especially if they do I/O and starved competing for global locks) , so it is hard to come up with suggestions. wb_workfn task blocking seems to be related  to a FS writeback being stuck, and if I read the backtrace right, it uses block interface as a backend.
So it can be many things, either overwhelmed block interface or insufficient instance kernel memory. We would also gladly take a look at the ENA side. Please provide us instance-id and timeline when the issue happened.
Could be really helpful if you could capture the perf output, and also monitor kernel memory usage.

iuridiniz, 2022-05-16:
HI @zorikm,

I will be glad if you can verify.

Instance ID: **i-0cb4a85e31b4ad438**
Timestamp (YYYY-MM-DD HH:MM:SS UTC) of freeze: **2022-05-06 13:15:00 UTC**

akiyano, 2022-05-18:
Hi @iuridiniz, 
We are looking into this and will update you here.

akiyano, 2022-05-22:
Hi @iuridiniz,
I looked at the instance you sent us, and I see the missing TX completions reset attempt.
However this reset does not seem to be the cause of the problem, it is only the result of it.
Something happens in the instance that locks up the cpus.
The ena driver doesn't get cpu time, and therefore isn't able to handle TX completions.
The ena driver tries to recover by scheduling a device reset, but the actual code that resets the device is never run, because the cpus stay locked up.
You need to find the root cause of the cpus lock up.
We would recommend contacting AWS support for further assistance

iuridiniz, 2022-05-31:
Hello Zorikm and Akiyano,

We think we found the source of our TX timeout, it's our workload. Kernel hung tasks are a consequence, not a cause.

We went deep in investigation and we discovered enough evidence that our workload is starving other processes in the same CPU.

Also, we found that ENA reset is working properly (in 2.5.0g). We got a heavy workload (but not sufficient to freeze up the machine) that triggered the ENA device reset and it was successful. So we had the same conclusion, we think that device reset procedure was starving too, so it's not an ENA issue.

Thank you all, your responses helped us to go in the correct direction to solve our problem.

PS: If you guys are interested in more details and the outcome, feel free to contact me in person. I'm closing this issue since it isn't an ENA issue.

closed 2022-05-31 by iuridiniz

iuridiniz, 2022-11-15:
Hello everyone, I've been contacted by some of you guys to clarify the root cause.

So this message is about this.

### What did we discovery?

Our workload runs some threads with realtime priority (SCHED_FIFO) and we got some hints that these were starving other process (including ksoftirqd threads who ran linux network drivers bottom halves)

### What kind of hints?

- kernel messages like this (some associated with a freeze instance):
  - `[1372035.302647] INFO: task kworker/u8:2:9108 blocked for more than 120 seconds.`
  - `[4126837.826790] sched: RT throttling activated`
- Graphs (not available at time of writing due recycle) showing a persistent usage of 100% in one isolated CPU Thread for some time while other CPU Threads are OK.

### What did we do?

After some talks with devel team, we conclude we don't need these SCHED_FIFO threads.
So, after removing these threads, we are running without any issues ever since (no downtime, no kernel messages, no abnormally isolated CPU Thread usage).

### What did we conclude?

We think that (due how SCHED_FIFO threads works in linux), one SCHED_FIFO thread occupied one CPU Thread for sufficient time to starve ksoftirqd thread on the same of the same CPU Thread, the ENA's watchdog (??) detected that and the rest is in the issue description.

davidarinzon, 2022-11-16:
Thank you for providing the insights of your investigation @iuridiniz