amzn/amzn-drivers #224: Fail ena_stats_get function with pktgen on AWS EC2

Mark840308, 2022-06-06:
Hi,

I'm trying to test latency with pktgen on EC2 instance. 
There is problem while pktgen trying to retrieve statistics from ena. Log message are as followed.

And is that possible to send packets using MAC address between two region instance with VPC set .

If this is not the right forum to get an answer please help me with a link where I can get some help.

DPDK version: 20.11.2
pktgen version: 20.11.3

```
Copyright (c) <2010-2020>, Intel Corporation. All rights reserved. Powered by DPDK
EAL: Detected 2 lcore(s)
EAL: Detected 1 NUMA nodes
EAL: Detected shared linkage of DPDK
EAL: Multi-process socket /var/run/dpdk/rte/mp_socket
EAL: Selected IOVA mode 'PA'
EAL: No available hugepages reported in hugepages-2048kB
EAL: Probing VFIO support...
EAL: VFIO support initialized
EAL:   Invalid NUMA socket, default to 0
EAL:   Invalid NUMA socket, default to 0
EAL: Probe PCI driver: net_ena (1d0f:ec20) device: 0000:00:06.0 (socket 0)
EAL:   Invalid NUMA socket, default to 0
EAL: Probe PCI driver: net_ena (1d0f:ec20) device: 0000:00:07.0 (socket 0)
EAL: No legacy callbacks, legacy socket not created

*** Copyright (c) <2010-2020>, Intel Corporation. All rights reserved.
*** Pktgen  created by: Keith Wiles -- >>> Powered by DPDK <<<

 Port: Name         IfIndex Alias        NUMA  PCI
    0: net_ena         0                   0   1d0f:ec20/00:06.0
    1: net_ena         0                   0   1d0f:ec20/00:07.0

Initialize Port 0 -- TxQ 1, RxQ 1
Src MAC 0a:4e:da:ec:ae:f3
 <Promiscuous mode Enabled>
Initialize Port 1 -- TxQ 1, RxQ 1
Src MAC 0a:57:cd:23:c0:af
 <Promiscuous mode Enabled> <Promiscuous mode Enabled>

  RX/TX processing lcore:   1 rx:  2 tx:  2
| Ports 0-1 of 2   <Main Page>  Copyright (c) <2010-2020>, Intel Corporation
  Flags:Port        : P------Lat         :0 P------Single      :1
Link State          :         <UP-10000-FD>         <UP-10000-FD>      ---Total Rate---
Pkts/s Max/Rx       :18446744073709551608/018446744073709551608/0                   1/0
       Max/Tx       :                   0/0                   0/0                   0/0
MBits/s Rx/Tx       :                   0/0                   0/0                   0/0
Broadcast           :                     0                     0
Multicast           :                     8                     8
Sizes 64            :                     0                     0
      65-127        :                     0                     0
      128-255       :                     0                     0
      256-511       :                     0                     0
      512-1023      :                     0                     0
      1024-1518     :                     0                     0
Runts/Jumbos        :                   8/0                   8/0
ARP/ICMP Pkts       :                   8/0                   8/0
Errors Rx/Tx        :                   0/0                   0/0
Total Rx Pkts       :                     0                     0
      Tx Pkts       :                     0                     0
      Rx MBs        :                     0                     0
      Tx MBs        :                     0                     0
                    :
Pattern Type        :               abcd...               abcd...
Tx Count/% Rate     :         Forever /100%         Forever /100%
Pkt Size/Tx Burst   :             80 /   32             64 /   32
TTL/Port Src/Dest   :         4/ 1234/ 5678         4/ 1234/ 5678
Pkt Type:VLAN ID    :       IPv4 / UDP:0001       IPv4 / TCP:0001
802.1p CoS/DSCP/IPP :             0/  0/  0             0/  0/  0
VxLAN Flg/Grp/vid   :      0000/    0/    0      0000/    0/    0
IP  Destination     :           192.168.1.1           192.168.0.1
    Source          :        192.168.0.1/24        192.168.1.1/24
MAC Destination     :     0a:57:cd:23:c0:af     0a:4e:da:ec:ae:f3
    Source          :     0a:4e:da:ec:ae:f3     0a:57:cd:23:c0:af
PCI Vendor/Addr     :     1d0f:ec20/00:06.0     1d0f:ec20/00:07.0

Pktgen:/> str
Pktgen:/> ena_stats_get(): Could not retrieve statistics from ENA

0 0 > 8
q         quitena_stats_get(): Could not retrieve statistics from ENA

0 0 > 8
ena_stats_get(): Could not retrieve statistics from ENA

ena_stats_get(): Could not retrieve statistics from ENA

```

shaibran, 2022-06-06:
This is likely since the get stat command is being called from the secondary process.
Stats gathering requires retrieving statistics through admin queue command which can be done only from the primary process.
See [documentation](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#81-generic-statistics) 

With that being said, we intend to change the PMD behavior in the next release v2.8.0 and to avoid the admin command usage in this flow which should allow gathering basic stats not only from the primary process.

Mark840308, 2022-06-07:
Thanks for your responds.

1. My pktgen command is:
```
enable 0 latency
str
```
2. my module is igb_uio
3. instance type is c5.large

I'm rebuilding whole dpdk and pktgen with new instance(C5.large, Amazon linux) and update dpdk, pktgen to newest version. 
Trying to figure out if there is some kernel configuration problem may cause impact.

If there is more information i'll leave comment after. 

Thanks a lot

shaibran, 2023-08-10:
This was not reproduced in-house with this DPDK and Pktgen versions, so this is likely application configuration issue

closed 2023-08-10 by shaibran