amzn/amzn-drivers #227: Support Dynamic Device Personalization (DDP)

loneizhao, 2022-06-15:
My customer is asking if Amazon ENA support DDP,  if not , any plan to support it ? Thanks

https://www.intel.com/content/www/us/en/architecture-and-technology/ethernet/dynamic-device-personalization-brief.html

davidarinzon, 2022-06-15:
Hi @loneizhao,

Thank you for providing the introduction link above.
Can you share some insights on the use-cases that the customer has with this functionality?

Thanks!

davidarinzon, 2022-06-26:
@loneizhao 
Do you possibly have information on how the customer is planning to use the feature? What are the use-cases?
You're welcome to contact me via Email if you wish.

davidarinzon, 2022-07-07:
Hi @loneizhao,
This feature is not planned to be supported by ENA at the moment.
Thanks.

closed 2022-07-07 by davidarinzon