amzn/amzn-drivers #235: DPDK ENA PMD Silently Drops Packets on Rx

matrackif, 2022-08-08:
Hello,

DPDK Version 22.03, ENA PMD version 2.6.0.

We have a C++ application that uses the [DPDK ENA PMD](https://github.com/DPDK/dpdk/tree/main/drivers/net/ena) in order to consume as many packets as possible in an EC2 instance cloud environment. In general, everything seems to work, but we observed some strange behavior. I will not paste the code directly, but the logic can be summarized follows:

1. Create one rx queue using the DPDK API function [rte_eth_rx_queue_setup()](https://doc.dpdk.org/api/rte__ethdev_8h.html#a36ba70a5a6fce2c2c1f774828ba78f8d). We tried varying values for the `nb_rx_desc` parameter, more on that later.
2. In an infinite loop, call `rte_eth_rx_burst(port , 0, pkts_burst, 8)`. What we do with the received packets is irrelevant for this discussion.   Our application always had enough mbufs in the memory pool to handle subsequent calls to `rte_eth_rx_burst()`, and we always called the function as fast as possible to make sure we drain the HW queue of packets. 

**Scenario:** 
We used the AWS port mirroring technology to receive packets on the DPDK port. AWS instances have a maximum bandwidth and PPS limits. However if these limits are exceeded AWS CloudWatch reports them, and we can query ethtool -S on machines that use the kernelspace ENA driver, and we can call [rte_eth_stats_get()](https://doc.dpdk.org/api/rte__ethdev_8h.html#adec226574c53ae413252c9b15f6f4bab) on the DPDK ENA PMD ports. We used iperf between 2 instances, and the outgoing traffic was mirrored to the DPDK port on a third instance.

**Observation of the problem:**
 We understand that production traffic is preferred over mirrored traffic in AWS. However CloudWatch reports exactly how many packets were actually sent out to the Traffic Mirror Target (TMT), with the [NetworkPacketsMirrorOut](https://docs.aws.amazon.com/vpc/latest/mirroring/traffic-mirror-cloudwatch.html) statistic. Now the rx ring size from the  `rte_eth_rx_queue_setup()` comes into play. 

1. When the ring size, i.e. `nb_rx_desc` was 512, we observed around 15-20% silent drops. This means `NetworkPacketsMirrorOut` values were about 10-15% larger than the total amount of packets received by `rte_eth_rx_burst()`.
2. When the ring size, i.e. `nb_rx_desc` was 1024, we observed around 5% silent drops.
3. When the ring size, i.e. `nb_rx_desc` was 2048 (the maximum allowed), we observed almost no silent drops. 

This is what we mean by "silent drops":

- `rte_eth_stats_get()` returns no drops. [imissed](https://doc.dpdk.org/api/structrte__eth__stats.html#a462703bfeca72b2d1039f588d8ddbf3c), [ierrors](https://doc.dpdk.org/api/structrte__eth__stats.html#aa4f59b58c69cf6d99f2bdfcad48c5774), [rx_nombuf](https://doc.dpdk.org/api/structrte__eth__stats.html#a8afc3f87397701e58b218d9dad03be8c) are all zero! The number of packets received, ([ipackets](https://doc.dpdk.org/api/structrte__eth__stats.html#ac14b69830000494630c8897b70c5fc54)) was equal to the total number of packets received by `rte_eth_rx_burst()`
- All allowance exceeded metrics (such as bw_in_allowance_exceeded, etc) in [rte_eth_xstats_get()](https://doc.dpdk.org/api/rte__ethdev_8h.html#a300d75b583c1f5acfe5b162a5d8c0ac1) also are zero

**Conclusion:**
One of the following is true:

1. The DPDK ENA PMD reports incorrect statistics. It does not catch the moment of HW queues dropping packets, since meddling with the RX ring size clearly influences the problem, but no AWS metric or DPDK API function actually shows the drop count.
2. The DPDK ENA PMD reports correct statistics. The reason it reports no rx drops is because AWS infrastructure drops the packets silently before they arrive on the NIC. This is unlikely since increasing ring size means we approach the correct `NetworkPacketsMirrorOut` value in CloudWatch. Which signifies that this is an application/PMD problem, not an AWS problem.
3. Our application is not performing some book-keeping needed by the ENA PMD in order for it to properly count rx drops. Looking at the ENA PMD source code, we noticed it creates some DPDK timers used for keep-alive. Calling [rte_timer_manage()](https://doc.dpdk.org/api/rte__timer_8h.html#aad81303c11541adfb88c62c47b863fa6) in a loop did not help.

Thank you

shaibran, 2022-08-10:
Let us review this and get back to you

shaibran, 2022-08-10:
We have already verified that Rx drops and limiters works properly on DPDK v22.03 by running the testpmd (executed it on c5n instances), but it did not include mirorring.
As next steps please provide:
1. the instance types being used so we can test in-house.
2. the instance ID and the region where it resides so we can review its statistics internally.

Thanks

matrackif, 2022-08-10:
Hi, 

Thanks for your reply

3 instances located in eu-west-1 Europe (Ireland)

1. First instance: The Traffic Mirror Source and iperf client instance: m5.4xlarge, 5Gbit baseline bandwith: id= i-0e60d2d4bbf4f3183
2. Second instance m6i.4xlarge: The iperf server instance. 6.25Gbit baseline bandwidth. id= i-0d49db29a1c6ac6d
3. Third instance,  m6i.4xlarge: Traffic Mirror Target with 2 ENIs, the builtin ENI is the management port and the second ENI is the DPDK PMD Driver enabled port which is the traffic mirror target. instance id= i-08f98e99f99505370

The situation is that instance 1 sends UDP datagrams using iperf to instance 2. The client can run the basic `./iperf -c <IP> -u -l 1024`, whereas the server simply runs `./iperf -u -s`. The outgoing traffic of instance 1 is mirrored to instance 3's DPDK enabled port. 

We expect that `NetworkPacketsMirrorOut` statistic on instance 1 matches the number of datagrams received by instance 3's DPDK port. But as I mentioned in my issue, we see that when the RX ring size is decreased, the DPDK port receives less and less packets compared to `NetworkPacketsMirrorOut`. These drops are not reported by AWS and the DPDK API.

If you need any more information, feel free to ask.

Thank you

matrackif, 2022-08-10:
Update:

I was not able to reproduce the issue with dpdk-testpmd. I looked at the source code for testpmd, and [all it does is instantly read the packet and free it in a busy loop, doing no processing in between.](https://github.com/DPDK/dpdk/blob/main/app/test-pmd/rxonly.c). 

I will modify it to sleep in-between calls to rx_burst, and stress it with iperf traffic, and check if that causes a silent drop issue. Will post later with my findings. I am trying to strain testpmd as much as possible in order to increase the chance of rx drops. By running something like this:

`dpdk-testpmd -l 0-1 -- -i --forward-mode=rxonly --rxd=128 --rxq=1 --total-num-mbufs=1025 --auto-start --disable-rss --record-burst-stats`

matrackif, 2022-08-11:
Silent drops HAVE been observed with dpdk-testpmd, they have nothing to do with port mirroring it seems. testpmd by default is extremely fast because it reads packets on a dedicated lcore in a busy loop. It is quite difficult to "overwhelm" the application with high bandwidth traffic. 

However if a sleep is inserted in between calls to [rte_eth_rx_burst()](https://doc.dpdk.org/api/rte__ethdev_8h.html#a3e7d76a451b46348686ea97d6367f102), then it is quite easy to reproduce silent drops. 

In order to reproduce silent rx drops: please do the following: 

1. Checkout DPDK version 22.03 `git clone --branch=v22.03 https://github.com/DPDK/dpdk.git`
2. Apply the attached patch [testpmd-sleep.txt](https://github.com/amzn/amzn-drivers/files/9309563/testpmd-sleep.txt) `git apply testpmd-sleep.txt`. This patch adds the command line argument --sleep-time=N, where N is the number of microseconds to sleep in between calls to the packet forwarding function, which in the case of --forward-mode=rxonly, is only [rte_eth_rx_burst()](https://doc.dpdk.org/api/rte__ethdev_8h.html#a3e7d76a451b46348686ea97d6367f102).
3. Run dpdk-testpmd, sleeping 100 microseconds after each call to rx_burst: `dpdk-testpmd -l 0-1 -- -i --forward-mode=rxonly --rxd=128 --rxq=1 --total-num-mbufs=1025 --auto-start --disable-rss --record-burst-stats --sleep-time=100`
4. Send a lot of traffic to the DPDK port, I used port mirroring + iperf, but it can be any method. 

dpdk-testpmd received only 2024017 packets, even though CloudWatch reported 3507384 `NetworkPacketsMirrorOut`:

[image]

```
######################## NIC statistics for port 0  ########################
  RX-packets: 2024017    RX-missed: 0          RX-bytes:  315746728
  RX-errors: 0
  RX-nombuf:  0
  TX-packets: 0          TX-errors: 0          TX-bytes:  0

  Throughput (since last show)
  Rx-pps:        93876          Rx-bps:    117157872
  Tx-pps:            0          Tx-bps:            0
  ############################################################################
```

Notice how no drops were reported by dpdk-testpmd. We observed the exact same behavior in our own application.

shaibran, 2022-08-14:
@matrackif we found that there is indeed a known gap of silent drops on the 6th generation hardware which currently doesn’t support reporting packet drops due to lack of receive buffers.
Issue is handled on high priority but I am unable to provide a resolution date yet since this is not being handled in PMD layer.
We will update our documentation accordingly.
Thanks you!

matrackif, 2022-08-16:
Thank you for your reply. I have a few 

1. So this is a ENA device issue, the logic in the ENA PMD C code  will not be corrected?
2. By "6th gen", I assume this problem occurs on m6i instances and not m5 instances? If so do m5 instances properly report rx drops? If that's the case should I retest this issue on m5.4xlarge instances?

Regards

shaibran, 2022-08-17:
1. you are correct, the logic in the ENA PMD C code will not change.
2. "6th gen" is indeed the m6i instances and not m5 instances, on which you should not see silent drops.

shaibran, 2023-02-09:
I will check and get back to you with the info.

BR,
Shai

shaibran, 2023-03-21:
@matrackif sorry for the delay. The issue is not fixed along with the traffic mirroring feature.
I am unable to provide an estimated time for resolution.

shaibran, 2023-08-10:
**An update from ENA:**
This is a known gap of silent drops on the 6th generation hardware which currently doesn’t support reporting packet drops due to lack of receive buffers (which is not a bug and can happen when application does not release buffers back to the receive queue fast enough). We recently implemented support to report such drops via a new rx_overrun statistic and plan to include it in our next release ENAv2.8.0 which is targeted to DPDKD23.11

closed 2023-08-10 by shaibran

jonmoon76, 2024-02-01:
@matrackif  - Has DPDK23.11 solved your problem?   

I am seeing similar 'silent drops' but including m5 and m5n instance types, i.e. not just restricted to m6in.

I also see that the bug identified was triggered due to a lack of mbufs, when you were clear that is not the case in your scenario.

Thanks.

xguerin, 2024-12-07:
> We recently implemented support to report such drops via a new rx_overrun statistic and plan to include it in our next release ENAv2.8.0 which is targeted to DPDKD23.11

Hello @shaibran, I'm taking the liberty to comment on this closed issue to ask whether or not the reporting of those silent drops is now working. I'm seeing odd packet loss on DPDK 23.11, sometimes up to 100s in a row, and the device is not reporting any packet miss. I have a hard time believing those drops are coming from the network.