amzn/amzn-drivers #236: setting lpc_size to a non-default value

chuckfloyd, 2022-08-09:
What is the preferred method of setting lpc_size so that the new value is implemented? It does not appear that there's a writable parameter for the value, and this requires a reboot or device reset. Will the module read a config file in /etc/modules-load.d? Do I need to modify the source and recompile the module with a new default value? I have a few systems that show increasing values of queue_#_rx_lpc_full.

davidarinzon, 2022-08-10:
Hi @chuckfloyd 

The way to change this value is through a module parameter, and it requires the reboot/reset as you've noted. Alternatively, you may do as you suggested, and alter the driver with a different value for the `lpc_size` module parameter, but I would not recommend this option.

Another viable option would be similar to what you've suggested, you can create a config file `ena.conf` in `/etc/modprobe.d` and specify `options ena lpc_size=<the size you want>`.

More information is available at https://github.com/amzn/amzn-drivers/tree/master/kernel/linux/ena#local-page-cache-lpc.

Feel free to reach out to me at darinzon@amazon.com for further assistance, if needed.

chuckfloyd, 2022-08-15:
Thanks! Since we're only seeing enough queues to enable LPC on very large instances, and all of those are experiencing increasing lpc_full counts at the default setting, I think we'll explore the module options route to increase lpc_size and observe.

chuckfloyd, 2022-08-31:
For my systems on CentOS 7, I had to include the new /etc/modprobe.d/ena.conf with the options line in the initramfs so that it was available when the driver first loaded during the boot process.

closed 2022-08-31 by chuckfloyd

AWSNB, 2022-08-31:
Thanks for the update Chuck,  glad you got it working
We should update our docs to cover this case