amzn/amzn-drivers #237: Compile error with kernel 5.19.2-1.el7.elrepo - CentOS 7.9

vvhungy, 2022-08-20:
I got a compilation error as below. Please check.
Thanks.

OS: CentOS 7.9.2009
Kernel 5.19.2-1.el7.elrepo.x86_64 - using elrepo repository (http://elrepo.org/tiki/HomePage).
ena driver version 2.7.4

> DKMS make.log for amzn-drivers-2.7.4 for kernel 5.19.2-1.el7.elrepo.x86_64 (x86_64)
> Sat Aug 20 13:09:58 +08 2022
> make: Entering directory '/var/lib/dkms/amzn-drivers/2.7.4/build/kernel/linux/ena'
> make -C /lib/modules/5.19.2-1.el7.elrepo.x86_64/build M=/var/lib/dkms/amzn-drivers/2.7.4/build/kernel/linux/ena modules
> make[1]: Entering directory '/usr/src/kernels/5.19.2-1.el7.elrepo.x86_64'
>   CC [M]  /var/lib/dkms/amzn-drivers/2.7.4/build/kernel/linux/ena/ena_netdev.o
> In file included from /var/lib/dkms/amzn-drivers/2.7.4/build/kernel/linux/ena/ena_netdev.c:28:
> /var/lib/dkms/amzn-drivers/2.7.4/build/kernel/linux/ena/ena_xdp.h: In function ‘ena_xdp_execute’:
> /var/lib/dkms/amzn-drivers/2.7.4/build/kernel/linux/ena/ena_xdp.h:176:31: error: passing argument 1 of ‘bpf_warn_invalid_xdp_action’ makes pointer from integer without a cast [-Werror=int-conversion]
>   176 |   bpf_warn_invalid_xdp_action(verdict);
>       |                               ^~~~~~~
>       |                               |
>       |                               u32 {aka unsigned int}
> compilation terminated due to -Wfatal-errors.
> cc1: all warnings being treated as errors
> make[2]: *** [scripts/Makefile.build:249: /var/lib/dkms/amzn-drivers/2.7.4/build/kernel/linux/ena/ena_netdev.o] Error 1
> make[1]: *** [Makefile:1849: /var/lib/dkms/amzn-drivers/2.7.4/build/kernel/linux/ena] Error 2
> make[1]: Leaving directory '/usr/src/kernels/5.19.2-1.el7.elrepo.x86_64'
> make: *** [Makefile:59: all] Error 2
> make: Leaving directory '/var/lib/dkms/amzn-drivers/2.7.4/build/kernel/linux/ena'
>

akiyano, 2022-08-22:
Hi @vvhungy,
Thanks for reporting this issue.
Please use the attached patch to resolve the issue.
To apply the patch, please copy it into the amzn-drivers directory and run:
`git am 0001-fix-bpf_warn_invalid_xdp_action-compilation-error.patch.txt`

[0001-fix-bpf_warn_invalid_xdp_action-compilation-error.patch.txt](https://github.com/amzn/amzn-drivers/files/9392177/0001-fix-bpf_warn_invalid_xdp_action-compilation-error.patch.txt)

This fix will be included in a future release.

vvhungy, 2022-08-23:
The patch worked as expected.
Tks.

davidarinzon, 2022-09-13:
This issue is resolved with the release of 2.8.0g driver

closed 2022-09-13 by davidarinzon