amzn/amzn-drivers #239: Failed to set RSS hash function in the device

soulofmitsui, 2022-08-27:
Hi,

This is really a question than an issue. We are using the DPDK ena driver on m5dn and i3en instances. It worked well. We recently tried to run the same on m5d instances and got the following error when setting up RSS:
```
ena_rss_hash_set(): Failed to set RSS hash function in the device
ena_rss_hash_update(): Failed to set RSS hash
```
Is that all instances without the 'n' suffix doesn't support RSS with DPDK driver?

Thanks,
Yang

shaibran, 2022-08-29:
RSS is supported on all HW, however, the [advanced features](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#11-rss-support) are supported only on subset of the HWs.
m5d is likely the 4th generation ENA which does not support it.

soulofmitsui, 2022-08-29:
Thanks shaibran. Can I assume all the instances with the 'n' suffix support changing RSS key? If not, is there a way to check for the ENA h/w version (assuming version > 4 supports RSS key change)?

shaibran, 2022-08-30:
This is correct for the 5th generation, but it's not straight forward to know it from the [documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/instance-types.html)
The 4th generation C/M/R instance types run on ENAv1
The 5th generation C/M/R instance types without the 'n' run on ENAv1
The 5th generation C/M/R instance types with the 'n' (enhanced networking) run on ENAv2
The 6th generation C/M/R instance types run on ENAv3
However, there are additional instance types which it is complicated to know what is the ENA HW on it, e.g., x/z/a/i families.
I will request modification to the official aws documentation to include the ENAv[1,2,3] for each instance type

shaibran, 2022-08-31:
It's not straight forward to know it from the AWS documentation, but you can assume that the advanced RSS feature are supported for the below instance types: 
* all “n” instances. 
* 6th generation (excluding c/m/r6g)
* For all other instances types you can use ```ethtool -X``` to figure it out (Linux)

soulofmitsui, 2022-09-02:
Thanks shaibran!

closed 2022-09-02 by soulofmitsui

shaibran, 2022-10-11:
short answer is yes.  When you run your PMD on instance with 'n' suffix it will support the new RSS capabilities.
But this is a just a by-product of the minimal PMD version which can run on the 'n' instance type rather than instance type capability.
RSS is not dependent on the instance type, but since newer PMD which supports enhanced RSS capabilities (e.g., changing RSS key) support these newer instance types ('n', 6th gen,...) if you will use the newer instances it will allow you to get it "for free". Alternatively, you can apply our latest backports and get this capability also on older HWs.