amzn/amzn-drivers #240: Limitations on the number of combined channels in XDP

akhota, 2022-09-16:
Hello,

The ena driver fails to load the xdp object if the combined channels exceeds half of the maximum allowed channel count.

```
$ ethtool -i eth1
driver: ena
version: 2.7.2g
firmware-version:
expansion-rom-version:
bus-info: 0000:00:06.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: yes

$ ethtool -l eth1
Channel parameters for eth1:
Pre-set maximums:
RX:             0
TX:             0
Other:          0
Combined:       8
Current hardware settings:
RX:             0
TX:             0
Other:          0
Combined:       8

(Use xdp-project sample program)
$ git clone https://github.com/xdp-project/xdp-tutorial.git
$ cd xdp-tutorial/basic01-xdp-pass/
$ git submodule update --init
$ make

$ sudo ip link set eth1 mtu 3498
$ sudo ip link set dev eth1 xdp obj xdp_pass_kern.o sec xdp
(snip)
Error: ena: Failed to set xdp program, there is no enough space for allocating XDP queues, Check the dmesg for more info.

$ dmesg -T
(snip)
[Fri Sep 16 14:29:02 2022] ena 0000:00:06.0 eth1: Failed to set xdp program, the Rx/Tx channel count should be at most half of the maximum allowed channel count. The current queue count (8), the maximal queue count (8)
```

I assume that this limit is determined by `ena_xdp_legal_queue_count()`.

* https://github.com/amzn/amzn-drivers/blob/ena_linux_2.8.0/kernel/linux/ena/ena_xdp.h#L81

Other NIC drivers, for example the Intel ixgbe driver, can load the xdp module even if all channels are combined.

```
$ ethtool -i enp1s0f0
driver: ixgbe
version: 5.15.0-47-generic
firmware-version: 0x00012b2c
expansion-rom-version:
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: yes

$ ethtool -l enp1s0f0
Channel parameters for enp1s0f0:
Pre-set maximums:
RX:             n/a
TX:             n/a
Other:          1
Combined:       8
Current hardware settings:
RX:             n/a
TX:             n/a
Other:          1
Combined:       8

$ sudo ip link set dev enp1s0f0 xdp obj xdp_pass_kern.o sec xdp

$ ip l sh enp1s0f0
3: enp1s0f0: <BROADCAST,MULTICAST> mtu 1500 xdp qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 00:1b:21:bc:04:a2 brd ff:ff:ff:ff:ff:ff
    prog/xdp id 33

$ sudo dmesg -T
(snip)
[Fri Sep 16 11:21:00 2022] ixgbe 0000:01:00.0: Multiqueue Enabled: Rx Queue count = 8, Tx Queue count = 8 XDP Queue count = 8
```

Why does the ena driver have this limitation?

Our environment is:

* AWS
  * AMI: Amazon Linux 2 AMI (HVM) - Kernel 5.10, SSD Volume Type
  * Instance type: c6i.xlarge - 32xlarge, a1.2xlarge
  * Driver version: ena 2.7.2g (build from source)
* Bare metal (for comparison)
  * OS: Ubuntu 22.04 LTS server
  * CPU architecture: x86_64
  * Driver version: ixgbe, included in 5.15.0-47-generic kernel

Thanks,

ShayAgros, 2022-09-19:
Hi,
this limitation comes from the implementation of XDP support in the ENA driver. We chose to do it to avoid synchronizing between concurrent flows of transmissions (both kernel-stack packets combined with XDP_REDIRECT and XDP_TX).

Can I ask what flow are you testing and if you see performance degradation using half of the queues ?

akhota, 2022-09-21:
Hi @ShayAgros ,

> Can I ask what flow are you testing and if you see performance degradation using half of the queues ?

We are using attached program for our testing.

* [multiqueue-process-udp-packet-main.tar.gz](https://github.com/amzn/amzn-drivers/files/9613284/multiqueue-process-udp-packet-main.tar.gz)

This program is a modification of `samples/bpf/xsk_fwd.c` included in the Linux kernel sources.
On the Amazon Linux 2 AMI, you can build as following procedure:

```
$ sudo yum install clang llvm gcc libbpf libbpf-devel bpftool kernel-headers make

$ tar xvpzf multiqueue-process-udp-packet-main.tar.gz
$ cd multiqueue-process-udp-packet-main/
$ make
```

We are using this program as following:

```
$ sudo ethtool -L eth1 combined 4   # a half of NIC queues
$ sudo ip link set eth1 mtu 3498

$ sudo ./xsk_fwd \
  -i eth1 -q 0 -c 0 \
  -i eth1 -q 1 -c 1 \
  -i eth1 -q 2 -c 2 \
  -i eth1 -q 3 -c 3 \
  -Nc --filename xsk_fwd_kern.o
```

In the above example, `xsk_fwd` listens to eth1, and thread 0-3 handling NIC queue 0-3. Each thread sends back the packets it receives.
`xsk_fwd_kern.o` redirects 13333/udp packets to user land via AF_XDP socket and passes the rest to kernel.
We are sending packets by TRex from the other instance in the same subnet.

On the `a1.2xlarge` instance type, the ENA NIC device has only two queues and only one queue is available for XDP.
In this case, the CPU that handling the queue reaches 100% utilization before the NIC bandwidth (10 Gbps) is full.

Similarly, even higher spec instances such as the `c6i.32xlarge` (bandwidth 50 Gbps, 8 queues) can only use half queues for XDP.
I haven't tried it yet, but I think maybe it can't fill the bandwidth.

zeacot, 2023-03-21:
> Hello,
> 
> The ena driver fails to load the xdp object if the combined channels exceeds half of the maximum allowed channel count. Why does the ena driver have this limitation?
> 
> ```
> $ ethtool -i eth1
> driver: ena
> version: 2.7.2g
> firmware-version:
> expansion-rom-version:
> bus-info: 0000:00:06.0
> supports-statistics: yes
> supports-test: no
> supports-eeprom-access: no
> supports-register-dump: no
> supports-priv-flags: yes
> 
> $ ethtool -l eth1
> Channel parameters for eth1:
> Pre-set maximums:
> RX:             0
> TX:             0
> Other:          0
> Combined:       8
> Current hardware settings:
> RX:             0
> TX:             0
> Other:          0
> Combined:       8
> 
> (Use xdp-project sample program)
> $ git clone https://github.com/xdp-project/xdp-tutorial.git
> $ cd xdp-tutorial/basic01-xdp-pass/
> $ git submodule update --init
> $ make
> 
> $ sudo ip link set eth1 mtu 3498
> $ sudo ip link set dev eth1 xdp obj xdp_pass_kern.o sec xdp
> (snip)
> Error: ena: Failed to set xdp program, there is no enough space for allocating XDP queues, Check the dmesg for more info.
> 
> $ dmesg -T
> (snip)
> [Fri Sep 16 14:29:02 2022] ena 0000:00:06.0 eth1: Failed to set xdp program, the Rx/Tx channel count should be at most half of the maximum allowed channel count. The current queue count (8), the maximal queue count (8)
> ```
> 
> I assume that this limit is determined by `ena_xdp_legal_queue_count()`.
> 
> * https://github.com/amzn/amzn-drivers/blob/ena_linux_2.8.0/kernel/linux/ena/ena_xdp.h#L81
> 
> Other NIC drivers, for example the Intel ixgbe driver, can load the xdp module even if all channels are combined.
> 
> ```
> $ ethtool -i enp1s0f0
> driver: ixgbe
> version: 5.15.0-47-generic
> firmware-version: 0x00012b2c
> expansion-rom-version:
> bus-info: 0000:01:00.0
> supports-statistics: yes
> supports-test: yes
> supports-eeprom-access: yes
> supports-register-dump: yes
> supports-priv-flags: yes
> 
> $ ethtool -l enp1s0f0
> Channel parameters for enp1s0f0:
> Pre-set maximums:
> RX:             n/a
> TX:             n/a
> Other:          1
> Combined:       8
> Current hardware settings:
> RX:             n/a
> TX:             n/a
> Other:          1
> Combined:       8
> 
> $ sudo ip link set dev enp1s0f0 xdp obj xdp_pass_kern.o sec xdp
> 
> $ ip l sh enp1s0f0
> 3: enp1s0f0: <BROADCAST,MULTICAST> mtu 1500 xdp qdisc noop state DOWN mode DEFAULT group default qlen 1000
>     link/ether 00:1b:21:bc:04:a2 brd ff:ff:ff:ff:ff:ff
>     prog/xdp id 33
> 
> $ sudo dmesg -T
> (snip)
> [Fri Sep 16 11:21:00 2022] ixgbe 0000:01:00.0: Multiqueue Enabled: Rx Queue count = 8, Tx Queue count = 8 XDP Queue count = 8
> ```
> 
> Our environment is:
> 
> * AWS
>   
>   * AMI: Amazon Linux 2 AMI (HVM) - Kernel 5.10, SSD Volume Type
>   * Instance type: c6i.xlarge - 32xlarge, a1.2xlarge
>   * Driver version: ena 2.7.2g (build from source)
> * Bare metal (for comparison)
>   
>   * OS: Ubuntu 22.04 LTS server
>   * CPU architecture: x86_64
>   * Driver version: ixgbe, included in 5.15.0-47-generic kernel
> 
> Thanks,

So how did you figured out to solve this issue? Did you modify the ENA Driver to remove the Limitation by yourself ? Please post it as me too facing same issue :(

ShayAgros, 2023-04-03:
Sorry for the lack of activity by my side, @zeacot what is the issue you're observing ?

Do you mean that you're unable to reach max BW using AF XDP (I'm assuming generic mode AF XDP and not the native in the driver) ?
Does the same CPU handles both the AF XDP program and the eBPF program ?

davidarinzon, 2023-08-27:
Closing this ticket due to inactivity, please re-open when/if more information can be provided.

closed 2023-08-27 by davidarinzon