amzn/amzn-drivers #245: Significant DPDK performance variance among C-type instance

eecheng87, 2022-10-07:
Hi, all

Recently, I try to port memcached, created by [Seastar](https://github.com/scylladb/seastar), based on DPDK to EC2 instances. I observe there is a significant performance variance among `C` type instance based on x86 CPU (for high performance web servers and high performance computing). I didn't test the instance powered by Arm-based AWS Graviton since Seastar can't be built on arm-based machine.

Here is the environment setup:
* Server: 2xlarge running memcached
* Client: 4xlarge running [memtier_benchmark](https://github.com/RedisLabs/memtier_benchmark)
* Both server and client are in the same security group and cluster placement group
* Attach an additional NIC to server for DPDK (it is also in the same security group)

The server side runs memcached with DPDK: `sudo ./build/release/apps/memcached/memcached --network-stack native --dpdk-pmd --dhcp 0 --host-ipv4-addr 172.31.10.228 --netmask-ipv4-addr 255.240.0.0 --smp 1 --port 11212`

The server side runs vanilla memcached: `sudo ./build/release/apps/memcached/memcached --smp 1 --max-slab-size 100 --network-stack posix --host-ipv4-addr 172.31.10.219 --port 11212
`

The client side runs memtier_benchmark: `./memtier_benchmark -s xx.xx.xx.xx -p 11212 --protocol=memcache_text --clients=16 --threads=16 --ratio=0:1 --key-pattern=R:R --key-minimum=16 --key-maximum=16 --data-size=32 --test-time=4 --hide-histogram`

Here is the experimental result, throughput to varied instance type:

|     | Vanilla | DPDK |
| --- | ------- | ---- |
| C5   | 180k | 580k |
| C5n  | 130k | 650k |
| C6i  | 180k | 360k |

I'm not sure the varience is due to the implementation defect from Seastar or AWS ENA driver. Do you guys have any comment about this experiment?

Also, I have another configuration shows there is a significant degradation. Both server and clinet are running on c6i.8xlarge. And found the throughput of DPDK degrades from 260k to 200k. This is pretty weird result and can't be explained by myself.

Following is the step I build seastar:

Prerequisite
```
sudo apt-get update
sudo apt install git build-essential libpcre3 libpcre3-dev zlib1g zlib1g-dev libssl-dev libgd-dev libxml2 libxml2-dev uuid-dev autoconf automake libtool net-tools python3-pip ninja-build
pip3 install pyelftools
```

Build Seastar
```
git clone https://github.com/scylladb/seastar.git
cd seastar/
sudo ./install-dependencies.sh
vim src/net/dpdk.cc -> replace default_ring_size from 512 to 1024
git submodule update --init
./configure.py --mode=release --enable-dpdk --without-tests
ninja -C build/release
```

Build igb_uio.ko
```
git clone git://dpdk.org/dpdk-kmods
cd dpdk-kmods/linux/igb_uio
make
```

Other setup
```
sudo sh -c "echo 1024 > /sys/kernel/mm/hugepages/hugepages-2048kB/nr_hugepages"

sudo mkdir /mnt/huge
sudo mount -t hugetlbfs nodev /mnt/huge

sudo modprobe uio
sudo insmod /home/ubuntu/dpdk-kmods/linux/igb_uio/igb_uio.ko wc_activate=1
sudo /home/ubuntu/seastar/dpdk/usertools/dpdk-devbind.py --bind=igb_uio ens6
```

Any suggestion will be appreciated! Thanks in advance!

shaibran, 2022-10-11:
Hi, we will look into this and update.
Thanks!

shaibran, 2022-11-02:
can you please share the dpdk version, the instance IDs you used for testing and the time when you ran the tests? I would like to check the logs from aws side

shaibran, 2022-11-02:
Re: _I'm not sure the variance in improvement rate is due to the implementation defect from Seastar or AWS ENA driver. Do you guys have any comment about this experiment_?

The C types you tested are from three different HW generations which have different capabilities on each generation. The variance can be related to the per-queue capabilities differences. With the DPDK it's more likely to hit the maximum.

Re: _ I have another configuration shows there is a significant degradation. Both server and clinet are running on c6i.8xlarge. 
And found the throughput of DPDK degrades from 260k to 200k. This is pretty weird result and can't be explained by myself._

This is likely caused by the higher pressure on the HW from the DPDK side. You can check for ENI metrics/Tx drops values, or you can provide me the details in my previous comment and I can check if there were drops on the ENA side during the test.

shaibran, 2022-11-13:
Hi @eecheng87, the seastar framework uses very old DPDK v19.05 which lacks multiple features required by the 6th generation HW, leading to performance degradation. The seastar framework lacks documentation regarding upgrade to a newer DPDK version.
We were able to reproduce this performance degradation using the DPDK v19.11.13 LTS version, which is the closest one working properly with the latest kernel and gcc, and DTS.
Once we applied the PMD [backports ](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk/backports) from DPDK v21.11 the performance improved.

eecheng87, 2022-11-15:
Hi @shaibran , sorry for late reply and really appreciate your survey. It makes sense that previous DPDK lacks multiple features required by the 6th generation HW. Can you describe more about how to reproduce the improvement? I suppose we can't use customized DPDK for seastar and it uses DPDK v21.08. Also, which patch I should applied? I didn't see matched version for DPDK used by seastar. Thanks!

shaibran, 2022-11-23:
simply apply all patches under [v19.11.10](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk/backports/v19.11.10) backports folder. This should add the features missing from the PMD in that DPDK version

shaibran, 2023-08-10:
Summary of the issue: This is a very old dpdk version which enters the slow path on our 6th generation instance types.
Recommended to apply latest backports to resolve such issue.

closed 2023-08-10 by shaibran