amzn/amzn-drivers #246: DPDK: max number of RX descriptors depends on #threads

mejedi, 2022-10-24:
We've noticed that the maximum number of RX descriptors supported by device depends on the number of DPDK threads.

```
$ dpdk-testpmd -v -l 0,1,2 --in-memory --no-telemetry --file-prefix vpp -a 0000:00:06.0 --log-level=debug --main-lcore=0 -- -i -a
show port info all
```

Reports 16536 as the maximum RX ring size. Now if we go with fewer threads (`-l` option):

```
$ dpdk-testpmd -v -l 0,1 --in-memory --no-telemetry --file-prefix vpp -a 0000:00:06.0 --log-level=debug --main-lcore=0 -- -i -a
show port info all
```

The number is reduced to 2048.

We can see in the source code that the number of EAL threads is passed to the device, which, supposedly sets the maximum ring size accordingly.

```c
static void ena_config_host_info(struct ena_com_dev *ena_dev)
{
    ...

    host_info->num_cpus = rte_lcore_count();
```
https://git.dpdk.org/dpdk-stable/tree/drivers/net/ena/ena_ethdev.c?h=v21.11.2#n444

Our educated guess is that `num_cpus` is a proxy of the instance size, which the device uses to limit itself in tiny instances. However the number of DPDK is NOT a reliable measure of the instance size. Please consider using the number of CPUs instead.

The current logic creates issues in VPP 22.10 as it manages threads itself and doesn't bother telling DPDK the number of threads used. Consequentially, the maximum RX queue size is limited to 2K only.

shaibran, 2022-10-30:
let us look into this

mejedi, 2023-03-11:
Any progress with this one?

shaibran, 2023-03-22:
Hi [mejedi](https://github.com/mejedi),
To summarize the problem:
The number of CPUs is not constant in the application which can vary between executions (controlled through `-l` variables).
Host info sets the number of currently active processors which is the number of available logical cores which is the number of threads. So in the case of VPP during initialization the value of `rte_lcore_count` is smaller than the actual supported number of CPUs on the instance type which limits your RX queue size.

As a temporary mitigation, you can manually modify the PMD [code](https://git.dpdk.org/dpdk-stable/tree/drivers/net/ena/ena_ethdev.c?h=v21.11.2#n444) to any value to enable the maximal queue depth.

For a long-term solution, since in DPDK the number of processors can vary, we will add a task to our backlog to ignore this value from the host information and allow maximum depth for DPDK. However, note that to avoid affecting existing running apps by this change, we will support this from a specific PMD version (of course we will add backports so that apps running on older DPDK versions can opt-in and port this change)

shaibran, 2023-08-10:
Summary: this is not a driver issue and will be handled in the device level (I already opened internal task to track it), however I am unable to share timeline for this.
As a mitigation, a workaround in dpdk code was provided.
I am closing this issue.

shaibran, 2023-10-05:
this will be supported only from future DPDK ENA driver versions (2.8.0 and later) to avoid changing the behavior of already running applications