amzn/amzn-drivers #247: Device statistics exported via ethtool

zeffron, 2022-10-24:
It would be very helpful to us to have the device statistics (those exposed via `rtnl_link_stats64`) also exposed via ethtool. This would allow us to use our tooling that uses ethtool to gather those metrics, reducing the number of metrics scraping tools we need to deploy/support. This would also help us save cycles on our performance critical servers.

Currently, it appears that many of those metrics are already exposed per-queue in ethtool, so it shouldn't be too hard to aggregate them. Some, such as drops, appear to be completely missing from ethtool, even per-queue.

davidarinzon, 2022-10-25:
Hi @zeffron 
Thank you for the suggestion, we will the request and update.

davidarinzon, 2022-11-13:
Hi @zeffron 
Aggregated stats are already available via `ifconfig` and `/cat/proc/dev`, the general guideline is to not duplicate statistics which are already available through other methods.

davidarinzon, 2022-12-11:
Hi @zeffron 
Do you have any further questions?

zeffron, 2022-12-12:
Can you please link to that general guideline? I cannot find it via searching or perusing the kernel documentation, and it seems to be violated by several other major driver vendors (such as Mellanox and Intel).

davidarinzon, 2022-12-14:
Hi @zeffron 
Thank you for your reply, you're welcome to follow-up with me offline at darinzon@amazon.com for additional information.

davidarinzon, 2023-03-13:
As noted above, this is not planned to be supported.

closed 2023-03-13 by davidarinzon