amzn/amzn-drivers #248: RSS Hash Key Query/Update Failed on c5a.8xlarge instances

yangsuli-netapp, 2022-10-25:
Hello,

I have a similar problem as the issue described in https://github.com/amzn/amzn-drivers/issues/169, where RSS Hash Key Query/Update failed on EC2 instances. 

More specifically, we are using DPDK on AWS c5a.8xlarge instances, with ENA drivers. We are unable to successfully establish connections because the static RSS key we use does not match the one which is actually used in the HW. 

We tried to query and update the RSS key using rte_eth_dev_rss_hash_conf_get() and ena_rss_hash_set(), but they both failed. Further tracing revealed that ena_rss_hash_set() failed because the supported_features vector does not have the ENA_ADMIN_RSS_HASH_INPUT bit set, which caused the function to support an OP_NOT_SUPPORED error.  ena_get_rss_hash_key() failed with a similar error, as apparently we cannot retrieve a key unless we have previously set it. 

I am a bit confused though, as per this link (https://doc.dpdk.org/guides/nics/overview.html), RSS Key update would be supported by the ENA driver. So does the above error mean that the underlying NIC hardware does not support such a feature, and we need to change to another EC2 instance type with a NIC that does support this?  Should we not expect RSS key query/update API calls to reliably succeed across different EC2 instances, even though ENA drivers support them? 

Thanks a lot for your help!

SUli

shaibran, 2022-10-30:
Hi, the DPDK documentation you mentioned https://doc.dpdk.org/guides/nics/overview.html refers to the supported features of the latest ENA PMD version.
Please refer to our [RSS documentation](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#113-rss-hash-key-update):
1. you need to use DPDK v21.08 (which contains ENA PMDv2.4.0) or later. Alternatively, you can apply our latest backports which supports modification of the RSS hash key.
2. I recall that the c5a generation used a static key, but I am not sure if it was a HW limitation or just lack of support in the SW. So I suggest to make sure your PMD fully supports the feature before migrate to newer HW generations.

yangsuli-netapp, 2022-10-31:
@shaibran  Thanks for your response and help! Some questions though:

1. I am using DPDK v21.11.2, which I believe contains PMD v2.4.0 or later, so it should support changing RSS key, at lease per the specification. 
2. How do I know if the underlying NIC has a HW limitation for updating RSS, if I am running on an EC2 instance? 

Thanks!

shaibran, 2022-11-02:
@yangsuli-netapp Correct, DPDK v21.11.2 means that the ENA PMD supports the feature.
See [instructions](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#113-rss-hash-key-update) for detecting if HW supports the feature from app.

closed 2022-11-13 by amitbern-aws