amzn/amzn-drivers #251: Compile error with kernel 4.18.0-425.3.1.el8.x86_64 on Rocky 8.7

alekrain, 2022-11-22:
Hello, I received the compile error below on Rocky 8.

OS: Rocky Linux release 8.7 (Green Obsidian)
Kernel Version: 4.18.0-425.3.1.el8.x86_64
ENA Driver Version: 2.8.0

```
make -C /lib/modules/4.18.0-425.3.1.el8.x86_64/build M=/home/rocky/amzn-drivers/kernel/linux/ena modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-425.3.1.el8.x86_64'
  CC [M]  /home/rocky/amzn-drivers/kernel/linux/ena/ena_netdev.o
  CC [M]  /home/rocky/amzn-drivers/kernel/linux/ena/ena_ethtool.o
/home/rocky/amzn-drivers/kernel/linux/ena/ena_ethtool.c:1220:20: error: initialization of ‘void (*)(struct net_device *, struct ethtool_ringparam *, struct kernel_ethtool_ringparam *, struct netlink_ext_ack *)’ from incompatible pointer type ‘void (*)(struct net_device *, struct ethtool_ringparam *)’ [-Werror=incompatible-pointer-types]
  .get_ringparam  = ena_get_ringparam,
                    ^~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:318: /home/rocky/amzn-drivers/kernel/linux/ena/ena_ethtool.o] Error 1
make[1]: *** [Makefile:1585: _module_/home/rocky/amzn-drivers/kernel/linux/ena] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-425.3.1.el8.x86_64'
make: *** [Makefile:64: all] Error 2
```

rjhornsby, 2022-12-03:
@akiyano thanks for the patch. worked great, and unblocks some packer->AMI work. appreciate you making that available to us while the next release is still in progress.

davidarinzon, 2023-01-16:
Hi @rjhornsby 
Can you please check the latest release? It should resolve your issue.

closed 2023-01-19 by davidarinzon