amzn/amzn-drivers #252: DPDK ENA PMD can not receive Packets on Rx

liamwei, 2022-11-24:
Hi All,

I an running testpmd using DPDK 19.11.13
It print the following info: show port info all

Link status: up
Link speed: 0 Mbps
Link duplex: full-duplex

the speed always is 0 Mbps. And It can not receive the packet on this port.

Please help me
Thanks

shaibran, 2022-11-24:
Hi [liamwei](https://github.com/liamwei) We were testing the driver v19.11.13 with\out the backports recently and no Rx regression was observed so we suspect some misconfiguration issue here. can you please provide:
1. exact command used for starting the testpmd
2. list of commands being passed in the testpmd interactive mode
3. instance type (for reproduction in house)

best regards

liamwei, 2022-11-24:
> Hi [liamwei](https://github.com/liamwei) We were testing the driver v19.11.13 with\out the backports recently and no Rx regression was observed so we suspect some misconfiguration issue here. can [...]

hi shaibran,
1. ./testpmd -c 7 -n 4 -- -i
2. after start the testpmd, then input the show port info all
3. for the instance information, I will get from my colleague

I think it may be that some configurations are missing, which makes it can not receive packets. Could you give me some suggestion to check the configuration?

Thanks you very much.

shaibran, 2022-11-24:
first command you are typing is the 'show port info all', but you should first type 'start' in order to start any IO processing in the interactive mode

liamwei, 2022-11-24:
> correct me if I got it wrong, but it seems that the first command you are typing is the 'show port info all', but you should first type 'start' in order to start any IO processing in the interactive mode

hi shaibran,

I have input the "start", then "show port info all". It the same as before.
Link status: up
Link speed: 0 Mbps
Link duplex: full-duplex

And I input the "show port stats all", there are very few data packets.  In fact, there are many data packets on this network card. Seems to be filtered out by something.
  ######################## NIC statistics for port 0  ########################
  RX-packets: 20         RX-missed: 0          RX-bytes:  1272
  RX-errors: 0
  RX-nombuf:  0
  TX-packets: 16         TX-errors: 0          TX-bytes:  1008

  Throughput (since last show)
  Rx-pps:            0          Rx-bps:          144
  Tx-pps:            0          Tx-bps:          144
  ############################################################################

Could you give some advice?

Thank you very much.

liamwei, 2022-11-25:
hi All,

My server is redhat 7.9, and the dpdk is v19.11.13.
My ena driver version is version: 2.0.3K

please help me, thanks.

semihalf-krawczyk-michal, 2022-11-25:
Hi @liamwei ,

Could you please take a look at this issue #102 and check if you're doing everything correctly as discussed over there and confirm?

The link speed 0 is a valid value, as described [in the ENA PMDs FAQ](https://github.com/amzn/amzn-drivers/blob/master/userspace/dpdk/README.md#13-general-faq).

Thanks,
Michał

liamwei, 2022-11-25:
hi @semihalf-krawczyk-michal 

Thank you very much.

Another question, this version can not enable promiscuous mode.
EAL: Error - exiting with code: 1
  Cause: Could not enable promiscuous mode for port0: Operation not supported
Is it normal?

thanks.

semihalf-krawczyk-michal, 2022-11-25:
The ENA doesn't support the promiscuous mode, as on AWS it's rather impossible to implement it, as the whole AWS VPC works on L3 (please see [this comment](https://github.com/amzn/amzn-drivers/issues/172#issuecomment-845131116) for more details).

You have to use valid Ethernet and IP address for the ENIs, otherwise they'll be filtered before reaching the ENI.

liamwei, 2022-11-25:
hi @semihalf-krawczyk-michal 

you mean if I use dpdk, we must enable kni, then config the IP, or It can not receive the packet. right?

thanks.

liamwei, 2022-11-25:
hi @semihalf-krawczyk-michal 

I mean, if I start the dpdk and do not configure IP through the kni, how can I route the external data packets to my network card.
Do I need to connect directly to the peer network card, or do I have other device as proxy?

thanks.

liamwei, 2022-11-29:
hi all,

My instance type is c6i.xlarge
instance ID is i-07e2cd57cb6d613a0

thanks.

semihalf-krawczyk-michal, 2022-11-29:
Hi @liamwei 

> I mean, if I start the dpdk and do not configure IP through the kni, how can I route the external data packets to my network card.
> Do I need to connect directly to the peer network card, or do I have other device as proxy?

The routing is happening in the networking infrastructure level. When you're allocating interface for your instance you can see that it has already IP address attached, so it knows what packets can be received by the interface.

You don't need KNI or assign the IP for the hardware to receive the packets which are addressed properly. Changing IP in the KNI won't change the VPC network infrastructure, it should be rather the other way around - you should first check the interface IP address in your AWS console, and then assign it to the KNI.

If you require any other assistance regarding DPDK and the AWS, please reach me and Shai Brandes via email at mk@semihalf.com and shaibran@amazon.com

Thanks,
Michał

shaibran, 2023-08-10:
configuration issue which we resolved via mail thread offline. closing, feel free to reopen if you have further questions

closed 2023-08-10 by shaibran