amzn/amzn-drivers #253: kni example fail

liamwei, 2022-11-28:
Hi All,

I am running kni example using DPDK 19.11.13

After startting the kni example, and config the ip on this vEth0. 

this is the result of ifconfig.
vEth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 9001
        inet 192.168.1.133  netmask 255.255.255.224  broadcast 192.168.1.159
        ether 02:28:82:f6:98:8e  txqueuelen 1000  (Ethernet)
        RX packets 341  bytes 21834 (21.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

There is no RUNNING in the flags. Is it normal?
And It can not receive the packet.

Please help me
Thanks

liamwei, 2022-11-29:
hi all,

My instance type is c6i.xlarge
instance ID is i-07e2cd57cb6d613a0

thanks.

liamwei, 2022-11-29:
hi all,

Any update?

thanks.

semihalf-krawczyk-michal, 2022-11-29:
Hi @liamwei 

Could you please double check the things below and update us?

* Are you sure you've bound the ENA device you're supposed to use with the KNI to the DPDK driver?
* I think that the KNI IP should be the same as the address of the ENI in the VPC configuration you're going to use for testing - could you change it to the IP assigned to the ENI and check if it helps? It should be the IP of the interface you can see before you're binding the ENA device to the igb_uio or vfio_pci (assigned by the DHCP server)
* How are you generating the Rx flow? What commands or programs are you using? Is it possible for you to share some instructions on how you're doing it with us?

I suggest to continue this discussion over email. Please email me and Shai Brandes with further findings at: mk@semihalf.com and shaibran@amazon.com

Thanks,
Michał

closed 2023-08-10 by shaibran