amzn/amzn-drivers #254: DPDK on AWS EC2 C5 instance, interfaces only sends traffic but not receiving

gianlucacvalastro, 2022-12-13:
Hi,

I'm using a C5 instance, with two private interfaces ens6 and ens7. Binding these interfaces with igb_uio and using dpdk applications when we try to send traffic from one interface to another, each interface is able to send traffic but no traffic is received on both interfaces. Trying to debug with dpdk pkt-gen it seems that only arp packet in broadcast are received, while trying to forward traffic between two interfaces no traffic is received on both interfaces. 

Thank you for your availability and for your support.

Best regards

shaibran, 2022-12-15:
Hi, since there is no incoming traffic this is likely some configuration issue.
1. Please make sure you followed the PMD readme https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk and check the ACL and security groups being configured which can lead to incoming packet drop.
2. Please share dpdk, pktgen and pmd versions being used and the command sequence you used to bring up and run the environment.

gianlucacvalastro, 2022-12-15:
Thanks @shaibran 

1. I followed the commands in the PMD readme to bring up the environment. For the ACL and security groups, we are checking it, thank you for this suggestion. I will let you know about this as quickly as possible.
2. We faced this no incoming traffic issue also using another public forwarding application, not only pktgen. I tried to use different version of dpdk, pktgen and pmd versions, always facing this issue.

I noted that regardless of the commands used for pktgen, with correct MAC and IP addresses, the pktgen output shows that only multicast/ARP packets are received. 

Thank you for your availability and for your support.

Best regards

closed 2023-08-10 by shaibran