amzn/amzn-drivers #256: Does ENA DPDK pmd driver do RSS hash for IPSEC ESP packet?

JasonJiang-VM, 2022-12-16:
I'm using ENA driver 2.0.2, I would like to check whether IPSEC ESP packet can be RSS fan out to different RSS queue or not.
First of all, I can get RSS key and RSS function, even by adding the rss_hash_conf_get function:
 240 static const struct eth_dev_ops ena_dev_ops = {
 241         .dev_configure        = ena_dev_configure,
 242         .dev_infos_get        = ena_infos_get,
 243         .rx_queue_setup       = ena_rx_queue_setup,
 244         .tx_queue_setup       = ena_tx_queue_setup,
 245         .dev_start            = ena_start,
 246         .dev_stop             = ena_stop,
 247         .link_update          = ena_link_update,
 248         .stats_get            = ena_stats_get,
 249         .xstats_get_names     = ena_xstats_get_names,
 250         .xstats_get           = ena_xstats_get,
 251         .xstats_get_by_id     = ena_xstats_get_by_id,
 252         .mtu_set              = ena_mtu_set,
 253         .rx_queue_release     = ena_rx_queue_release,
 254         .tx_queue_release     = ena_tx_queue_release,
 255         .dev_close            = ena_close,
 256         .dev_reset            = ena_dev_reset,
 257         .rss_hash_conf_get    = ena_dev_rss_hash_conf_get,  <<<<
 258         .reta_update          = ena_rss_reta_update,
 259         .reta_query           = ena_rss_reta_query,
 260 };

But from the output, it seems it complained and return :
ena_dev_rss_hash_conf_get(): : Failed to get hash function -95( not supported).
According to this, I have two questions: 
- If I would like to get the rss hash key, do I need to upgrade the driver version to >2.4.0?
- Even after 2.4.0, can ESP packet can be RSS and hashed to different rx queue? Or it is treated as IP packets, and do nothing for this. From the README, it seems ENA RSS only works for TCP/UDP, not others( IP, or ESP here).

shaibran, 2022-12-18:
The static key used prior to v2.4.0 is available [here](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#11-rss-support). In v2.4.0 and later the application sets its own RSS hash key.
The HW calculates the hash only for UDP/TCP, and all other types will be received on queue#0.

JasonJiang-VM, 2022-12-19:
Thanks shaibran for your input. Appreciate it.
I did some testing with IPSEC ESP packets. In my testing with IPSEC ESP, we have 4 pairs IPSEC tunnels between two EC2(c5n) instances, it seems that ESP packets flow can be received by different queues, not only the queue0, we saw 3 ~ 4 PCPUs being used on the receiving side. If IPSEC ESP is only processed on queue0, that should be only 1 PCPU used. Could you please help me to clarify this?

JasonJiang-VM, 2022-12-19:
And BTW, I checked ENA specification 1.0 document, about RSS said: 
•	Receive-Side Scaling (RSS) and configurable indirection table based on 4-tuple hash for TCP and UDP packets and 2-tuple for non-TCP/UDP as well as IP Fragments 
•	CRC32 and Toeplitz hash (16-bit and 32-bit) for RSS hash

Do you mean it is not true for DPDK driver(2-tuple for non-TCP/UDP as well as IP Fragments)? Is it possible to enable it for dpdk driver as well?

Thanks
Jason

shaibran, 2022-12-20:
Hi Jason, 2-tuple is indeed used by the HW for non-TCP/UDP packets' hash calculation. I will update the PMD documentation accordingly.

JasonJiang-VM, 2022-12-20:
Thanks shaibran. It's cool~
For a further question, regarding the hash of IP packets, there is a script to get the hash value:
https://github.com/amzn/amzn-ec2-ena-utilities/blob/main/ena-toeplitz/toeplitz_calc.py
I learned from you that there is a static hash key: be:ac:01:fa:6a:42:b7:3b:80:30:f2:0c:77:cb:2d:a3:ae:7b:30:b4:d0:ca:2b:cb:43:a3:8f:b0:41:67:25:3d:25:5b:0e:c2:6d:5a:56:da
that script request a source port and destination port, should I make them all 0 to get the hash packets:
-bash-4.2$ ./toeplitz_calc.py -t 10.117.8.100 -R 0 -r 10.117.9.121 -T 0
Sending traffic from 10.117.8.100:0 to 10.117.9.121:0
to an instance with a default RSS key that cannot be changed

Should result in the hash for all drivers:              0x55505550 (RSS table entry : 80) <<<<<<
-bash-4.2$ ./toeplitz_calc.py -t 10.117.8.100 -R 0 -r 10.117.9.121 -T 0 -k be:ac:01:fa:6a:42:b7:3b:80:30:f2:0c:77:cb:2d:a3:ae:7b:30:b4:d0:ca:2b:cb:43:a3:8f:b0:41:67:25:3d:25:5b:0e:c2:6d:5a:56:da
Sending traffic from 10.117.8.100:0 to 10.117.9.121:0
to an instance which supports changing the key

Should result in the following hash for each driver:
DPDK                                                    0x760faaaf (RSS table entry: 47) <<<<< 
FreeBSD                                                 0x760faaaf (RSS table entry: 47)
Linux (before setting the key with ethtool)             0xebfb4c38 (RSS table entry: 56)
Linux (after setting the key with ethtool)              0x1404b3c7 (RSS table entry: 71)
Windows                                                 0x1404b3c7 (RSS table entry: 71)

So please help to confirm:  for a IP packets, when get the hash with toeplitz hash, how to input the source port and destination port(all 0?), and why the result is different if I input the default hash key: be:ac:01:fa:6a:42:b7:3b:80:30:f2:0c:77:cb:2d:a3:ae:7b:30:b4:d0:ca:2b:cb:43:a3:8f:b0:41:67:25:3d:25:5b:0e:c2:6d:5a:56:da

Thanks a million,
Jason

JasonJiang-VM, 2022-12-23:
I did some testing for the hash of IP packets. I uses a IPSEC tunnel to transmit packets:
Local IP:         10.100.0.40 
Remote IP:     10.100.0.36
-bash-4.2$ ./toeplitz_calc.py -t 10.100.0.40 -T 0 -r 10.100.0.36 -R 0
Sending traffic from 10.100.0.40:0 to 10.100.0.36:0
to an instance with a default RSS key that cannot be changed

Should result in the hash for all drivers:              0xa7b0a7b (RSS table entry : 123)
-bash-4.2$ ./toeplitz_calc.py -t 10.100.0.36 -T 0 -r 10.100.0.40 -R 0
Sending traffic from 10.100.0.36:0 to 10.100.0.40:0
to an instance with a default RSS key that cannot be changed

Should result in the hash for all drivers:              0xa7b0a7b (RSS table entry : 123)
index=123, queue=3
So on both side, it should use queue #3 to receive packets. But per my testing, is is queue #1 on one side, queue #0 on the other side. 
And I use another IP pairs, it showed the same result, the toeplitz_calc.py give one results for this IP pair, but the receive queues are different on both side, one is queue#2, the other is queue#0.
-bash-4.2$ ./toeplitz_calc.py -t 10.100.0.42 -T 0 -r 10.100.0.41 -R 0
Sending traffic from 10.100.0.42:0 to 10.100.0.41:0
to an instance with a default RSS key that cannot be changed

Should result in the hash for all drivers:              0x10f110f1 (RSS table entry : 113)
index=113, queue=1

So here is question, for a IP packet, how ENA driver does the hash? Any comments are appreciated.
Thanks
Jason

shaibran, 2022-12-25:
ENA PMD does not calculate anything, it configures the HW which in turn performs the actual calculation.
The script you mentioned simulates the expected behavior per each driver configuration.
Drivers configures the HW differently (DPDK/freeBSD follow the standard implementation opposed to Linux)

ShayAgros, 2023-01-02:
The script's output is further discussed in amzn/amzn-ec2-ena-utilities#1

shaibran, 2023-01-09:
resolving this one, see https://github.com/amzn/amzn-ec2-ena-utilities/issues/1 for the details regarding the script

closed 2023-01-09 by amitbern-aws