amzn/amzn-drivers #257: Failed to recognize ena on Ubuntu 20.04 LTS

sakaia, 2023-01-30:
I tried to execute DPDK on Amazon ENA but failed with following error (No probed ethernet devices).
Is there any suggestions for this? (dpdk-devbind recognized ena devices on c5 instance on ubuntu 20.04LTS also failed on 22.04LTS)
If the DPDK version is not appropriate, would you provide me a appropriate version?
I am looking around following document but DPDK 22.11 is not described.
  https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk

testpmd execution
===
sudo /usr/local/bin/dpdk-testpmd
EAL: Detected CPU lcores: 2
EAL: Detected NUMA nodes: 1
EAL: Detected static linkage of DPDK
EAL: Multi-process socket /var/run/dpdk/rte/mp_socket
EAL: Selected IOVA mode 'PA'
TELEMETRY: No legacy callbacks, legacy socket not created
testpmd: No probed ethernet devices
testpmd: create a new mbuf pool <mb_pool_0>: n=155456, size=2176, socket=0
testpmd: preferred mempool ops selected: ring_mp_mc
Done
No commandline core given, start packet forwarding
io packet forwarding - ports=0 - cores=0 - streams=0 - NUMA support enabled, MP allocation mode: native

  io packet forwarding packets/burst=32
  nb forwarding cores=1 - nb forwarding ports=0
Press enter to exit

Telling cores to stop...
Waiting for lcores to finish...

  +++++++++++++++ Accumulated forward statistics for all ports+++++++++++++++
  RX-packets: 0              RX-dropped: 0             RX-total: 0
  TX-packets: 0              TX-dropped: 0             TX-total: 0
  ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

Done.

Bye...
===

Compiling procedure

===
sudo apt update
sudo apt install -y meson pip net-tools jq
sudo pip install pyelftools
wget https://fast.dpdk.org/rel/dpdk-22.11.1.tar.xz
tar xf dpdk-22.11.1.tar.xz
cd dpdk-stable-22.11.1/
meson setup build
cd build
ninja
sudo ninja install
sudo ldconfig
sudo /usr/local/bin/dpdk-hugepages.py -p 2M --setup 1G
sudo /usr/local/bin/dpdk-hugepages.py --show
Node Pages Size Total
0    512   2Mb    1Gb

sudo /usr/local/bin/dpdk-devbind.py --status

Network devices using kernel driver
===================================
0000:00:05.0 'Elastic Network Adapter (ENA) ec20' if=ens5 drv=ena unused= *Active*
0000:00:06.0 'Elastic Network Adapter (ENA) ec20' if=ens6 drv=ena unused=
0000:00:07.0 'Elastic Network Adapter (ENA) ec20' if=ens7 drv=ena unused=
===

shaibran, 2023-01-31:
Hi, let me check this and I will update

shaibran, 2023-01-31:
Hi [sakaia](https://github.com/sakaia).
ENA PMD did not add any changes to DPDK22.11, so you can use DPDK22.07.
Ubuntu Server 20.04 LTS AMI changed the way the vfio-pci is being distributed. Instead of releasing it as a module, it's being built-in the kernel. See mitigation in the ENA PMD [documentation](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#622-vfio-pci).
I hope this answers your query.

shaibran, 2023-02-02:
please see [quick start guide](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#4-quick-start-guide) for the proper sequence. you must bind to a specific port. In the case of Ubuntu Server 20.04 LTS AMI, this requires the additional step mentioned above in case you are using vfio-pci.

sakaia, 2023-02-03:
> Hi [sakaia](https://github.com/sakaia). ENA PMD did not add any changes to DPDK22.11, so you can use DPDK22.07. Ubuntu Server 20.04 LTS AMI changed the way the vfio-pci is being distributed. Instead [...]

Thank you.
I am finally succeeded running testpmd with recognizing ENA.
I take an option for using linux-image-5.4.0-1060-aws for Ubuntu 20.04 LTS.　

I have one question on this installation procedure.
During the installation I need to edit from linux-image-unsigned to linux-image.
Without this, I cannot download the linux kernel source code.
Is is appropriate operation?

- dpdk/ena: Adjust vfio script to the apt behavior
  - https://github.com/amzn/amzn-drivers/commit/6f9618f85ce76a9c67bf6879581b72a4e5228c61

shaibran, 2023-02-05:
yes, this is the direct download procedure

closed 2023-02-06 by sakaia