amzn/amzn-drivers #260: Packet loss goes to 98%  between c5.xlarge and c5.xlarge

sakaia, 2023-02-09:
I try to transfer the packet by using testpmd. But it goes to 98% packet loss.
Is there any idea to reduce the packet loss? 

Network Configuration is follows. 

- fj-dpdksakaia6 (c5.xlarge)
  - 192.168.139.101 06:0d:37:08:8d:d1 (general)
  - 192.168.142.144 06:04:3e:0a:4d:8f (DPDK)
- fj-dpdksakaia7
  - 192.168.172.231 06:88:3c:84:a0:eb (general)
  - 192.168.188.36  06:75:b2:5e:5b:c3 (DPDK)

Sender testpmd

```
sudo /usr/local/bin/dpdk-testpmd -- -i --no-numa --tx-ip=192.168.142.144,192.168.188.36 --stats-period 10 --eth-peer=0,06:75:b2:5e:5b:c3  --forward-mode=txonly
```

Receiver testpmd

```
sudo /usr/local/bin/dpdk-testpmd -- -i --forward-mode=rxonly --stats-period 10 --eth-peer=0,06:04:3e:0a:4d:8f --stats-period 10
```

Sender output

```
EAL: Detected CPU lcores: 4
EAL: Detected NUMA nodes: 1
EAL: Detected static linkage of DPDK
EAL: Multi-process socket /var/run/dpdk/rte/mp_socket
EAL: Selected IOVA mode 'PA'
EAL: VFIO support initialized
EAL: Using IOMMU type 8 (No-IOMMU)
EAL: Probe PCI driver: net_ena (1d0f:ec20) device: 0000:00:06.0 (socket -1)
TELEMETRY: No legacy callbacks, legacy socket not created
Interactive-mode selected
Set txonly packet forwarding mode
testpmd: create a new mbuf pool <mb_pool_0>: n=171456, size=2176, socket=0
testpmd: preferred mempool ops selected: ring_mp_mc

Warning! port-topology=paired and odd forward ports number, the last port will pair with itself.

Configuring Port 0 (socket 0)
Port 0: 06:04:3E:0A:4D:8F
Checking link statuses...
Done
Error during enabling promiscuous mode for port 0: Operation not supported - ignore
testpmd> start tx_first
txonly packet forwarding - ports=1 - cores=1 - streams=1 - NUMA support disabled, MP allocation mode: native
Logical Core 1 (socket 0) forwards packets on 1 streams:
  RX P=0/Q=0 (socket 0) -> TX P=0/Q=0 (socket 0) peer=06:75:B2:5E:5B:C3

  txonly packet forwarding packets/burst=32
  packet len=64 - nb packet segments=1
  nb forwarding cores=1 - nb forwarding ports=1
  port 0: RX queue number: 1 Tx queue number: 1
    Rx offloads=0x0 Tx offloads=0x18000
    RX queue: 0
      RX desc=0 - RX free threshold=0
      RX threshold registers: pthresh=0 hthresh=0  wthresh=0
      RX Offloads=0x0
    TX queue: 0
      TX desc=0 - TX free threshold=0
      TX threshold registers: pthresh=0 hthresh=0  wthresh=0
      TX offloads=0x18000 - TX RS bit threshold=0
testpmd> stop
Telling cores to stop...
Waiting for lcores to finish...

  ---------------------- Forward statistics for port 0  ----------------------
  RX-packets: 0              RX-dropped: 0             RX-total: 0
  TX-packets: 5288929        TX-dropped: 231793148     TX-total: 237082077
  ----------------------------------------------------------------------------

  +++++++++++++++ Accumulated forward statistics for all ports+++++++++++++++
  RX-packets: 0              RX-dropped: 0             RX-total: 0
  TX-packets: 5288929        TX-dropped: 231793148     TX-total: 237082077
  ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

Done.
testpmd> quit

Stopping port 0...
Stopping ports...
Done

Shutting down port 0...
Closing ports...
ena_rx_queue_release(): Rx queue 0:0 released
ena_tx_queue_release(): Tx queue 0:0 released
Port 0 is closed
Done

Bye...
```

Receiver output

```
EAL: Detected CPU lcores: 4
EAL: Detected NUMA nodes: 1
EAL: Detected static linkage of DPDK
EAL: Multi-process socket /var/run/dpdk/rte/mp_socket
EAL: Selected IOVA mode 'PA'
EAL: VFIO support initialized
EAL: Using IOMMU type 8 (No-IOMMU)
EAL: Probe PCI driver: net_ena (1d0f:ec20) device: 0000:00:06.0 (socket -1)
TELEMETRY: No legacy callbacks, legacy socket not created
Interactive-mode selected
Set rxonly packet forwarding mode
Warning: NUMA should be configured manually by using --port-numa-config and --ring-numa-config parameters along with --numa.
testpmd: create a new mbuf pool <mb_pool_0>: n=171456, size=2176, socket=0
testpmd: preferred mempool ops selected: ring_mp_mc

Warning! port-topology=paired and odd forward ports number, the last port will pair with itself.

Configuring Port 0 (socket 0)
Port 0: 06:75:B2:5E:5B:C3
Checking link statuses...
Done
Error during enabling promiscuous mode for port 0: Operation not supported - ignore
testpmd> start
rxonly packet forwarding - ports=1 - cores=1 - streams=1 - NUMA support enabled, MP allocation mode: native
Logical Core 1 (socket 0) forwards packets on 1 streams:
  RX P=0/Q=0 (socket 0) -> TX P=0/Q=0 (socket 0) peer=06:04:3E:0A:4D:8F

  rxonly packet forwarding packets/burst=32
  nb forwarding cores=1 - nb forwarding ports=1
  port 0: RX queue number: 1 Tx queue number: 1
    Rx offloads=0x0 Tx offloads=0x18000
    RX queue: 0
      RX desc=0 - RX free threshold=0
      RX threshold registers: pthresh=0 hthresh=0  wthresh=0
      RX Offloads=0x0
    TX queue: 0
      TX desc=0 - TX free threshold=0
      TX threshold registers: pthresh=0 hthresh=0  wthresh=0
      TX offloads=0x18000 - TX RS bit threshold=0
testpmd> stop
Telling cores to stop...
Waiting for lcores to finish...

  ---------------------- Forward statistics for port 0  ----------------------
  RX-packets: 5090000        RX-dropped: 0             RX-total: 5090000
  TX-packets: 0              TX-dropped: 0             TX-total: 0
  ----------------------------------------------------------------------------

  +++++++++++++++ Accumulated forward statistics for all ports+++++++++++++++
  RX-packets: 5090000        RX-dropped: 0             RX-total: 5090000
  TX-packets: 0              TX-dropped: 0             TX-total: 0
  ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

Done.
```

shaibran, 2023-02-12:
can you please share the region, instance IDs and the time you executed this run? I can check the stats from EC2 side

sakaia, 2023-02-13:
> can you please share the region, instance IDs and the time you executed this run? I can check the stats from EC2 side

Thank you for your consideration. My test environment is follows.

The availability region is 
- ap-northeast-1a

The instance ids are follows
- i-0dd47f93aa36b87ce
- i-0328f54404754b538

The time is around 
Feb 9 2023 8:00 PM GMT+9

sakaia, 2023-02-13:
P.S. 
This packet loss also occured on t3.xlarge<=> c5.xlarge.
And It is also occured on testpmd => tcpdump.

shaibran, 2023-02-13:
thanks, I will look into the stats on ec2 and update

shaibran, 2023-02-13:
I did not see any issues for the transmitter i-0328f54404754b538 (specific adapter eni-01e55625dfbf87d27)
Testpmd txonly mode seems to flood the queues (TX-dropped statistics is the amount of packets that couldn't be submitted to tx queue because it full).

sakaia, 2023-02-14:
Thank you for your investigation

This problem still continues even if I change the instance from c5,xlarge to I try to use m5n.8xlarge.

And its throughput is 60MiB/s (even in m5n.xlarge).
Is this currentl ENA performance?(AWS Document says 25Gbps for m5n.xlarge)

Instance ID is i-0f738fff438325096

For testpmd side
```
sudo /usr/local/bin/dpdk-testpmd -a 0000:00:06.0 -- -i --no-numa --tx-ip=192.168.190.3,192.168.144.30 --stats-period 10 --eth-peer=0,06:21:f5:b6:9c:8b --forward-mode=txonly
```

For tcpdump side,

```
sudo tcpdump -w - |pv -bert >/dev/null
```

Output is follows

```
tcpdump: listening on ens5, link-type EN10MB (Ethernet), capture size 262144 bytes
^C33GiB 0:01:05 [58.1MiB/s]
45175956 packets captured
45177370 packets received by filter
0 packets dropped by kernel
```

sakaia, 2023-02-14:
By setting following command, the performance is boosted to 961MiB/s on m5n.8xlarge (i-0f738fff438325096).
But this performance is less than 25Gbps. It is roughly 1/3 of the hardware performance (8Gbps/25Gbps). 

```
set txpkts 2176
``` 

```
$ sudo tcpdump -w - |pv -bert >/dev/null
tcpdump: listening on ens5, link-type EN10MB (Ethernet), capture size 262144 bytes
 352GiB 0:09:49 [ 961MiB/s]
```

AWSNB, 2023-02-14:
Hi

The .xlarge instances can get burst their network bandwidth to 10Gbps for a medium duration, but the baseline guaranteed bandwidth is less than 10Gbps
(described here https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/monitoring-network-performance-ena.html)

As for the results on m5n.8xlarge:  what total instance bandwidth is 25Gbps,  each single flow (defined as TCP tuple or UDP tuple)  as described in details here:
https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/ec2-instance-network-bandwidth.html

sakaia, 2023-02-14:
Thank you for your description.

I try to do iperf3 between m5n.8xlarge instances (i-0f738fff438325096,  i-0e4deb33432883bbc),
instead of DPDK.

But I cannot get 25Gbps.(around 9Gbps)
This is a expected result?

server

```
iperf3 -s
```

client
```
iperf3 -c 192.168.144.30
```

```
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 192.168.175.73, port 45684
[  5] local 192.168.144.30 port 5201 connected to 192.168.175.73 port 45696
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  1.06 GBytes  9.13 Gbits/sec
[  5]   1.00-2.00   sec  1.11 GBytes  9.53 Gbits/sec
[  5]   2.00-3.00   sec  1.11 GBytes  9.53 Gbits/sec
[  5]   3.00-4.00   sec  1.11 GBytes  9.53 Gbits/sec
[  5]   4.00-5.00   sec  1.11 GBytes  9.53 Gbits/sec
[  5]   5.00-6.00   sec  1.11 GBytes  9.53 Gbits/sec
[  5]   6.00-7.00   sec  1.11 GBytes  9.53 Gbits/sec
[  5]   7.00-8.00   sec  1.11 GBytes  9.53 Gbits/sec
[  5]   8.00-9.00   sec  1.11 GBytes  9.53 Gbits/sec
[  5]   9.00-10.00  sec  1.11 GBytes  9.53 Gbits/sec
[  5]  10.00-10.04  sec  48.1 MBytes  9.53 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.04  sec  11.1 GBytes  9.49 Gbits/sec                  receiver
```

akiyano, 2023-02-14:
To get maximum performance from iperf3 you need to use multiple streams.
Please use the following client command:

`iperf3 -c 192.168.144.30 -P8`

This will create 8 streams. You can change the 8 to whichever number works best for your use case of course.

amitbern-aws, 2023-02-14:
sakaia,

You are running a single-flow traffic in the same cluster placement group, you should achieve up to 10 Gbps.
(https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/ec2-instance-network-bandwidth.html)

Add more streams in parallel to reach to 25 Gbps

`iperf3 -c 192.168.144.30 -P 32`

Thanks

sakaia, 2023-02-15:
Thank you, I

I am checking -Pn with various number. And I am getting around 90% and 99% compared to an ideal value.

> To get maximum performance from iperf3 you need to use multiple streams. Please use the following client command:
> 
> `iperf3 -c 192.168.144.30 -P8`
> 
> This will create 8 streams. You can change the 8 to whichever number works best for your use case of course.

おおむね以下の感じで出ました。-Pの数をぎりぎりにすると、99%まで近づくみたいです。

```
$ iperf3 -c 192.168.144.30 -P3
```

```
[SUM]   0.00-10.00  sec  26.4 GBytes  22.7 Gbits/sec    0             sender
[SUM]   0.00-10.03  sec  26.4 GBytes  22.6 Gbits/sec                  receiver
```

```
$ iperf3 -c 192.168.144.30 -P8
```
```
[SUM]   0.00-10.00  sec  28.9 GBytes  24.8 Gbits/sec    0             sender
[SUM]   0.00-10.04  sec  28.9 GBytes  24.7 Gbits/sec                  receiver
```

sakaia, 2023-02-15:
Then I have two questions for this issue.

1. Single flow performance for dpdk-testpmd/tcpdump is slower than iperf3.
    This is correct understanding?
2. Is there any option to send parallel flow for testpmd?

sakaia, 2023-02-15:
> testpmd is just an open source example application provided and maintained by the DPDK community which shows the usage of the generic PMD API and its network stack is not optimized for performance, for testpmd please refer to the [command line options](https://doc.dpdk.org/guides/testpmd_app_ug/run_app.html#testpmd-command-line-options)

Thank you for  your comment

Which software I should use for simple testing？

shaibran, 2023-02-15:
[ENA DTS](https://github.com/amzn/amzn-ec2-ena-utilities/tree/main/ena-dts) better demonstrate AWS ENA networking device capabilities

sakaia, 2023-02-16:
Thank you for your suggestion.

I try to use following tool. But it seems to be too old (2 years passed from last changes and it uses python2)
Is there any good tool? 

> [ENA DTS](https://github.com/amzn/amzn-ec2-ena-utilities/tree/main/ena-dts) better demonstrate AWS ENA networking device capabilities

shaibran, 2023-02-16:
we are about to release a new version to our GitHub, but it still requires few modifications.

shaibran, 2023-08-10:
ENA performance benchmarking application is under construction. It requires stabilization and migrating it to python 3 and will be released to ENA github in the next few months, however, we cannot provide estimated completion date.
I am closing this issue

closed 2023-08-10 by shaibran