amzn/amzn-drivers #261: Ethertype 0xAEFE (ECPRI) support for AWS EC2 instances

gianlucacvalastro, 2023-02-14:
Dear all,

I'm a newbie with AWS EC2 instances. Running dpdk applications on a C5 instance, with ENA private interfaces, binded with igb_uio, is it possible to receive ethernet frame with Ethertype 0xAEFE (ECPRI)? Do AWS ENA interfaces support this? If not, is it possible to enable this capability on ENA interfaces?    

Thank you for your availability and for your support.

Best regards

gianlucacvalastro, 2023-02-15:
Any update about this?
Thanks

nafeabshara, 2023-02-15:
The underlying EC2 network is all IP based and not layer2.   So the only traffic running in your. vPC network and to the ENA interfaces is IPv4/iPv6/ARP and those are the only 3 ether type you could seeEC2 network/ENA do support eCPRI-over-IP and it a popular use caseSent from my iPhone

gianlucacvalastro, 2023-02-15:
Dear @nafeabshara thanks for your detailed and fast response. Unfortunately, we have to test a dpdk based application that allows only ecpri over ethernet and not over ip/udp. So, according to your response, in our case, there is not the possibility to test this application on a C5 instance, right?

Thank you for your availability and for your support.

Best regards

shaibran, 2023-03-12:
@gianlucacvalastro that is correct, no possibility to perform this

gianlucacvalastro, 2023-03-13:
Dear @shaibran,

thanks for your response.

Thank you for your availability and for your support.

Best regards

shaibran, 2023-08-10:
Summary: ethernet frame with Ethertype 0xAEFE (ECPRI) will not be supported as the underlying EC2 network is all IP based and not layer2.
Closing

closed 2023-08-10 by shaibran