amzn/amzn-drivers #263: ena driver showed high tx drop rate

JasonJiang-VM, 2023-02-15:
Hello, we are performance performance testing on aws environment with c5n.metal and c5n.18xlarge instance. 
The topology is:
  VM-CLIENT (c5n.18xlarge) ---- CME1(edge node 1 with VPN tunnel) -- VPN tunnel -- CME2 (edge node 2 with VPN tunnel) --- VM-SERVER(c5n.18xlarge)

During the testing, we running iperf3 client in the VM-CLIENT and iperf server in the  VM-SERVER. And the traffic will  go through the VPN tunnel between CME1 and CME2. And we found high tx drop on CME1.
We are using ENA driver 2.0.2 with DPDK 19.11. It's hard to upgrade the DPDK to newest.

perimeter-edge> get physical-port fp-eth1
Wed Feb 15 2023 UTC 09:24:05.994
Physical Port
UPT PASSTHRU CAPABLE : false
ADMIN_STATUS         : up
DRIVER               : net_ena
DUPLEX               : full
ESP_RSS_INFO         : {'enabled': False}
ID                   : 1
LINK                 : up
MAC                  : 02:87:05:6b:41:d9
MAX_RX_PKTLEN        : 9216
MAX_RX_QUEUES        : 32
MAX_TX_QUEUES        : 32
MBUFSIZE             : 2048
MIN_RX_BUFSIZE       : 64
MTU                  : 9000
NAME                 : fp-eth1
OFFLOAD_CAPABILITIES : TX_IPV4_CKSUM TX_UDP_CKSUM TX_TCP_CKSUM RX_IPV4_CKSUM RX_UDP_CKSUM RX_TCP_CKSUM
PCI                  : 0000:81:00:00
POLLING_STATUS       : active
REDIRECT-TO-KERNEL   : True
RETA_SIZE            : 128
RSS_HASH_FUNC        : err
RX_QUEUE             : 16
RX_QUEUE_SIZE        : 1024
RXCONF               : {'enable_scatter': 0, 'enable_lro': 0, 'jumbo_frame': 0, 'max_rx_pkt_len': 1518, 'scattered_rx': 0}
SOCKET               : 0
SPEED                : 0
TX_QUEUE             : 16
TX_QUEUE_SIZE        : 1024

We tried to change the tx ring size, but it seems 1024 is the biggest one.
For the tx packet drop, please see attachment for reference.  Please kindly give advice and how to fix it.
Thanks
Jason
<img width="1440" alt="Screen Shot 2023-02-07 at 13 29 16" src="https://user-images.githubusercontent.com/71364
[image]
665/218988583-7302dd57-df23-4956-9a42-7682e9e196bd.png">

zorikm, 2023-02-15:
Thank you @JasonJiang-VM,
Could you please provide your instance-id of CME1? Also it is not clear from your description what instance type was used for CME1 and CME2. How many flows do you use in your iperf3 test?

JasonJiang-VM, 2023-02-16:
Hi zorikm, 
Both CME1 and CME2 are using c5n.metal. 
Here is the commands on the VM-CLIENT (c5n.18xlarge): 
The port number is from [5001 - 5064]
iperf -c 10.100.80.12 -p 5001 -u -b 1000M -l 1250 -t 360000&
....
iperf -c 10.100.80.12 -p 5058 -u -b 1000M -l 1250 -t 360000&
....
iperf -c 10.100.80.12 -p 5064 -u -b 1000M -l 1250 -t 360000&

On the server side on the VM-SERVER(c5n.18xlarge):
iperf -u -s 10.100.80.12 -p 5001&
....
iperf -u -s 10.100.80.12 -p 5058&
....
iperf -u -s 10.100.80.12 -p 5064&

Bests
Jason

shaibran, 2023-02-16:
Hi, This is likely due to application flooding the queues.
Can you please share the instances IDs and the estimated time you executed the test? we can double check the stats on ec2 side.
You can check ENA statistics, see the stats [documentation ](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#8-driver-statistics) but most of the relevant stats (tx drops due to flooding HW queues, exceeding instance limiters) are supported from newer PMD versions only, so you'll need to apply backports.

JasonJiang-VM, 2023-02-17:
The instance id is : CME1 [i-0fd8248d0c759ca8d](https://us-west-2.console.aws.amazon.com/ec2/home?region=us-west-2#InstanceDetails:instanceId=i-0fd8248d0c759ca8d)  And CME2:  [i-055a6dbc90754a207](https://us-west-2.console.aws.amazon.com/ec2/home?region=us-west-2#InstanceDetails:instanceId=i-055a6dbc90754a207), we will performance testing during Beijing time 9:00am to 9:00pm.  We found aws underlay drop packets as well.
For the backports, I will give a try

JasonJiang-VM, 2023-02-17:
BTW, will you help to check whether the aws underlay networking sending pause frame to those two instances. Typically, the tx drop counter also means aws underlay networking can't process packets fast enough. 

Thanks
Jason

shaibran, 2023-08-20:
closing this issue. For future search sake, driver stats should expose when instance limiters are being hit, but this requires upgrading the ena to latest driver, see [driver statistics](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#8-driver-statistics) usage for more details

closed 2023-08-20 by shaibran