amzn/amzn-drivers #266: DPDK RSS set dest port only field

ricmli, 2023-03-31:
Hi All,

I would like to report an issue regarding setting dst port only hash field with DPDK PMD. Currently, it seems that the driver is not able to set the destination port hash field only, which is causing some difficulties in my project.

Based on my research in the doc and code, I have found that the feature to set dst port only hash field was introduced in DPDK v21.08 and ENA v2.4.0. However, it appears that this feature is not currently supported by the available ENA hardware.

I am wondering if there is any workaround in the driver that can be used to enable this feature, or if it requires hardware support. Additionally, I am curious to know if there are any plans to add support for this feature in the future.

Thank you for your time and attention.

ricmli, 2023-04-07:
The code to set rss_hf
```
 struct rte_eth_rss_conf* rss_conf;
 rss_conf = &port_conf.rx_adv_conf.rss_conf;
 rss_conf->rss_hf = RTE_ETH_RSS_NONFRAG_IPV4_UDP | RTE_ETH_RSS_L4_DST_ONLY |
                         RTE_ETH_RSS_L3_DST_ONLY;
 ...
 rte_eth_dev_configure(port_id, nb_rx_q, nb_tx_q, &port_conf);
```

The log from ENA PMD
```
ena_rss_hash_set(): Setting RSS hash fields is not supported. Using default values: 0xc30
```

irrelevelephant, 2023-04-08:
I have the same issue, but with Linux driver. I've tested on c5n, c6gn, and c7, and get `EOPNOTSUPP` in all cases. The driver readme states:

> Different combinations of L2/L3/L4 fields can be configured as inputs for hash functions.

Is this inaccurate, or are there some other steps we can take to achieve this configuration?

akiyano, 2023-04-09:
Hi @irrelevelephant,
Regarding your question about the Linux driver.
The sentence from the README: `Different combinations of L2/L3/L4 fields can be configured as inputs for hash functions.` is indeed inaccurate. 
Currently the ENA Linux driver uses the following input from the packet to calculate the RSS hash:
1. IP layer's Source IP and Destination IP
2. (if available) Transport Layer's Source port and Destination port. 

This is not configurable.

irrelevelephant, 2023-04-10:
@akiyano thank you for the quick response! Is the limitation in hardware/firmware, and hence also applicable to the DPDK driver?

If so, can you share any information about why the driver tree has admin controls for the headers used in the RSS hash? Was this planned as a feature at some point? We'd be very interested in such a feature.

shaibran, 2023-04-10:
Hi [ricmli](https://github.com/ricmli), regarding your question, it is a HW limitation. We are considering improvements to the RSS configuration, however, there is no timeline that I can share at this time.
Please see the DPDK readme [FAQ section](https://github.com/amzn/amzn-drivers/tree/master/userspace/dpdk#13-general-faq) which explains the warning log you saw `Setting RSS hash fields is not supported. Using default values: 0xc30` (This happens since RSS hash key is supported but hash functions is not.)

ricmli, 2023-04-10:
@shaibran Got it, thank you for the response! Looking forward to the improvements.

closed 2023-04-30 by amitbern-aws