amzn/amzn-drivers #270: Build error with kernel 6.3

erayaslan, 2023-05-24:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.8.6

### Custom Code

No

### OS Platform and Distribution

Linux rog 6.3.3-gentoo-dist #1 SMP PREEMPT_DYNAMIC Wed May 17 15:53:09 -00 2023 x86_64 AMD Ryzen 9 5900HX with Radeon Graphics AuthenticAMD GNU/Linux

### Bug description

Building with kernel 6.3 fails with
```
eraya@rog /tmp/amzn-drivers/kernel/linux/ena $ make
make -C /lib/modules/6.3.3-gentoo-dist/build M=/tmp/amzn-drivers/kernel/linux/ena modules
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-pc-linux-gnu-gcc (Gentoo 13.1.1_p20230513 p2) 13.1.1 20230513
  You are using:           gcc (Gentoo 13.1.1_p20230520 p2) 13.1.1 20230520
  CC [M]  /tmp/amzn-drivers/kernel/linux/ena/ena_netdev.o
  CC [M]  /tmp/amzn-drivers/kernel/linux/ena/ena_ethtool.o
  CC [M]  /tmp/amzn-drivers/kernel/linux/ena/ena_lpc.o
  CC [M]  /tmp/amzn-drivers/kernel/linux/ena/ena_phc.o
  CC [M]  /tmp/amzn-drivers/kernel/linux/ena/ena_xdp.o
  CC [M]  /tmp/amzn-drivers/kernel/linux/ena/dim.o
  CC [M]  /tmp/amzn-drivers/kernel/linux/ena/ena_devlink.o
/tmp/amzn-drivers/kernel/linux/ena/ena_devlink.c: In function ‘ena_devlink_params_get’:
/tmp/amzn-drivers/kernel/linux/ena/ena_devlink.c:111:15: error: implicit declaration of function ‘devlink_param_driverinit_value_get’; did you mean ‘devl_param_driverinit_value_get’? [-Werror=implicit-function-declaration]
  111 |         err = devlink_param_driverinit_value_get(devlink,
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |               devl_param_driverinit_value_get
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:252: /tmp/amzn-drivers/kernel/linux/ena/ena_devlink.o] Error 1
make[1]: *** [Makefile:2025: /tmp/amzn-drivers/kernel/linux/ena] Error 2
make: *** [Makefile:85: all] Error 2
```

### Reproduction steps

```bash
The following commits seem to be relevant:

https://github.com/torvalds/linux/commit/075935f0ae0fbbe469a911d685f6cc59de892700
https://github.com/torvalds/linux/commit/fb8421a94c5613fee86e192bab0892ecb1d56e4c
```

### Expected Behavior

Thank you

### Actual Behavior

.

davidarinzon, 2023-05-24:
Hi @erayaslan 
Thank you for raising this, we are aware of it and it will be fixed in the next release.

davidarinzon, 2023-05-28:
Hi @erayaslan 
The issue is expected to be fixed in version 2.8.7 which was just released (https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.8.7), you're welcome to check it and provide feedback.

erayaslan, 2023-05-28:
Works as expected. Thank you for the quick turn around. Closing. @davidarinzon

closed 2023-05-28 by erayaslan