amzn/amzn-drivers #271: [Bug]: RHEL 8.8 kernel: error: macro "netif_napi_add" passed 4 arguments, but takes just 3

ziggythehamster, 2023-05-27:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.8.6

### Custom Code

No

### OS Platform and Distribution

(kind of a pain to run `uname -a`)

Rocky Linux 8, kernel 4.18.0-477.10.1.el8_8.x86_64

### Bug description

It appears that RHEL/Rocky 8.8 requires the same alternate code path as RHEL 9.2 at https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/kcompat.h#L1096

### Reproduction steps

```bash
1. Attempt to compile ena 2.8.6 on Rocky Linux 8.8 with the latest kernel installed.
```

### Expected Behavior

It compiles

### Actual Behavior

It attempts to pass too many arguments to `netif_napi_add`

### Relevant log output

```shell
x86_64: Building module:
    x86_64: Cleaning build area...
    x86_64: make -j2 KERNELRELEASE=4.18.0-477.10.1.el8_8.x86_64 -C ena BUILD_KERNEL=4.18.0-477.10.1.el8_8.x86_64...(bad exit status: 2)
    x86_64: DKMS make.log for ena-2.8.6 for kernel 4.18.0-477.10.1.el8_8.x86_64 (x86_64)
    x86_64: Fri May 26 23:54:30 UTC 2023
    x86_64: make: Entering directory '/var/lib/dkms/ena/2.8.6/build/ena'
    x86_64: make -C /lib/modules/4.18.0-477.10.1.el8_8.x86_64/build M=/var/lib/dkms/ena/2.8.6/build/ena modules
    x86_64: make[1]: Entering directory '/usr/src/kernels/4.18.0-477.10.1.el8_8.x86_64'
    x86_64:   CC [M]  /var/lib/dkms/ena/2.8.6/build/ena/ena_netdev.o
    x86_64:   CC [M]  /var/lib/dkms/ena/2.8.6/build/ena/ena_ethtool.o
    x86_64: In file included from /var/lib/dkms/ena/2.8.6/build/ena/ena_netdev.h:9,
    x86_64:                  from /var/lib/dkms/ena/2.8.6/build/ena/ena_netdev.c:25:
    x86_64: /var/lib/dkms/ena/2.8.6/build/ena/kcompat.h: In function ‘ena_netif_napi_add’:
    x86_64: /var/lib/dkms/ena/2.8.6/build/ena/kcompat.h:1100:50: error: macro "netif_napi_add" passed 4 arguments, but takes just 3
    x86_64:   netif_napi_add(dev, napi, poll, NAPI_POLL_WEIGHT);
    x86_64:                                                   ^
    x86_64: compilation terminated due to -Wfatal-errors.
    x86_64: make[2]: *** [scripts/Makefile.build:317: /var/lib/dkms/ena/2.8.6/build/ena/ena_netdev.o] Error 1
    x86_64: make[2]: *** Waiting for unfinished jobs....
    x86_64: In file included from /var/lib/dkms/ena/2.8.6/build/ena/ena_netdev.h:9,
    x86_64:                  from /var/lib/dkms/ena/2.8.6/build/ena/ena_ethtool.c:10:
    x86_64: /var/lib/dkms/ena/2.8.6/build/ena/kcompat.h: In function ‘ena_netif_napi_add’:
    x86_64: /var/lib/dkms/ena/2.8.6/build/ena/kcompat.h:1100:50: error: macro "netif_napi_add" passed 4 arguments, but takes just 3
    x86_64:   netif_napi_add(dev, napi, poll, NAPI_POLL_WEIGHT);
    x86_64:                                                   ^
    x86_64: compilation terminated due to -Wfatal-errors.
    x86_64: make[2]: *** [scripts/Makefile.build:317: /var/lib/dkms/ena/2.8.6/build/ena/ena_ethtool.o] Error 1
    x86_64: make[1]: *** [Makefile:1616: _module_/var/lib/dkms/ena/2.8.6/build/ena] Error 2
    x86_64: make[1]: Leaving directory '/usr/src/kernels/4.18.0-477.10.1.el8_8.x86_64'
    x86_64: make: *** [Makefile:85: all] Error 2
    x86_64: make: Leaving directory '/var/lib/dkms/ena/2.8.6/build/ena'
    x86_64: STDERR: Error! Bad return status for module build on kernel: 4.18.0-477.10.1.el8_8.x86_64 (x86_64)
```
```

### Contact Details

gablk@amazon.com

amitbern-aws, 2023-05-27:
Thanks ziggythehamster for reaching to our support,
this is a known issue which will be fixed on version 2.8.7

davidarinzon, 2023-05-28:
Hi @ziggythehamster 
The issue is expected to be fixed in version 2.8.7 which was just released (https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.8.7), you're welcome to check it and provide feedback.

closed 2023-06-03 by davidarinzon