amzn/amzn-drivers #273: [Bug]: the devlink_set_features has been removed since Kernel 6.3

mehrdadrad, 2023-07-31:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.8.8

### Custom Code

No

### OS Platform and Distribution

Fedora 6.4.6-200.fc38.x86_64

### Bug description

The `devlink_set_features` function has been removed since Kernel 6.3 but the ENA driver still has it [here](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ena_devlink.c#L290) at master.
https://lore.kernel.org/netdev/20230107094909.530239-3-jiri@resnulli.us/T/

### Reproduction steps

```bash
During compilation on Kernel 6.4, you can see the below error:

/var/lib/dkms/amzn-drivers/2.8.8/build/kernel/linux/ena/ena_devlink.c:290:9: error: implicit declaration of function ‘devlink_set_features’; did you mean ‘netif_skb_features’? [-Werror=implicit-function-declaration]
  290 |         devlink_set_features(devlink, DEVLINK_F_RELOAD);
      |         ^~~~~~~~~~~~~~~~~~~~
      |         netif_skb_features
```

### Expected Behavior

.

### Actual Behavior

.

davidarinzon, 2023-08-01:
Thank you for raising this issue @mehrdadrad 
We will look into fixing this in the nearest release.

davidarinzon, 2023-08-10:
Hi @mehrdadrad 
We've released 2.8.9g with the fix, you're welcome to try it out and provide feedback.

closed 2023-08-16 by mehrdadrad