amzn/amzn-drivers #274: [Support]: queue_*_tx_llq_buffer_copy is too high

whatmatrix, 2023-08-07:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

2.8.8g

### Custom Code

No

### OS Platform and Distribution

6.1.42-399.aws

### Support request

My ethernet interface is mostly sending IPv4 acks and small packets. However, I see a very high tx_llq_buffer_copy in ethtool. I set force_large_llq_header=1 but it does not make a difference.

From tcpdump, I do not see exotic tcp options and payloads are < 100b. The following is a typical packet.

```
 Flags [.], ack 4876, win 32406, options [nop,nop,TS val 3521313542 ecr 1742829286], length 0
 Flags [P.], seq 193:289, ack 3280, win 32406, options [nop,nop,TS val 747437730 ecr 1982392270], length 96
```

I use z1d.metal instance and m5zn.3xlarge instance. I confirmed that `lspci -vvv` shows Region 2 in the ENA devices. I do not see LLQ related errors in dmesg.

I use the most recent Clear Linux. Since Clear Linux has a built-in ENA driver, I blacklisted in the kernel command-line using `initcall_blacklist=ena_init` and installed the ENA DKMS driver. I needed to rename the driver to awsena due to name conflicts with the built-in driver.

### Contact Details

minjae.hwang@gmail.com

davidarinzon, 2023-08-08:
Hi @whatmatrix 

Looking at `llq_buffer_copy` counter, I assume that you've checked the documentation at https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html. We went through the documentation and identified that there are inaccuracies in it, which will be fixed.
This counter counts the amount of times the data that needs to be copied to the LLQ part is not linear in the skb.
This is not an error metric/statistic.

whatmatrix, 2023-08-08:
Hi @davidarinzon 

Thank you very much for your answer. Yes, you are right that I read that documentation.

Could you elaborate more about "the LLQ part is not linear"? I am just curious if there is a way to encourage LLQ or overhead of LLQ (if there is any) by changing kernel parameters or applications.

davidarinzon, 2023-08-08:
Hi @whatmatrix,
LLQ (Low Latency Queues) is an improvement of the data path. Quoting from our Best Practices page:
```
Q: What is ENA LLQ aka ENAv2?

A: ENA LLQ is a Tx submission ring that is located in ENA device DRAM behind PCI BAR2.
The driver maps this area as Write Combined, and pushes there the Tx submission ring
descriptors along with packet headers. LLQs improves latency and PPS.
Starting with ENA driver v2.0 LLQ is the default mode of operation of majority of the instance types.
```
Therefore, I wouldn't think about LLQ as an overhead.

By "linear" I mean that the packet headers, which are pushed into the TX submissions ring mentioned above, of an incoming packet (which comes in an skb) are within the linear part of the skb or not.

davidarinzon, 2023-08-17:
Hi @whatmatrix 
Do you have any further questions?

closed 2023-08-27 by davidarinzon