amzn/amzn-drivers #278: [Support]: vfio-pci driver with DPDK issues on AWS

igorfardoc, 2023-09-05:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

-

### Custom Code

No

### OS Platform and Distribution

Linux 6.1.41-63.114.amzn2023.x86_64

### Support request

Hi!
I am trying to make DPDK work on my c6i.8xlarge instance with two additional NIC in the same subnet (inbound permissions seems to be okay), but have some problems probably with vfio-pci driver.
When I run testpmd application, some packets are sent, but no packets are received.
I saw some similar issues like [this](https://github.com/amzn/amzn-drivers/issues/102) or [this](https://github.com/amzn/amzn-drivers/issues/252), but did not find useful solution there.
Also I tried to ping NIC on standard ena driver and it worked as expected, but when I changed driver to vfio-pci using dpdk-devbind, all packets of ping are dropped.

Any thoughts what am I doing wrong?

shaibran, 2023-09-06:
Hello [igorfardoc](https://github.com/igorfardoc), thank you for reaching out to us.

I suspect this is related to an issue with this distribution when working with multiple ENIs. The distribution team is working on mitigation but I am unable to provide a timeline for the fix.
Please migrate to AL2 and this should likely allow you to work properly with multiple ENIs till the internal issue is mitigated.

If issue persists (and also as best practice for future issues), please always share the exact DPDK version details and if any other patches/backports were applied, so we will be able to attempt to reproduce in-house. In addition you can provide the transmitting/receiving instance IDs, ENI IDs, their region, and the time frame of the occurrence, so we can check our internal logs for issues.

BR,
Shai

igorfardoc, 2023-09-06:
I created a new instance with AMI amzn2-ami-kernel-5.10-hvm-2.0.20230822.0-x86_64-gp2. Packets can not be received. More info:
Instance Id: i-080a56cc04e3516d9
I use dpdk-stable-21.11.5, tried to use igb_uio driver with WC mode compiled from dpdk-kmods repo according to [instructions](https://github.com/amzn/amzn-drivers/blob/master/userspace/dpdk/README.md#621-igb_uio) and also tried vfio-pci without any patching, only set noiommu mode.
Network interfaces eni-02095cff7367112c0 and eni-00b9c59c23f4cd8d0 are used.

igorfardoc, 2023-09-06:
> Hello [igorfardoc](https://github.com/igorfardoc), thank you for reaching out to us.
>
> I suspect this is related to an issue with this distribution when working with multiple ENIs. The distribution [...]

I created a new instance with AMI amzn2-ami-kernel-5.10-hvm-2.0.20230822.0-x86_64-gp2. Packets can not be received. More info:
Instance Id: i-080a56cc04e3516d9
I use dpdk-stable-21.11.5, tried to use igb_uio driver with WC mode compiled from dpdk-kmods repo according to [instructions](https://github.com/amzn/amzn-drivers/blob/master/userspace/dpdk/README.md#621-igb_uio) and also tried vfio-pci without any patching, only set noiommu mode.
Network interfaces are eni-02095cff7367112c0 and eni-00b9c59c23f4cd8d0 are used.

shaibran, 2023-09-06:
Thanks, I will review our logs

igorfardoc, 2023-09-06:
> 

Also tried to run skeleton application, and get this error:
`sudo ./basicfwd
EAL: Detected CPU lcores: 16
EAL: Detected NUMA nodes: 1
EAL: Detected shared linkage of DPDK
EAL: Multi-process socket /var/run/dpdk/rte/mp_socket
EAL: Selected IOVA mode 'PA'
EAL: No available 1048576 kB hugepages reported
EAL: VFIO support initialized
EAL: Probe PCI driver: net_ena (1d0f:ec20) device: 0000:00:06.0 (socket 0)
EAL: Probe PCI driver: net_ena (1d0f:ec20) device: 0000:00:07.0 (socket 0)
TELEMETRY: No legacy callbacks, legacy socket not created
Port 0 MAC: 0a 2b 0b fa 38 1c
EAL: Error - exiting with code: 1
  Cause: Cannot init port 0
`

shaibran, 2023-09-06:
AL2 with DPDK 21.11 LTS was previously tested with igb_uio and it works well. In addition I looked in the instance on our side and I do not see any drops so this is likely some configuration issue.
DPDK 21.11 has ENA v2.5.0 in it which does not support multi-process and is ~2 YO driver, so it is better to migrate to later DPDK releases which also provide improved debug logs.
I suggest to verify that security group, ACL and route table are configured properly (specifically since I noticed that the two ENIs eni-02095cff7367112c0 and eni-00b9c59c23f4cd8d0 reside on the same instance), and stick to one configuration, e.g., DPDK 22.11 + igb_uio and try to execute the testpmd on different instances where each interface has a single additional ENI.

igorfardoc, 2023-09-06:
Thanks!
This fixed the issue

closed 2023-09-06 by igorfardoc

shaibran, 2023-09-07:
Glad to hear, and thank you for choosing EC2!
I will resolve the issue