amzn/amzn-drivers #279: [Bug]: ENA can drop succesful XDP redirect packets on error

zeffron, 2023-09-13:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.9.1

### Custom Code

No

### OS Platform and Distribution

Linux latest

### Bug description

Kernel eBPF mailing has found a bug in the ENA driver's handling of errors in XDP in the NAPI poll. It seems to be present in this out-of-tree driver as well. I'm also not 100% convinced that the in-tree driver is fixing it correctly, as it looks like there's a different function used for flushing for ENA?

https://lore.kernel.org/bpf/20230908135748.794163-2-bigeasy@linutronix.de/

### Reproduction steps

```bash
Have an error after one packet in the NAPI poll is successfully redirected.
```

### Expected Behavior

Successfully redirected packets are flushed.

### Actual Behavior

Successfully redirected packets are not flushed

### Contact Details

bpf@vger.kernel.org

zeffron, 2023-09-13:
Looks like my concern about in-tree not having fixed correctly isn't valid there, but the in-tree fix may not be directly portable here, because in-tree has [removed xdp_do_flush_map](https://lore.kernel.org/bpf/20230908143215.869913-1-bigeasy@linutronix.de/)

akiyano, 2023-09-13:
Hi,

We are aware of issue fixes that happen in the upstream driver.
We backport them to the github driver in subsequent github driver releases.
Is there anything else we can help you with regarding this?

Thanks,
Arthur

zeffron, 2023-09-13:
Nope, that's it, thanks!

davidarinzon, 2023-10-24:
Hi @zeffron 

The change that you've mentioned has been backported and integrated as part of 2.10.0g release (https://github.com/amzn/amzn-drivers/releases/tag/ena_linux_2.10.0), more specifically in https://github.com/amzn/amzn-drivers/commit/d7c3540e91a70d31fd16c24800b865acdf3326ea.

Let us know if you have additional concerns.

closed 2023-11-26 by davidarinzon