amzn/amzn-drivers #284: [Documentation]: Map of EFA driver state in upstream kernels

stewartsmith, 2023-10-04:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.

### Driver Type

Linux kernel driver for Elastic Fabric Adapter (EFA)

### Driver Tag/Commit

efa_linux_2.5.0

### Documentation issue description

Documentation on the state of the EFA driver in the upstream Linux kernel, and how that relates to the driver trees here would be beneficial for Linux distributions.

Ideally, all bug fixes would go to the appropriate stable trees, and upstream Linux is up to date with EFA drivers.

### Contact Details

trawets@amazon.com

mrgolin, 2024-04-18:
@stewartsmith Thanks for reaching out with this suggestion and apologies for the long handling time. We have updated EFA driver README file to address this request and it now has a paragraph describing the relation between the source in mainline kernel, in various Linux distributions' kernels and in this repository.
https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/efa/README#L29

mrgolin, 2025-01-01:
Closing the issue, feel free to reach out for follow-up questions.

closed 2025-01-01 by mrgolin