amzn/amzn-drivers #285: [Bug]: Linux ENA driver in crash loop

TimeToogo, 2023-10-23:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

2.8.6g

### Custom Code

No

### OS Platform and Distribution

Linux [omitted].ec2.internal 6.1.29-50.88.amzn2023.x86_64 #1 SMP PREEMPT_DYNAMIC Tue Jun  6 00:47:06 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

### Bug description

It seems one of our instances has the ENA driver in a crash loop, preventing it functioning properly.

**Instance ID:** `i-0a4ee90ceaaa6eb83`

This keeps appearing the syslog:
```
[86492.786178] ena 0000:00:05.0 ens5: Keep alive watchdog timeout.
[86492.787110] ena 0000:00:05.0: ENA device version: 0.10
[86492.787814] ena 0000:00:05.0: ENA controller version: 0.0.1 implementation version 1
[86492.788943] ena 0000:00:05.0 ens5: Trigger reset is on
[86492.855833] ena 0000:00:05.0 ens5: tx_timeout: 1
[86492.856445] ena 0000:00:05.0 ens5: suspend: 1
[86492.857079] ena 0000:00:05.0 ens5: resume: 1
[86492.857658] ena 0000:00:05.0 ens5: wd_expired: 9467
[86492.858315] ena 0000:00:05.0 ens5: interface_up: 10476
[86492.956012] ena 0000:00:05.0 ens5: interface_down: 10475
[86492.956748] ena 0000:00:05.0 ens5: admin_q_pause: 0
[86492.957415] ena 0000:00:05.0 ens5: reset_fail: 0
[86492.958059] ena 0000:00:05.0 ens5: queue_0_tx_cnt: 3502
[86493.055847] ena 0000:00:05.0 ens5: queue_0_tx_bytes: 4719871
[86493.056667] ena 0000:00:05.0 ens5: queue_0_tx_queue_stop: 0
[86493.057494] ena 0000:00:05.0 ens5: queue_0_tx_queue_wakeup: 0
[86493.058283] ena 0000:00:05.0 ens5: queue_0_tx_dma_mapping_err: 0
[86493.155981] ena 0000:00:05.0 ens5: queue_0_tx_linearize: 0
[86493.156742] ena 0000:00:05.0 ens5: queue_0_tx_linearize_failed: 0
[86493.157509] ena 0000:00:05.0 ens5: queue_0_tx_napi_comp: 1221
[86493.158236] ena 0000:00:05.0 ens5: queue_0_tx_tx_poll: 1221
[86493.256087] ena 0000:00:05.0 ens5: queue_0_tx_doorbells: 1252
[86493.256892] ena 0000:00:05.0 ens5: queue_0_tx_prepare_ctx_err: 0
[86493.257693] ena 0000:00:05.0 ens5: queue_0_tx_bad_req_id: 0
[86493.355542] ena 0000:00:05.0 ens5: queue_0_tx_llq_buffer_copy: 3282
[86493.356446] ena 0000:00:05.0 ens5: queue_0_tx_missed_tx: 0
[86493.357166] ena 0000:00:05.0 ens5: queue_0_tx_unmask_interrupt: 1221
[86493.357966] ena 0000:00:05.0 ens5: queue_0_tx_xsk_need_wakeup_set: 0
[86493.455819] ena 0000:00:05.0 ens5: queue_0_tx_xsk_wakeup_request: 0
[86493.456686] ena 0000:00:05.0 ens5: queue_0_rx_cnt: 1808
[86493.457421] ena 0000:00:05.0 ens5: queue_0_rx_bytes: 473534
[86493.458194] ena 0000:00:05.0 ens5: queue_0_rx_rx_copybreak_pkt: 1510
[86493.556010] ena 0000:00:05.0 ens5: queue_0_rx_csum_good: 1713
[86493.556773] ena 0000:00:05.0 ens5: queue_0_rx_refil_partial: 0
[86493.557553] ena 0000:00:05.0 ens5: queue_0_rx_csum_bad: 0
[86493.558255] ena 0000:00:05.0 ens5: queue_0_rx_page_alloc_fail: 0
[86493.656125] ena 0000:00:05.0 ens5: queue_0_rx_skb_alloc_fail: 0
[86493.656894] ena 0000:00:05.0 ens5: queue_0_rx_dma_mapping_err: 0
[86493.657650] ena 0000:00:05.0 ens5: queue_0_rx_bad_desc_num: 0
[86493.755615] ena 0000:00:05.0 ens5: queue_0_rx_bad_req_id: 0
[... 129 lines cut ...]
### Reproduction steps

```bash
Not sure, seems specific to single instance.
```

### Expected Behavior

ENA driver should not crash.

### Actual Behavior

ENA driver in crash loop.

### Additional Data

```bash
**Region:** `us-east-1`
**Instance ID:** `i-0a4ee90ceaaa6eb83`
**Instance Type:** `c6i.large`
```

### Contact Details

elliot@recall.ai

davidarinzon, 2023-10-23:
Hi @TimeToogo 

Thank for reaching out.
We recommend reaching out to AWS support in the standard support channels and convey all the information mentioned for a detailed analysis.

davidarinzon, 2023-10-23:
Hi @TimeToogo 

We have identified that your launched instance have been launched on a server with rare hardware issues, leading to network connection problems. We sincerely apologize for any inconvenience this may have caused, and want to assure you that we are actively working to resolve the issue. As part of our response, we are removing this server from our fleet to prevent any further disruptions. Please relaunch your instance.

TimeToogo, 2023-10-23:
Wow, thanks for the quick investigation and follow up @davidarinzon, really appreciate that.

Is it possible to get some credit applied to our account as this instance never got cleaned up until manual intervention leading to an increased AWS bill?

davidarinzon, 2023-10-23:
Hi @TimeToogo 

We are here to assist with technical questions and issues related to the different ENA drivers. For billing, please contact AWS billing and/or support. You may check https://aws.amazon.com/compute/sla/ for more info.

TimeToogo, 2023-10-23:
Thanks @davidarinzon I will reach out

closed 2023-10-23 by TimeToogo