amzn/amzn-drivers #289: [Bug]: /sys/class/net/xxx/speed is not readable

pstavirs, 2023-12-14:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

ena_linux_2.11.0-1-g765c823 

### Custom Code

No

### OS Platform and Distribution

Linux ip-172-31-5-201 6.2.0-1012-aws #12~22.04.1-Ubuntu SMP Thu Sep  7 14:01:24 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux  

### Bug description

speed in sysfs is not readable with or without sudo. mtu works however.

### Reproduction steps

```bash
1. cat /sys/class/net/xxx/speed

Replace `xxx` with the interface name
```

### Expected Behavior

Speed in Mbps to be printed

### Actual Behavior

Get an `Invalid argument` error

### Relevant log output

```shell
ubuntu@ip-172-31-5-201:~$ cat /sys/class/net/ens5/speed
cat: /sys/class/net/ens5/speed: Invalid argument
ubuntu@ip-172-31-5-201:~$ cat /sys/class/net/ens5/mtu
3000
ubuntu@ip-172-31-5-201:~$ sudo cat /sys/class/net/ens5/speed
cat: /sys/class/net/ens5/speed: Invalid argument
ubuntu@ip-172-31-5-201:~$ sudo cat /sys/class/net/ens5/mtu
3000
ubuntu@ip-172-31-5-201:~$ ethtool -i ens5
driver: ena
version: 2.11.0g
```

amitbern-aws, 2023-12-14:
Hi pstavirs,
EC2 instance does not report interface link speed. 
EC2 only provides the up-to bandwidth per instance type, which is shared between all network interfaces running on the instance.

Customers can retrieve their instance's bandwidth via AWS site:  https://aws.amazon.com/ec2/instance-types/
or via AWS CLI
For example:
```
aws ec2 describe-instance-types --instance-type c5n.xlarge --region us-east-1 | jq -r '.InstanceTypes[0].NetworkInfo.NetworkPerformance'
Up to 25 Gigabit
```

Thanks

pstavirs, 2023-12-14:
@amitbern-aws Thanks - I understand that EC2 has a total bandwidth cap per instance type. But in my view that's something akin to a " traffic policer" at the instance level. Interface level speed is a different entity. Or is that the 'ena' driver being essentially a paravirtual driver does not really have any real/physical speed and that's why it is a considered decision not to show any speed?

amitbern-aws, 2023-12-14:
@pstavirs
Yes.  It's an informed decision not to report the link speed for individual interfaces of an instance.  Since the bandwidth limit is aggregated across all the interfaces of the instance, the individual interface link speed is not an accurate metric and should not be used.

pstavirs, 2023-12-16:
@amitbern-aws IMHO link speed and instance bandwidth limit are two different things and the presence of the latter should not exclude reporting the former

nafeabshara, 2023-12-19:
Can you educate us why link speed would matter when it is not representing the actual bandwidth of the instance ?

We debated this internally when we define ENA and its attributes, and decided against publishing link speed because it will be more confusing to average customer because typically /speed is associated with bandwidth for non-virtualized and on-prem systems

pstavirs, 2023-12-20:
@nafeabshara Happy to present my perspective. Note that I'm a network engineer and my use case for the EC2 instance is also as a VNF, so the perspective is colored by that. **But it is applicable even for general non VNF apps.**

Let's consider an on-prem non-VM system with one link of X bandwidth. The fact that X bandwidth (in bits/sec) is available does not mean that the system will be able to utilize it fully due to CPU/memory/app-design (or packet size!) constraints.

Assuming the system is talking to another end point, the network in between may have a different and possibly **lower bandwidth Y** although the link speed is X.

I would consider the instance bandwidth to be something similar to this network bandwidth constraint.

Note that even if the app/customer uses the instance bandwidth - there is no guarantee that the system can utilize the full bandwidth due to the same constraints I mentioned above (CPU/memory et.al.)

Another way to view this multiple levels of bandwidth availability (link speed, instance policer, network bandwidth) is to see it as a hierachical QOS or policer.

amitbern-aws, 2024-01-07:
@pstavirs,
Thanks for your feedback.  We reviewed it internally, and we currently plan to leave the current supported implementation as-is.  As mentioned above, you may retrieve the instance's up-to bandwidth via AWS CLI.

Thanks

pstavirs, 2024-01-07:
@amitbern-aws Thanks for considering the feedback, Amit!