amzn/amzn-drivers #290: [Bug]: Why does the adjustment of the interrupt coalescing parameters seem not to take effect?

Data-crea, 2023-12-21:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

6.1.66-91.160.amzn2023.x86_64

### Custom Code

No

### OS Platform and Distribution

Linux ip-172-31-23-xxx.us-east-2.compute.internal 6.1.66-91.160.amzn2023.x86_64 #1 SMP PREEMPT_DYNAM

### Bug description

I'm testing the interrupt coalescing command for the ENA network card, and am mainly concerned with the tuning and effects of the rx usecs parameter. The testing tool uses the Nginx rpm package that comes with ena (only access_log is turned off) and wrk.

First use ethtool -C ens5 adaptive-rx off to turn off dim.
Then I used ethtool -C ens5 rx-usecs xx to adjust the parameters, and found that when the value of rx-usecs is greater than 64, the system (number of interrupts, cpu utilization, pps) does not seem to have any change, even if xx=1000.

Is that why? Is the underlying hardware limiting this, for example, limiting the maximum effective value of rx-usecs to 64us?

### Reproduction steps

```bash
client: ./wrk -c 400 -t 4 -d 3000 http://172.31.ipa.ipb:80
server: /sbin/nginx -c /etc/nginx/nginx.conf (with access_log disable)

The number of server cpu is 32, the number of qp is 16.
The irq list is {28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43}.
I have not change the irq affinity and keep this default.

I use 'dool' to check irq numers, and use 'top' and 'sar -n DEV 1' to check cpu utilization and pps.
dool --cpu --int -I 28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43 -n
https://github.com/scottchiefbaker/dool
```

### Expected Behavior

What is the act of digging deep into the underlying hardware? 
Why does rx-usecs no longer take effect after it is greater than 64?

### Actual Behavior

What is the act of digging deep into the underlying hardware? 
Why does rx-usecs no longer take effect after it is greater than 64?

Data-crea, 2023-12-21:
Actual test results display:
Disable adaptive-rx，rx-usecs = 1, tx-usecs=64:
[image]
[image]
[image]

Disable adaptive-rx，rx-usecs = 32, tx-usecs=64:
[image]
[image]
[image]

Disable adaptive-rx，rx-usecs = 64, tx-usecs=64:
[image]
[image]
[image]

Disable adaptive-rx，rx-usecs = 96, tx-usecs=64:
It behaves exactly the same as  rx-usecs=64

davidarinzon, 2023-12-21:
Hi @Data-crea 
Thank you for raising this concern.
In order to validate a few things, would it be possible for you to share the AWS instance information on which you've produced these results?

Data-crea, 2023-12-21:
> Hi @Data-crea Thank you for raising this concern. In order to validate a few things, would it be possible for you to share the AWS instance information on which you've produced these results?

Hi!
client:
instance id i-01fe5336720dc9258
public IPv4 DNS ec2-18-117-125-49.us-east-2.compute.amazonaws.com
public IPv4 addr 18.117.125.49
c6in.8xlarge
al2023-ami-2023.3.20231218.0-kernel-6.1-x86_64

server: 
instance id i-01b53b89a6f3cd137 
public IPv4 DNS ec2-18-224-4-108.us-east-2.compute.amazonaws.com
public IPv4 addr 18.224.4.108
c6in.8xlarge
al2023-ami-2023.3.20231218.0-kernel-6.1-x86_64
And I replace the ena.ko maked by:https://github.com/amzn/amzn-drivers    path: make kernel/linux/ena/

Data-crea, 2023-12-21:
In order to reduce the impact of tx interrupts, we can set it larger and then continue to find what happen on rx-usecs:
ethtool -C ens5 tx-usecs 1000000

I'm also still checking the reason and trying to find the answer :) 
Thanks David!

davidarinzon, 2023-12-21:
Hi @Data-crea 

On AWS EC2 gen6 instance types, the device uses the minimum value between TX and RX interrupt moderation values, given that TX interrupt moderation is static on 64usec, the behavior is as you've described.
We are planning to support both values in future generations.

Data-crea, 2023-12-22:
> Hi @Data-crea
> 
> On AWS EC2 gen6 instance types, the device uses the minimum value between TX and RX interrupt moderation values, given that TX interrupt moderation is static on 64usec, the behavior is as you've described. We are planning to support both values in future generations.

Hi, David! Thank you very much for your reply!

Do you mean device tx-usecs is fixed to 64?

I used 'taskset -c 0 sockperf tp -i 172.31.ipa.ipb -m 200 -t 3000 -p' on the server side for testing:

Turn off adaptive-rx.
When tx-usecs = 1, rx-usecs =20, txpps=860000/s, irqs=107k, this means that one tx irq is triggered per 9.3usec.

When tx-usecs = 64, rx-usecs =20, txpps=860000/s, irqs=37k, this means that one tx irq is triggered per 27usec.

According to what you said, irq usecs of device=min(rx-usecs, tx-usecs)=min(20,64)=20usec

So it seems that tx-usecs is adjustable. 

Since tx-usecs can be adjusted, can we set tx-usecs to a large value, such as 10000, and let's look at the issue of rx-usecs that I am exploring. But even if tx-usecs =10000, the effectiveness of rx-usecs seems to be limited to 64 usecs and below.

Thanks a lot!

davidarinzon, 2023-12-22:
Hi @Data-crea 
As you've previously mentioned, TX interrupt moderation can be adjusted using `ethtool`. The default value is is 64usec.
DIM algorithm is based on the traffic and it works in several buckets, it may be that based on the nature of the traffic that you're trying, DIM algorithm doesn't go higher than 64usec.

Data-crea, 2023-12-22:
> Hi @Data-crea As you've previously mentioned, TX interrupt moderation can be adjusted using `ethtool`. The default value is is 64usec. DIM algorithm is based on the traffic and it works in several [...]

I don't think this argument convinces me. I checked the ENA driver source code:
[image]

In EQE mode, the range of netdim rx-usecs is [1,8,64,128,256].
So why do you say " it may be that based on the nature of the traffic that you're trying, DIM algorithm doesn't go higher than 64usec."?

At the same time, I said that I have used 'ethtool -C ens5 adaptive-rx off' to turn off DIM.
Under this premise, I verified the effect of manual adjustment of rx-usecs, and then I opened the issue mentioned in the issue.

Data-crea, 2023-12-23:
Thanks! David! I think I have tested successfully!

Finally, I found that the underlying device seems to have some restrictions on rx-usecs?

For example, 0-8 rxusecs are device pinned to 8usecs, 64 usecs are device pinned to 50usecs, 128 usecs are device pinned to 90usecs, and 256 usecs are device pinned to 180usecs.

I tested a lot of data and calculated these? Is this right?

amitbern-aws, 2023-12-25:
@Data-crea
We will try to reproduce your findings.
Stay tuned for updates
Thanks

Data-crea, 2023-12-26:
> @Data-crea We will try to reproduce your findings. Stay tuned for updates Thanks

Thanks! Amitbern!

And DIM seems to coalesce packets when the traffic is low, such as nginx + wrk -c 1/-c 10.
Do you have want to optimize this?

davidarinzon, 2023-12-26:
Hi @Data-crea 
Thank you for your inquiries.
Providing a summarized answer to your queries from several comments.

On AWS EC2 Gen-6 instance-types, the device uses the minimum value set for Tx and Rx interrupt moderation. In addition, in this instance-type generation, the device supports a limited number of pre-defined values which are not fully aligned with the DIM buckets values, which is inline to the deviation you experience.

Independent Tx and Rx interrupt moderation values, improved accuracy, and granularity are supported on newer and all future generation instance-types (e.g. c7gn, r8g)

Moreover, the device currently supports time-based interrupt moderation. There is a roadmap plan to consider adding packet rate information in the future.

Data-crea, 2023-12-27:
> Hi @Data-crea Thank you for your inquiries. Providing a summarized answer to your queries from several comments.
>
> On AWS EC2 Gen-6 instance-types, the device uses the minimum value set for Tx and Rx [...]

Thank you very much for your answers!
I'll go and test further, but I'd like to know *why the different instance types are treated differently*?

davidarinzon, 2023-12-28:
Hi @Data-crea 

AWS offers a diverse range of instance types, each tailored to meet specific performance, computational, and hardware requirements.
The underlying hardware components, including networking interfaces, vary among these instance types.
Over time, AWS address bugs and introduces improvements to the underlying hardware.
New instance types often benefit from these advancements, incorporating fixes for previous hardware limitations and
leveraging refined algorithms to provide a better overall experience.

Data-crea, 2023-12-29:
OK, thank you very much! My question has been fully answered!
Best regards.