amzn/amzn-drivers #291: [Bug]: Inconsistent conntrack_allowance_available stats per interface

mtirpak, 2023-12-21:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

n/a

### Custom Code

No

### OS Platform and Distribution

Ubuntu 22.04.3, Linux 5.15.136

### Bug description

Two interfaces of the same EC2 instance report different `conntrack_allowance_available` stat values:
 - eth0 shows a continuously changing value (green)
 - eth1 occasionally changes only, with many hours of delay (yellow)
 
On this instance, most of the traffic is actually not routed via eth0.

[image]

### Reproduction steps

```bash
Attach a second ENI to an EC2 instance and generate traffic via that additional interface. Monitor the conntrack_allowance_available on both interfaces for an extended period.
```

### Expected Behavior

`conntrack_allowance_available` is an instance level metric as confirmed by AWS support. Therefore, it is expected that every interface of the instance shows the same value approximately.

If this is not the case, then I would like to ask whether scraping the metric on the first interface of the instance always gives an accurate result. Is it safe to scrape only eth0 to understand the current allowance and ignore the metric on every other interface? Could this be consistently done on every instance then?

In contrast, `conntrack_allowance_exceeded` is incremented at the interface level according to our experience. Could you please confirm that the exceeded counter is pegged on the interface that failed to create the new flow?

### Actual Behavior

The `conntrack_allowance_available` stats are updated with many hours of delay on the second interface of the same instance.  

### Contact Details

miklos.tirpak@emnify.com

davidarinzon, 2023-12-24:
Hi @mtirpak 

Thank you for raising these questions.
We are looking into this and will get back to you soon.

davidarinzon, 2023-12-31:
Hi @mtirpak 

AWS has investigated this issue and we can confirm that under specific circumstances, there is a delay in updating the `conntrack_allowance_available` metric reported for some of the interfaces, even though the actual tracked connection allowance is shared across all interfaces attached to the same network card. This is a reporting only issue and does not affect the actual number of tracked connections still available for the instance. We have isolated the defect to the underlying host’s software and we will provide an update in this thread once it has been resolved. As a temporary workaround until then, reading the `conntrack_allowance_available` metric that was most recently updated will give the accurate result.
If you have any further questions or comments regarding this matter, please contact us at: http://aws.amazon.com/support

closed 2024-04-24 by davidarinzon