amzn/amzn-drivers #292: [Bug]: bpftrace cannot obtain ena_io_poll return value

Data-crea, 2023-12-29:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

6.1.66-91.160.amzn2023.x86_64

### Custom Code

No

### OS Platform and Distribution

 6.1.66-91.160.amzn2023.x86_64 #1 SMP PREEMPT_DYNAMI

### Bug description

Try to get the return value using the following command:
bpftrace -e 'kretprobe: ena_io_poll {@[nsecs/1000000000]=count();}'

bpftrace cannot obtain the ena_io_poll return value. How should I get it? 
Modify the code? I added noinline which doesn't seem to work.

### Reproduction steps

```bash
Excute the following cmd:
bpftrace -e 'kretprobe: ena_io_poll {@[nsecs/1000000000]=count();}'
```

### Expected Behavior

Get the corresponding return value.

### Actual Behavior

Cannot get the corresponding return value.

### Additional Data

```bash
# bpftrace -e 'kretprobe: ena_io_poll {@[nsecs/1000000000]=count();}'
Attaching 1 probe...
cannot attach kprobe, probe entry may not exist
ERROR: Error attaching probe: 'kretprobe:ena_io_poll'
```

davidarinzon, 2023-12-31:
Hi @Data-crea 

This is not related to the ENA Linux driver but rather possibly to the installation and requirements of the `bpftrace` tool.
I recommend checking the public documentation on how to enable and run `bpftrace` succesfully.

I have tried the command above on Amazon Linux 2 and Ubuntu 22.04 after installing the required packages and it worked well.
`sudo bpftrace -e 'kretprobe: ena_io_poll {@[nsecs/1000000000]=count();}'`

amitbern-aws, 2024-01-03:
Hi @Data-crea,
We can confirm that this issue has been replicated in the Amazon Linux 2023 kernel (version 6.1). 
Although the problem is not related to the ENA driver, we have internally assigned it to the Amazon Linux kernel team.

To mitigate this issue, you can:
1. Use Amazon Linux 2 (5.10)
2. Use kretfunc
```
sudo bpftrace -e 'kretfunc:ena:ena_io_poll {@[nsecs/1000000000]=count();}'
```

We will provide an update once we receive a resolution from the Amazon Linux team.

Thank you for bringing this to our attention.

Data-crea, 2024-01-04:
Hi @amitbern-aws @davidarinzon 

> Hi @Data-crea, We can confirm that this issue has been replicated in the Amazon Linux 2023 kernel (version 6.1). Although the problem is not related to the ENA driver, we have internally assigned it [...]

Thank you very much for your effective response! I will try this!

> 
> We will provide an update once we receive a resolution from the Amazon Linux team.
> 
> Thank you for bringing this to our attention.

amitbern-aws, 2024-01-04:
Update from kernel team:
- The bug introduced in commit (["tracing/kprobes: Return EADDRNOTAVAIL when func matches several symbols"](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v6.1.60&id=1f38ead73f25a889b15fca0ae78b8712dab77692)) merged in upstream kernel v6.1.60 and broke the kprobe on kernel module symbols.
- The fix (["tracing/kprobes: Fix symbol counting logic by looking at modules as well"](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable-rc.git/commit/?h=linux-6.1.y&id=abbcb10d6691e6f98838753c216c92ecd560a59a)) will be merged to v6.1.71.

Amazon Linux 2023 will take the fix in the coming weeks,
Will update once its available.

Thanks

Data-crea, 2024-01-04:
I see it! Thanks again!

closed 2024-01-04 by Data-crea