amzn/amzn-drivers #294: [Support]: Number of XDP queues with ena

pstavirs, 2024-01-06:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.
- [X] I have followed the [AWS official troubleshoot documentation](https://docs.aws.amazon.com/AWSEC2/latest/UserGuide/troubleshooting-ena.html).
- [X] I have followed the driver [readme](https://github.com/amzn/amzn-drivers) and [best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst).

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Driver Tag/Commit

 6.2.0-1017-aws

### Custom Code

No

### OS Platform and Distribution

Linux  6.2.0-1017-aws #17~22.04.1-Ubuntu SMP Fri Nov 17 21:07:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

### Support request

I'm using a `c5.4xlarge` EC2 instance which has 16vCPUs (HT enabled)

Ideally, we should have max of 16 queues for `ens5`, but the ethtool shows max as only 8 - why?
```
Channel parameters for ens5:
Pre-set maximums:
RX: n/a
TX: n/a
Other: n/a
Combined: 8
Current hardware settings:
RX: n/a
TX: n/a
Other: n/a 
Combined: 8
```

But when I try to attach a XDP program I get an error -
```
libbpf: Kernel error message: ena: Failed to set xdp program, there is no enough space for allocating XDP queues, Check the dmesg for more info         
```
and the following message in dmesg
```
[  411.021969] ena 0000:00:05.0 ens5: Failed to set xdp program, the Rx/Tx channel count should be at most half of the maximum allowed channel count. The current queue count (8), the maximal queue count (8)   
```
Why is this limitation?

Effectively we can have only `n/4` queues with XDP where `n` is the number vCPUs.

Does this mean that I should if I need 'm' queues, I should use an EC2 instance type with `m*4` queues?

### Contact Details

pstavirs@ostinato.org

amitbern-aws, 2024-01-06:
Hi @pstavirs,

1. Please read [ENA best practices](https://github.com/amzn/amzn-drivers/blob/master/kernel/linux/ena/ENA_Linux_Best_Practices.rst) for queue count info.

2. The max queue count depends on the instance type and instance size you are using. Usually the number of queues exposed per ENI is calculated as MIN(MAX_NUM_QUEUES_PER_ENI, NUM_OF_VCPUS) MAX_NUM_QUEUES_PER_ENI is 8 for most of the instance types and up to 32 for network accelerated instances.
c5 instance family only support max of 8 queues,
You can try the [c5n](https://aws.amazon.com/blogs/aws/new-c5n-instances-with-100-gbps-networking/) family for larger queue count (upto 32).

3. When XDP is activated, there is a queue restriction set to half of the maximum permissible channel count. This limitation stems from the implementation of XDP support in the ENA driver. This decision was made to prevent the need for synchronization between concurrent flows of transmissions, (both kernel-stack packets combined with XDP_REDIRECT and XDP_TX).

Thanks

davidarinzon, 2024-02-21:
Hi @pstavirs 
Do you have any further questions?

pstavirs, 2024-02-21:
Not at the moment. Thank you for all the help.

The ENA best practices document was very helpful. It would be good to link it from the Linux Kernel ENA README.md

closed 2024-02-21 by pstavirs

davidarinzon, 2024-02-21:
Thank you for the feedback @pstavirs