amzn/amzn-drivers #296: [Feature request]: ENA TX, RX hardware timestamping

likewise, 2024-03-01:
### Preliminary Actions

- [X] I have searched the existing [issues](https://github.com/amzn/amzn-drivers/issues) and didn't find a duplicate.

### Driver Type

Linux kernel driver for Elastic Network Adapter (ENA)

### Motivation

Is there future possible support for TX and RX hardware packet timestamping on ENA, or is this roadmapped somewhere?

Also what is the closest we can get with the current kernel but also DPDK drivers wrt timestamping in software?

### Solution

IEEE1588

### Contact Details

leon@sidebranch.com

amitbern-aws, 2024-03-01:
Hi likewise,
1. We are looking into incorporating HW TX/RX packet timestamping in ENA device in the future, its definitely in our roadmap, so stay tuned for updates.
2.  Can you explain/elaborate on your DPDK question?

Thanks

likewise, 2024-03-04:
2. I wondered if we can already use software timestamping in the DPDK RX/TX loops. I will look into this myself also.

Thanks.

closed 2024-03-04 by likewise

likewise, 2024-03-22:
Hi Amit,

Is there a way to find out about the roadmap, especially on what type of instances this will become available (first) and on what generation of instances this will not be rolled out?

I noticed PTP support is only on ARM based instances for example?

Thanks.

amitbern-aws, 2024-03-25:
Hi likewise,

We don't publicly disclose our roadmap details,
Currently, PTP is available in Asia Pacific (Tokyo) Region, on all R7g instances,
But our team is indeed working on incorporating HW TX/RX packet timestamping on several ENA devices in the future.
We appreciate your understanding and ongoing support and we will update you as soon as we can.
Should you have any further questions or require assistance, please don't hesitate to reach out.

Thanks

amitbern-aws, 2024-03-27:
PTP is now available also in US East (N. Virginia) region on all R7g instances, and we will be expanding support to additional AWS Regions and EC2 Instance types.

https://aws.amazon.com/about-aws/whats-new/2024/03/amazon-time-sync-service-microsecond-accurate-time-us-east-n-virginia-region/

oliora, 2024-11-04:
Would be interested in getting HW TX/RX packet timestamping on EC2 instances as well

utkarshgupta137, 2025-06-11:
Hi, we would also be really interested in getting hardware tx/rx timestamping. Any updates on the roadmap?